liquidctl/liquidctl #462: Add ability to give custom names to sensors

ykuksenko, 2022-06-03:
I use the Corsair Commander Pro.
When listing the sensors I see 
```
Corsair Commander Pro
├── Temperature 1     27.1  °C
├── Temperature 2     34.6  °C
├── Temperature 3     32.1  °C
├── Temperature 4     28.9  °C
├── Fan 1 speed       1082  rpm
├── Fan 2 speed        900  rpm
├── Fan 3 speed       1047  rpm
├── Fan 4 speed        758  rpm
├── Fan 5 speed       1203  rpm
├── Fan 6 speed        534  rpm
...
```
It would be nice to be able to rename sensors from defaults so the output would look like:
```
├── CPU Temp                 27.1  °C
├── Chipset Temp             34.6  °C
├── Exhaust Temp             32.1  °C
├── Intake Temp              28.9  °C
├── Front Fan 1 speed        1082  rpm
├── Front Fan 2 speed        900   rpm
├── Top Fan 1 speed          1047  rpm
├── Top Fan 2 speed          758   rpm
├── Read Fan speed           1203  rpm
├── Bottom Fan speed         534   rpm
...
```

It would be good if all supported devices could be adjustable as I also use `Asetek 690LC (assuming EVGA CLC)`
Also I use the Prometheus exporter so if the names could be exported there would be awesome.
The exporter works nicely in a container btw. Thanks for the software!

jonasmalacofilho, 2022-06-03:
How would these be specified?  On a config file?

ykuksenko, 2022-06-04:
A config file would be great.
It may also be nice to identify each device in case there are more than one of the same type of device. - Maybe by usb address or serial number if that is available.

jonasmalacofilho, 2022-06-05:
> A config file would be great.

Yeah, I think that implementing what you're suggestion makes most sense with a config file...

... which is our plans, albeit somewhat stalled: #195.

---

> It may also be nice to identify each device in case there are more than one of the same type of device. - Maybe by usb address or serial number if that is available.

That we already have.  We just don't show these details on the human-readable `status` output because in most cases it's unnecessary or can easily be inferred by the user.  But the user can filter the device by USB port, bus address, serial number, etc.  And the machine-readable `status --json` includes all details.