liquidctl/liquidctl #463: RM1000i convert to C OverflowError

Th3M1k3y, 2022-06-03:
Sometimes getting an error when requesting status

Executing `liquidctl status`

Sometimes returning

`ERROR: unexpected error with Corsair RM1000i: OverflowError('Python int too large to convert to C int')`

other times it works as expected

### Debug
```
[DEBUG] (cli) (main): liquidctl: 1.9.1
[DEBUG] (cli) (main): platform: Windows-10-10.0.22000-SP0
[DEBUG] (cli) (_log_requirements): python: 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)]
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: version n/a (smbus)
[DEBUG] (cli) (_log_requirements): libusb-package: 1.0.26.0
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:1c0d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): found CorsairHidPsu: Corsair RM1000i
[DEBUG] (usb) (find_devices): HID device: 046d:c626 (usage_page=0x0001 usage=0x0008)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1b49 (usage_page=0xffc2 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 0a92:00d1 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1b49 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1915:520a (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1b49 (usage_page=0xffc0 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1b49 (usage_page=0xffc2 usage=0x0003)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1b49 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 9886:0002 (usage_page=0xff31 usage=0x0074)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1b49 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xffbc usage=0x0088)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
[DEBUG] (usb) (find_devices): USB device: 0b05:18f3
[DEBUG] (usb) (find_devices): USB device: 0a92:00d1
[DEBUG] (usb) (find_devices): USB device: 046d:c626
[DEBUG] (usb) (find_devices): USB device: 1915:520a
[DEBUG] (usb) (find_devices): USB device: 2109:0813
[DEBUG] (usb) (find_devices): USB device: 2109:0813
[DEBUG] (usb) (find_devices): USB device: 1022:149c
[DEBUG] (usb) (find_devices): USB device: 2109:2813
[DEBUG] (usb) (find_devices): USB device: 1b1c:1c0d
[DEBUG] (usb) (find_devices): USB device: 10de:1ada
[DEBUG] (usb) (find_devices): USB device: 046d:c52b
[DEBUG] (usb) (find_devices): USB device: 9886:0002
[DEBUG] (usb) (find_devices): USB device: 2109:2813
[DEBUG] (usb) (find_devices): USB device: 1022:149c
[DEBUG] (usb) (find_devices): USB device: 1b1c:1b49
[DEBUG] (usb) (find_devices): USB device: 1022:149c
[DEBUG] (cli) (main): device: Corsair RM1000i
[DEBUG] (usb) (clear_enqueued_reports): discarded 2 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 03:88:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 02:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 03:d2:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: fe:03:52:4d:31:30:30:30:69:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (cli) (log): detailed error: unexpected error with Corsair RM1000i: OverflowError('Python int too large to convert to C int')
Traceback (most recent call last):
  File "C:\Users\micha\AppData\Local\Programs\Python\Python39\lib\site-packages\liquidctl\cli.py", line 463, in main
    status = dev.get_status(**opts)
  File "C:\Users\micha\AppData\Local\Programs\Python\Python39\lib\site-packages\liquidctl\driver\corsair_hid_psu.py", line 235, in get_status
    return self._get_status_directly()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python39\lib\site-packages\liquidctl\driver\corsair_hid_psu.py", line 160, in _get_status_directly
    ('Current uptime', self._get_timedelta(_CORSAIR_READ_UPTIME), ''),
  File "C:\Users\micha\AppData\Local\Programs\Python\Python39\lib\site-packages\liquidctl\driver\corsair_hid_psu.py", line 297, in _get_timedelta
    return timedelta(seconds=secs)
OverflowError: Python int too large to convert to C int
[ERROR] (cli) (log): unexpected error with Corsair RM1000i: OverflowError('Python int too large to convert to C int')
```

---

Additional information about the system:

- Affected device(s): Corsair RM1000i
- Operating system and version: Windows 11 Pro - 21H2 - Build 22000.675
- Liquidctl installation method: pip
- Output of `liquidctl --version`: liquidctl v1.9.1 (Windows-10-10.0.22000-SP0)

jonasmalacofilho, 2022-06-03:
This data

```
[DEBUG] (usb) (read): read 64 bytes: fe:03:52:4d:31:30:30:30:69:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```

is the standard response to initialization handshake.

Is there another process (another liquidctl instance, or other software like iCue, CorsairLink, HWiNFO) trying to access the device at the same time?

jonasmalacofilho, 2022-06-03:
Also, please confirm that you `initialize`'d the device before this execution of `status`, as it's also weird that the previous responses in that log are all mostly empty/zero.

Th3M1k3y, 2022-06-03:
Got it working, not entirely sure what was wrong. I had icue installed before, but uninstalled it before trying with liquidctl.

I did initialize before doing status, but it got me to think something maybe went wrong during it, so instead of just restarting my computer, I tried to shut it down and remove power completely for a few minutes, then trying again, and now it works every time.

closed 2022-06-03 by Th3M1k3y

jonasmalacofilho, 2022-06-03:
Glad to hear.  Let me know if the issue returns.

commit 4c981275 (2023-01-12): corsair_hid_psu: assert that the response matches the command sent

A rare and so far not well understood issue with these PSUs has been an
occasional exception being thrown from timedelta:

    OverflowError: Python int too large to convert to C int

In both cases for which we have debug infos, #463 and #192 (part), it
seems that the cause is an invalid response from the PSU, probably
because of a race with another program simultaneously accessing it.

(A third case has been reported on Discord, but so far no debug info has
been provided to us. Still, it seems likely that once again there's
a [... cut]