liquidctl/liquidctl #464: Corsair Commander Core giving error: 'response does not match command'

Grooty12, 2022-06-04:
I have just gotten the Corsair H100i Capellix liquid cooler, and when i try to use any command with it, this output is given: `ERROR: unexpected error with Corsair Commander Core (experimental): AssertionError('response does not match command')`

Basically all commands used gives the error (initialize, status, set). The expected behavior is that they work, but for all of them this happens:
```
$ liquidctl --match core status
ERROR: unexpected error with Corsair Commander Core (experimental): AssertionError('response does not match command')
```

Here is the output of the --debug flag:
```
$ liquidctl --match core status --debug                                                                                                     1 ✘ 
[DEBUG] (cli) (main): liquidctl: 1.9.1
[DEBUG] (cli) (main): platform: Linux-5.18.1-arch1-1-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.10.4 (main, May 14 2022, 05:21:19) [GCC 12.1.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.5.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0037 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0053 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0051 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0036 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): found CommanderCore: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0xff58 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a3e (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b05:18a3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:1b33
[DEBUG] (usb) (find_devices): USB device: 0b05:18a3
[DEBUG] (usb) (find_devices): USB device: 8087:07dc
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 1b1c:0a3e
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 1024 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:02:0a:db:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 4 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 1024 bytes: 08:05:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00: [... cut ...]
[ERROR] (cli) (log): unexpected error with Corsair Commander Core (experimental): AssertionError('response does not match command')
```
Also, when I use the commands, my fans LED's basically blink once. Really weird!
---

Additional information about the system:

- Affected device(s): Corsair H100i Capellix (Commander Core)
- Operating system and version: Arch Linux
- Liquidctl installation method: official Arch Linux repository
- Output of `liquidctl --version`: `liquidctl v1.9.1 (Linux-5.18.1-arch1-1-x86_64-with-glibc2.35)`

jonasmalacofilho, 2022-06-04:
Partially related to #455 (although it doesn't involve the initial "incorrect data type" error seen there).

Cc: @ParkerMc

ParkerMc, 2022-06-04:
Yeah, it's related to the firmware update. I have it working locally but I need to clean it up some. Feel free to bug me ever now and then if I don't get it PRed soon.

jonasmalacofilho, 2022-06-05:
@ParkerMc ,

That's great to hear.

Do you think we might be able to include these fixes in the next release?  Since those might not be trivial and/or include new features, I would probably need to merge them by the 20th.

penapena, 2022-07-30:
Any update on this issue? I have the same response as OP. Running version 1.10.0.

wildewouter, 2022-08-10:
@ParkerMc you have any updates on this? Appreciate the work you’re all putting in.

ParkerMc, 2022-08-12:
> @ParkerMc you have any updates on this? Appreciate the work you’re all putting in. 

I have the concept for the fix down. It's a rework of the exact flow. I just need to make time to get it in. Maybe this weekend I'll find some time. I'm always open to explaining what needs to be done if someone has more time then me to get it done

wildewouter, 2022-08-15:
I'd love to contribute. Not sure if it'd be any good considering I'm completely new to the project.

hykelvinlee42, 2022-08-22:
> I have the concept for the fix down. It's a rework of the exact flow. I just need to make time to get it in. Maybe this weekend I'll find some time. I'm always open to explaining what needs to be [...]

@ParkerMc I'd love to help contributing. What concept do you have in mind?

ParkerMc, 2022-08-27:
> @ParkerMc I'd love to help contributing. What concept do you have in mind?

There are two things that need to be done
1. Reading(in `_send_command`) should repeat until it gets a packet that starts with `00`(that is the actual response) other packets can just be ignored
2. The general flow needs to be changed
    Currently, it's: reset -> set_mode -> read and/or write
    It should become: open endpoint(currently called set_mode) -> read and/or write -> close endpoint(currently called reset)

commit a2b43914 (2022-09-13): commander_core: support 2.10.219 firmware (#501)

Fixes: #464
Related: #455
closed 2022-09-13 by jonasmalacofilho