liquidctl/liquidctl #465: Unable to Initialize NZXT RGB & Fan Controller on Mac OS

jorge-rodr, 2022-06-05:
Hi,

I had been using liquidctl on Mac OS 13.3 mac with Corsair HX750i for several months without any issues. I recently bought a NZXT RGB & Fan Controller and now that both of these devices are plugged, I see that the NZXT RGB & Fan Controller is unable to initialize with the error message "**could not open NZXT RGB & Fan Controller, possibly due to insufficient permissions**". 
After looking into this issue I realized that in the installation instructions it was said that:

> "Access permissions are not a concern on platforms like macOS"

which then made me believe I might have encountered a bug? I followed the instructions to install liquidctl with brew exactly as they are given in this project.

I executed the commands to list, initialize and show status. 
When I tried to initialize I always get the same error message ("ERROR: could not open NZXT RGB & Fan Controller, possibly due to insufficient permissions").

```
$ liquidctl list
Device #0: NZXT RGB & Fan Controller
Device #1: Corsair HX750i
$ liquidctl initialize -n 0 
ERROR: Error: could not open NZXT RGB & Fan Controller, possibly due to insufficient permissions
$ liquidctl initialize -n 1
$ liquidctl status -n 1
Corsair HX750i
├── Current uptime                     18:12:06  
├── Total uptime             144 days, 17:12:06  
├── Temperature 1                          42.8  °C
├── Temperature 2                          34.8  °C
├── Fan control mode                   Hardware  
├── Fan speed                                 0  rpm
├── Input voltage                        230.00  V
├── +12V OCP mode                    Multi rail  
├── +12V output voltage                   12.09  V
├── +12V output current                    8.50  A
├── +12V output power                     98.00  W
├── +5V output voltage                     5.00  V
├── +5V output current                     4.88  A
├── +5V output power                      25.00  W
├── +3.3V output voltage                   3.31  V
├── +3.3V output current                   1.75  A
├── +3.3V output power                     6.00  W
├── Total power output                   116.00  W
├── Estimated input power                128.00  W
└── Estimated efficiency                     91  %
 $ liquidctl status -n 0    
ERROR: Error: could not open NZXT RGB & Fan Controller, possibly due to insufficient permissions
```

---

Additional information about the system:

- Affected device(s): NZXT RGB & Fan Controller
- Operating system and version: Mac OS mac 13.3.1 virtualized on Proxmox 7
- Liquidctl installation method: official brew installation for Mac OS
- Output of `liquidctl --version`: liquidctl v1.8.1 (homebrew; liquidctl 1.8.1)
- Python version: 3

---
As instructed the same commands shown above were run with --debug flags. See Below
```

user@mac ~ % liquidctl list
Device #0: NZXT RGB & Fan Controller
Device #1: Corsair HX750i
user@mac ~ % liquidctl initialize -n 0 
ERROR: Error: could not open NZXT RGB & Fan Controller, possibly due to insufficient permissions
user@mac ~ % liquidctl initialize -n 1
user@mac ~ % liquidctl status -n 1
Corsair HX750i
├── Current uptime                     18:12:06  
├── Total uptime             144 days, 17:12:06  
├── Temperature 1                          42.8  °C
[... 194 lines cut ...]
├── Temperature 1                          43.0  °C
├── Temperature 2                          35.0  °C
├── Fan control mode                   Hardware  
├── Fan speed                                 0  rpm
├── Input voltage                        230.00  V
├── +12V OCP mode                    Multi rail  
├── +12V output voltage                   12.09  V
├── +12V output current                    8.25  A
├── +12V output power                     98.00  W
├── +5V output voltage                     5.00  V
├── +5V output current                     4.69  A
├── +5V output power                      23.00  W
├── +3.3V output voltage                   3.31  V
├── +3.3V output current                   1.38  A
├── +3.3V output power                     4.50  W
├── Total power output                   126.00  W
├── Estimated input power                138.00  W
└── Estimated efficiency                     91  %

user@mac ~ % ls -l /usr/bin/python*
-rwxr-xr-x  76 root  wheel  167120 Mar 26 08:21 /usr/bin/python3
user@mac ~ % liquidctl --version
liquidctl v1.8.1 (homebrew; liquidctl 1.8.1)

```

jonasmalacofilho, 2022-06-05:
Hi @jorge-rodr,

First, thanks for the detailed report!

This is tricky one: everything works with one HID, but not with the other.  Are there any Mac or Proxmox settings that are different between these devices?

It's also noteworthy that the HID path is `DevSrvsID:4294968429`, but IIRC that's very different from what I've seen before from other users on Mac.  Although they weren't running it on Proxmox...

Anyway, if all settings are exactly the same for both devices, then we'll need to know the exact error the kernel returned to hidapi; unfortunately it isn't reported to us or stored anywhere.  Maybe Mac OS logged it... can you check it?  And do you have any experience with debugging C libraries on Mac?

jorge-rodr, 2022-06-05:
Hi @jonasmalacofilho,

I do not think there are different settings for these devices.  The Mac OS VM has direct access to the usb controllers via vfio passthrough. Also, both usb devices are attached to the same USB hub (a NZXT usb hub gen3). 

Regarding debugging C experience I have none. Maybe we could do a session on Skype, Microsoft Tems, or Webex to collect the logs you need? Or if you have some procedure on what to do I could try following that too.

I have been looking into what you said that some codes were different. An I have seen these codes in the file called 71-liquidctl.rules found in extra/linux 

```
NZXT RGB & Fan Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="2009", TAG+="uaccess"

NZXT RGB & Fan Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="200e", TAG+="uaccess"

NZXT RGB & Fan Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="2010", TAG+="uaccess"
```

I also see that in IOREG the device has a value of "2010" as idProduct, as in the last of those previous lines.

- [See this screenshot for IOREG details of the NZXT RGB fan controller.](https://imgur.com/a/sNuoOan)
- [See this screenshot for the Corsair PSU device ](https://imgur.com/a/xAEqANO)

jorge-rodr, 2022-06-05:
Also, with the aid of an app called Hackintool I was able to extract the following kernel logs:

`2022-06-05 09:18:21.242395+0200  localhost kernel[0]: (IOUSBHostFamily) <IOUSBHostFamily invocation function for block in AppleUSBHostPort::enumerateDeviceComplete(IOUSBHostDevice*, int)> 000010.532264 AppleUSB20HubPort@00810000: AppleUSBHostPort::enumerateDeviceComplete_block_invoke: enumerated 0x1e71/2010/010c (NZXT USB Device) at 12 Mbps
`

jorge-rodr, 2022-06-11:
This issue is solved. The reason this happened was that that the USB ports in the virtualized Mac VM were not properly mapped. After remapping with the tool [USBToolBox](https://github.com/USBToolBox/tool), now liquidctl works as intended. Now that the mapping is correct, initializing the NZXT RGB & Fan Controller device on Mac OS Monterey is no longer a problem.

closed 2022-06-11 by jorge-rodr

jonasmalacofilho, 2022-06-11:
Oh, good, and thanks for letting me know.