liquidctl/liquidctl #466: Corsair Commander Core error "operation not supported by driver for Corsair Commander Core"

DrScartch, 2022-06-06:
forgive the bad formating, new to all of this, went from win10 to manjaro this week

when trying to set a fan speed curve in liquid ctl I get the following error "ERROR: operation not supported by driver for Corsair Commander Core (experimental)"

```
 sudo liquidctl --match core set fan speed 20 10 30 20 40 30 50 45 60 70 90 100
```
this is what returns the error and i was expecting it to set the fan speed based on temps with the speed in %s or give me an indication the command was set or a misuse of it.

```
 sudo liquidctl --match core set fan speed 20 10 30 20 40 30 50 45 60 70 90 100 --debug

[DEBUG] (cli) (main): liquidctl: 1.9.1
[DEBUG] (cli) (main): platform: Linux-5.15.41-1-MANJARO-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.10.4 (main, Mar 23 2022, 23:05:40) [GCC 11.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.5.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1532:021e (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:006c (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): found CommanderCore: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 0b05:18f3
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 1532:006c
[DEBUG] (usb) (find_devices): USB device: 1532:021e
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[INFO] (cli) (log): detailed error: operation not supported by driver for Corsair Commander Core (experimental): NotSupportedByDriver()
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 469, in main
    _device_set_speed(dev, args, **opts)
  File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 286, in _device_set_speed
    dev.set_speed_profile(args['<channel>'].lower(), profile, **opts)
  File "/usr/lib/python3.10/site-packages/liquidctl/driver/commander_core.py", line 113, in set_speed_profile
    raise NotSupportedByDriver
liquidctl.error.NotSupportedByDriver
[ERROR] (cli) (log): operation not supported by driver for Corsair Commander Core (experimental)
```
Additional information about the system:

- Affected device(s): Corsair Commander Core
- Operating system and version: Manjaro Linux x86_64 
- Kernel 5.15.41-1-MANJARO
- Liquidctl installation method: Via arch linux repository? (went into konsole and did 
```
sudo pacman -S liquidctl
```
PC is on 32gb ram:corsair vengence ddr4 3600, ryzen 9 5900x, and radeon 6700xt
- Output of `liquidctl --version`: liquidctl v1.9.1 (Linux-5.15.41-1-MANJARO-x86_64-with-glibc2.35)

Notably the vengence isnt showing up in liquidctl too, but that is a whole different thing. 

Ty for reading and hopefully its understandable.

jonasmalacofilho, 2022-06-06:
Hi @DrScartch,

Only fixed fan/pump speeds (or, more accurately, duty values) are supported [right now](https://github.com/liquidctl/liquidctl/blob/v1.9.1/docs/corsair-commander-core-guide.md#programming-the-pump-and-fan-speeds).

---

> Notably the vengence isnt showing up in liquidctl too, but that is a whole different thing.

SMBus devices are also currently only supported on Intel boards, due to some kernel driver differences (related: https://github.com/liquidctl/liquidctl/issues/461#issuecomment-1146310583).

DrScartch, 2022-06-06:
Ahhhhh, thanks @jonasmalacofilho . That helps a lot. TY I guess now I either just alt tab when I need to adjust speeds or write some script or macro to execute a command for the fan speeds.

jonasmalacofilho, 2022-06-06:
For the script, take a look at `extra/yoda`... it's not production ready, but it's a starting point.

closed 2022-07-03 by jonasmalacofilho