liquidctl/liquidctl #467: Corsair One i300

ntpapatheodorou, 2022-06-12:
I bought a Corsair One i300 and I want to use it with Manjaro.
It isnt listed on liquidctl list.
Buy running lsusb I get these.
`Bus 004 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 004 Device 005: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS578 SATA 6Gb/s
Bus 004 Device 003: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 004 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0db0:82c7 Micro Star International USB Audio
Bus 001 Device 004: ID 1b1c:0c14 Corsair Corsair ONE
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 007: ID 8087:0033 Intel Corp. 
Bus 001 Device 006: ID 1462:7d29 Micro Star International MYSTIC LIGHT 
Bus 001 Device 010: ID 2109:8817 VIA Labs, Inc. USB Billboard Device   
Bus 001 Device 009: ID 046d:c541 Logitech, Inc. USB Receiver
Bus 001 Device 012: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 011: ID 1a40:0801 Terminus Technology Inc. USB 2.0 Hub
Bus 001 Device 008: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 005: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

Is there any hope that I can make it work or should I change to Windows (I really dont want too).
It seems difficult decision as in the long term I will have overheating issues by using Linux without the ability of fan/pump configuration.

Thanks in advance.

jonasmalacofilho, 2022-06-16:
Hi,

Can you provide the information asked in our _[Support for a new device](https://github.com/liquidctl/liquidctl/issues/new?assignees=&labels=new+device&template=new_device.md&title=%3CBrand+name%3E+%3Cmodel+name%3E)_ issue template?

> Is there any hope that I can make it work [...]

Probably yes.

Thanks

ntpapatheodorou, 2022-06-16:
@jonasmalacofilho thanks for your reply.
Here is my template.

Device type: **AIO liquid cooler of Corsair One i300**
Product page: **[https://www.corsair.com/us/en/Categories/Products/Systems/CORSAIR-ONE/CORSAIR-ONE-Compact-Gaming-PC/p/CS-9020032-NA](url)**
First-party software: Corsair **iCue**
Other commonly used tools: **no one that im aware of,**

The device appears to support the following features:
 - monitoring of temperatures or other environment sensors:  **yes**
 - monitoring of fan or pump speeds or duty cycles:  **yes**
 - monitoring of voltages, currents or power:  **yes**
 - configurable fan or pump speeds:  **yes**
 - configurable voltages or current/power limits: **yes** 
 - configurable lighting of embedded LEDs: **yes**
 - configurable lighting of accessories like RGB fans or LED strips: **yes**
 - **example image here [https://i.imgur.com/FJURohD.png](url) **

The device is physically connected to the host system using:
**- USB: to motherboard like any AIO corsair liquid cooler**

The device uses which protocols to communicate with the host system:
**I dont know to be honest.**

Additional information about the device:
```
$ libusb -v
Bus 001 Device 004: ID 1b1c:0c14 Corsair Corsair ONE
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c14 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 Corsair ONE
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7

```

Finally, I can help with:
- testing changes on Linux: yes
- testing changes on Windows: yes
- testing changes on Mac OS: no dont have access to mac os
- attempting to capture USB traffic:  yes if im guided
- attempting to capture I²C/SMBus traffic: yes if im guided
- analyzing traffic data:  yes if im guided
- documenting the protocol:  yes if im guided
- implementing the driver:  yes if im guided.

assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2022-07-03:
Thanks for the additional information.  Unfortunately, this device likely uses a different protocol than anything we've seen before, or at least a new extension to one of the old protocols.

The next step is to capture some USB traffic between it and the host and start analyzing it.  Our internal _[Capturing USB traffic](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md)_ guide will help you get started.

ntpapatheodorou, 2022-07-30:
@jonasmalacofilho done it should I upload the file somewhere ?

jonasmalacofilho, 2022-07-31:
@ntpapatheodorou yes, please.  Can you send a PR to [liquidctl/collected-device-data](https://github.com/liquidctl/collected-device-data/)?

ntpapatheodorou, 2022-08-17:
@jonasmalacofilho you got them right ?

jonasmalacofilho, 2022-08-18:
[Yes](https://github.com/liquidctl/collected-device-data/pull/12#issuecomment-1206648617), thank you.

Haven't had the time to look at it yet, unfortunately.