liquidctl/liquidctl #471: KeyError when setting RGB color for LED strip

XFaonAE, 2022-06-21:
I am able to set the fan speeds with
`sudo -s`
`liquidctl --match smart set fan1 speed 100`

But when I try RGB i get a key error for the color
`liquidctl --match smart set led1 color red`

---

Additional information about the system:

- Affected device(s): 
- Operating system and version: 
- Liquidctl installation method: 
- Output of `liquidctl --version`:

jonasmalacofilho, 2022-06-21:
What device do you have?

jonasmalacofilho, 2022-06-21:
Actually, nevermind.

The issue is that you haven't specified a valid lighting mode:

```
$ liquidctl --help
liquidctl – monitor and control liquid coolers and other devices.

Usage:
[ ... ]
  liquidctl [options] set <channel> color <mode> [<color>] ...
[ ... ]
```

On a related note, "red" wont also be recognized as a color: take a look at _[Supported color specification formats](https://github.com/liquidctl/liquidctl#supported-color-specification-formats)._

closed 2022-06-21 by jonasmalacofilho