liquidctl/liquidctl #472: Corsair Commander fan power option

Lepidos, 2022-07-01:
Hi,
Is there a fan power on option? On windows I disabled fans 4-6, that now I need to turn. 🤦‍♂️

jonasmalacofilho, 2022-07-03:
Hi,

Which of the Corsair Commander variants do you have?

Have you tried just setting the speed for the corresponding fans to something > 0%?

Additionally, can you provide the complete output of `liquidctl --match commander initialize --debug` and `liquidctl --match commander status --debug`?

Lepidos, 2022-07-04:
variant:
cl-9011110-ww

Yes I tried setting the speed > 0%.

```
[DEBUG] (cli) (main): device: Corsair Commander Pro
[DEBUG] (keyval) (__init__): data in /run/user/1001/liquidctl/vid1b1c_pid0c10/loc7
[INFO] (commander_pro) (initialize): bound to corsair_cpro kernel driver, assuming it is already initialized
[WARNING] (commander_pro) (_get_static_info_from_hwmon): some attributes cannot be read from corsair_cpro kernel driver
[DEBUG] (commander_pro) (_get_static_info_from_hwmon): [0, 0, 0, 0]
[DEBUG] (keyval) (store): stored temp_sensors_connected=[0, 0, 0, 0] (in /run/user/1001/liquidctl/vid1b1c_pid0c10/loc7/temp_sensors_connected)
[DEBUG] (hwmon) (get_string): read fan1_label: fan1 3pin
[DEBUG] (hwmon) (get_string): read fan2_label: fan2 3pin
[DEBUG] (hwmon) (get_string): read fan3_label: fan3 4pin
[DEBUG] (commander_pro) (_get_static_info_from_hwmon): [1, 1, 2, 0, 0, 0]
[DEBUG] (keyval) (store): stored fan_modes=[1, 1, 2, 0, 0, 0] (in /run/user/1001/liquidctl/vid1b1c_pid0c10/loc7/fan_modes)
Corsair Commander Pro
├── Temperature probe 1     No  
├── Temperature probe 2     No  
├── Temperature probe 3     No  
├── Temperature probe 4     No  
├── Fan 1 control mode      DC  
├── Fan 2 control mode      DC  
├── Fan 3 control mode     PWM  
├── Fan 4 control mode     N/A  
├── Fan 5 control mode     N/A  
└── Fan 6 control mode     N/A  
```

2nd command:
```
[DEBUG] (cli) (main): device: Corsair Commander Pro
[DEBUG] (keyval) (__init__): data in /run/user/1001/liquidctl/vid1b1c_pid0c10/loc7
[INFO] (commander_pro) (get_status): bound to corsair_cpro kernel driver, reading status from hwmon
[DEBUG] (keyval) (load): loaded temp_sensors_connected=[0, 0, 0, 0] (from /run/user/1001/liquidctl/vid1b1c_pid0c10/loc7/temp_sensors_connected)
[DEBUG] (keyval) (load): loaded fan_modes=[1, 1, 2, 0, 0, 0] (from /run/user/1001/liquidctl/vid1b1c_pid0c10/loc7/fan_modes)
[DEBUG] (hwmon) (get_string): read fan1_input: 555
[DEBUG] (hwmon) (get_string): read fan2_input: 547
[DEBUG] (hwmon) (get_string): read fan3_input: 669
[DEBUG] (hwmon) (get_string): read in0_input: 12114
[DEBUG] (hwmon) (get_string): read in1_input: 4995
[DEBUG] (hwmon) (get_string): read in2_input: 3391
Corsair Commander Pro
├── Fan 1 speed      555  rpm
├── Fan 2 speed      547  rpm
├── Fan 3 speed      669  rpm
├── +12V rail      12.11  V
├── +5V rail        5.00  V
└── +3.3V rail      3.39  V
```

jonasmalacofilho, 2022-07-05:
This feature (disabling fan channels) is news to us.  And, it seems, to the `corsair_cpro` HWMON driver too.

Can you unload the kernel driver (`modprobe -r corsair_cpro`) and run `liquidctl --match commander initialize --debug` again?  This might at least tells us how these "disabled" channels are reported (at the protocol level).

---

P.S. Passing `--direct-access` to liquidctl should have a similar effect, but it could also mess us the expectations of the kernel driver, so it's better to unload the driver.  You can reload it afterwards with `modprobe corsair_cpro`...

Lepidos, 2022-07-05:
`fan_modes=b'\x01\x01\x02\x00\x00\x00'`
 This is interesting.

jonasmalacofilho, 2022-07-05:
Apparently there must be some other/new way for the device to tell the host that some fan channels are disabled.

@MarshallAsch , IIRC you have a Commander Pro... can you look into this when you have some time?

---

P.S. Regarding the different types for `fan_modes`, that's unlikely to cause problems in practice, but I'll fix it later today nonetheless.

Lepidos, 2022-07-05:
Yes I have no temp probes connected.
On windows i disabled fan **port** 4-6.

jonasmalacofilho, 2022-07-06:
@MarshallAsch, I see.  I think about the UI side of it...

Can you and/or maybe @Lepidos figure out the messages required to set each fan channel to each of the possible states?

- auto detect mode
- manually set PWM control
- manually set DC control
- disabled

After that I'll try to take a look at how to implement this in liquidctl, as well as a patch for the kernel driver as well.

MarshallAsch, 2022-07-06:
Yup I'll take a look at some new packet traces tonight (and take a look at my old data dumps to see if I documented it originally) it feels like something I looked at originally but I do not recall if I wrote it down anywhere

MarshallAsch, 2022-07-06:
The original pr has some links to the protocol and those commands are documented there. Here is one if them https://github.com/audiohacked/OpenCorsairLink/issues/70. (I'll implement something with this today)

MarshallAsch, 2022-07-06:
@Lepidos the command that you need will be 

`self._send_command(0x28, [0x02, <fan num>, _FAN_MODE_<MODE>])`

where `<fan num>` is in `0-5` for fans `1-6` and `<MODE>` is one of `DC`, `PWM` or `DISCONNECTED`

You can issue that command wherever you want and it will persist. There is no CLI interface for it. 

@jonasmalacofilho what do you think of some options on the `initialize` command like `--fan[1-6]-mode=[DC,PWM,OFF]`?

MarshallAsch, 2022-07-06:
@Lepidos there is now a WIP branch https://github.com/liquidctl/liquidctl/pull/474 that adds flags to the `initialize` command.

jonasmalacofilho, 2022-07-06:
@MarshallAsch,

> [...] and `<MODE>` is one of `DC`, `PWM` or `DISCONNECTED`

How about to re-enable auto-detection?

> what do you think of some options on the `initialize` command like `--fan[1-6]-mode=[DC,PWM,OFF]`?

I mostly like it, although I would prefer if we could add just a single option.

Perhaps `--fan-mode=<channel>:<mode>[,...]` (e.g. `--fan-mode=4:auto,5:dc,6:off`)?  Also, by keeping the syntax fairly generic (`<key>:<value>[,...]`), its parser could be useful for other future features.

MarshallAsch, 2022-07-06:
I know this is a bad implementation I wanted to get something up quickly as a poc to show that it works and this is a possible implementation. I'm very  unhappy with the flags I added

jonasmalacofilho, 2022-08-03:
@MarshallAsch ,
 
> ohhh I really like that its a lot better. Ill make that change tonight

Gentle ping... do you need some help?

And did you end up finding out:

> How about to re-enable auto-detection?

MarshallAsch, 2022-09-27:
a ping for @Lepidos, did you end up testing the implementation in #474?

>  Ill make that change tonight

One very very long night later, it has been implemented. 

> And did you end up finding out:
>> How about to re-enable auto-detection?

No I did not 😢

Lepidos, 2022-09-27:
I tried, just could not get the right syntax.
Will give it another shot this week.

MarshallAsch, 2022-09-27:
> I tried, just could not get the right syntax.

the syntax has changed a bit from the original format. you can ping me on the discord server if you would like to ask some 'live' questions or would like me to tag along for the attempt

commit c03b445d (2022-10-17): commander_pro: add support for changing fan modes during init (#474)

Closes: #472.
closed 2022-10-17 by jonasmalacofilho