liquidctl/liquidctl #475: [FR] Add doc for profiles accepted to yoda

SteveBronder, 2022-07-07:
Am I reading it right that the profile given to yoda will be read as '(min_temp, max_temp),(min_duty, max_duty)'? Sorry for the silly question but it was kind of hard to figure out what to google so I could understand the fan profile scheme

SteveBronder, 2022-07-07:
Nope I was totally wrong, it's a set of tuples of `(temp, fan speed %)` so a curve can be passed like 

```
# 20c -> 20% fan, 30c = 30% fan, ...
(20,20), (30,30), (45, 60), (50, 100)
```

jonasmalacofilho, 2022-07-07:
> Nope I was totally wrong, it's a set of tuples of `(temp, fan speed %)` so a curve can be passed like

That's right.

For completeness, the comment at the top of the file has a few examples:

https://github.com/liquidctl/liquidctl/blob/9b058bacaf78916e2d6987e63977f7d2349ab733/extra/yoda#L37-L41

SteveBronder, 2022-07-07:
That's reasonable. For the future it might be nice to have a line in the docs such as 

```
The profile is passed as a single quote string of pairs giving each temperatures fan speed %. For example '(20, 50), (40, 65), (50, 100)' has the fan speed at 50% for temperatures at or below 20 C, 65% fan speed for 40 C, and speeds of 100% for temperatures equal to or above 50 C. Fan speeds are interpolated between the points using exponential smoothing.
```

jonasmalacofilho, 2022-07-07:
Thanks for the suggestion... I slightly adjusted it and added it to the top level comment (which is the only place where yoda is documented).

commit 7390e840 (2022-09-19): extra: yoda: document how profiles are passed into the tool

Closes: #475