liquidctl/liquidctl #477: Kraken X60: alert threshold option broken or unsupported by the device

jonasmalacofilho, 2022-07-08:
### Discussed in https://github.com/liquidctl/liquidctl/discussions/476

<div type='discussions-op-text'>

<sup>Originally posted by **Morsghost** July  8, 2022</sup>
I'm on an old Kraken X60 and I've configured liquidctl to replace CAM after years of trouble with it, so far the fan controls are working well so thanks for that. The only issue I'm having is that the default alert threshold of 45C is much too low, and the --alert-threshold command doesn't seem to be recognized. Does this work on legacy devices, is there any usage information I might be missing? Thanks. </div>

Morsghost, 2022-07-08:
How would I go about capturing data between CAM and the X60? Wireshark? I'm a little leary of reinstalling cam and removing the libusb driver since it took a few tries to get everything working in liquidctl, but I can try to help out since there probably aren't many X60s left around for testing.

jonasmalacofilho, 2022-07-09:
> How would I go about capturing data between CAM and the X60? Wireshark?

Yeah, and there's a bit more information about it in our [internal docs](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md).

> I'm a little leary of reinstalling cam and removing the libusb driver since it took a few tries to get everything working in liquidctl, but I can try to help out since there probably aren't many X60s left around for testing.

I understand... thanks, btw.

And what issues did you have with the LibUSB & driver setup, do you remember?

commit 5a042e49 (2022-08-11): asete: document the alerts are not supported by the X40 and X60

As there is no way for liquidctl to differentiate between a X60 and a
X61, this is the best we can do.

Closes: #477
closed 2022-08-11 by jonasmalacofilho