liquidctl/liquidctl #480: Commander Pro Get Max RPM command

MarcasRealAccount, 2022-07-20:
So I have been playing around with libusb in a C++ environment trying to figure out how to send the external temperature to the commander pro for fan curves, and I stumbled upon a question in the driver code stating "[need to figure out how to find out what the max rpm is for the given fan](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/commander_pro.py#L423)".
And I was looking at the current protocol information, and noticed a missing entry `0x22`, which turns out to be a getter to get the max rpm for a fan. The way to calculate the rpm from the percentage is: `percentage * max / 256`, because it turns out the Duty as you guys call it isn't really just 0 to 100, but 0 to 255, tho I doubt it'll ever use anything above like 2000 rpm anyways, I know my fans only go up to about 1600 rpm and they are the LL fans...

Oh and it would seem like passing 0x06 and up results in an error code in byte 0 of value 0x18, no clue what it means, but I noticed that.

So anyways the command is as follows:
### Get max fan RPM value -  `0x22`

Request:

| Byte index | Description |
| ---------- | ----------- |
|  0x00 | 0x22 |
|  0x01 | fan number (0x00 - 0x05) |

Response:

| Byte index | Description |
| ---------- | ----------- |
| 0x00 | 0x00 |
| 0x01 | max rpm MSB |
| 0x02 | max rpm LSB |

MarcasRealAccount, 2022-07-20:
Actually I just figured out it is actually a command to get the percentage back...

Kinda weird how it still gave me the correct value as a maximum.  
But it could be that it has multiple access values or something,
Lemme try just passing in extra values and see what happens.

MarcasRealAccount, 2022-07-20:
Oh completely forgot to add the spec so far:
### Get fan percentage value -  `0x22`

Request:

| Byte index | Description |
| ---------- | ----------- |
|  0x00 | 0x22 |
|  0x01 | fan number (0x00 - 0x05) |

Response:

| Byte index | Description |
| ---------- | ----------- |
| 0x00 | 0x00 |
| 0x01 | unknown |
| 0x02 | percentage  |

MarcasRealAccount, 2022-07-20:
Ok after a bit of Wireshark action, I might've found 2 new commands:
### Set fan external temp -  `0x26`

Request:

| Byte index | Description |
| ---------- | ----------- |
|  0x00 | 0x26 |
|  0x01 | fan number (0x00 - 0x05) |
|  0x02 | temp MSB |
|  0x03 | temp LSB |

### Set channel brightness -  `0x39`

Request:

| Byte index | Description |
| ---------- | ----------- |
|  0x00 | 0x39 |
|  0x01 | channel |
|  0x02 | percentage |

Ok after testing the first one I can say it is absolutely correct :D
*Finally I can move over to linux without temperature problems*

jonasmalacofilho, 2022-07-21:
Thanks for looking into these a bit more.

> Actually I just figured out it is actually a command to get the percentage back...

Oh... : \

Well, we can still do it manually, by setting all fans to 100% and then reading and caching their maximum reported rpm speed.

It could be done in `initialize` (when there's no hwmon driver), and/or in `set_speed_profile` (but caching the result so it's generally only required once).

> A side note to `0x26` you have to periodically send the command, or the commander pro defaults to 1000 RPM, at least does that for me, might be some default specified somehow...

That looks like a fail-safe, which makes sense, in case the software sending the external temp itself crashes.  The specific fail-safe rpm/duty value might be configurable too.

jonasmalacofilho, 2022-07-21:
Forgot to say that even if `0x22` only reports the duty/percentage, we could still use it to augment the data reported by `get_status()`.

MarcasRealAccount, 2022-07-21:
> Well, we can still do it manually, by setting all fans to 100% and then reading and caching their maximum reported rpm speed.

Yeah that is fully possible, then maybe doing it async would be best, considering the RPM received from `0x21` isn't instant, and I doubt anyone would like to have it wait a second for each commander pro (At least I wouldn't want that)...

> The specific fail-safe rpm/duty value might be configurable too.

Sadly it seems a bit difficult to figure that one out, as it doesn't seem like ICue is configuring it on init.

Tho I will say there was this `0x03` getter command that I'm kinda curious what is.
I received `F8, 0D, B3, C9` from one and `B2, 99, E5, CB` from the other.

And then there's also `0x01`, `0x04`, `0x05`, `x07`, `0x08` and `0x09`, which I know does succeed and do exist, but idk what they do...  
I know one of those end up putting the commander pros into update mode or something, making ICue the only way of updating them and restoring them without a reboot (I'm just guessing that it cancels the update mode after a reboot)...

jonasmalacofilho, 2022-07-21:
> Yeah that is fully possible, then maybe doing it async would be best, considering the RPM received from `0x21` isn't instant, and I doubt anyone would like to have it wait a second for each commander pro (At least I wouldn't want that)...

I don't think that's possible, because other things could happen in the meantime and change the duty from under us.  (And liquidctl isn't setup for async, especially not the CLI, which is ostensibly "one-shot" style).

But some devices already require this to function at all, and I think 1 or 2 seconds isn't that bad, if it's only needed once per boot.  (A user with two devices could also do this in parallel).

> Sadly it seems a bit difficult to figure that one out, as it doesn't seem like ICue is configuring it on init.

Oh well...  I don't think it's super important, and it's still good to know that there's _some_ fail-safe functionality.

But have you checked whether it doesn't just fall back to the previously set fixed duty?  I've seen that with another device.

> I know one of those end up putting the commander pros into update mode or something, making ICue the only way of updating them and restoring them without a reboot (I'm just guessing that it cancels the update mode after a reboot)...

Probably better to not test that.  One of the devices I own does _not_ get out of "update" mode by itself, even after a reboot.  Or at least it didn't one time, when it got stuck there by accident... later I was able to complete the firmware update and get it unstuck, but only through the first party software.

MarcasRealAccount, 2022-07-21:
> But have you checked whether it doesn't just fall back to the previously set fixed duty? I've seen that with another device.

You know that is probable, cuz I did set the fixed rpm to 1000 at the start, gonna check it real quick.

> Probably better to not test that. One of the devices I own does not get out of "update" mode by itself, even after a reboot.

Ah ok, guess I'll leave them unknown for now...

MarcasRealAccount, 2022-07-21:
> You know that is probable, cuz I did set the fixed rpm to 1000 at the start, gonna check it real quick.

Ok I tested it and it didn't care if I set the fixed rpm previously to 2000.  
Still went to 1000 rpm when I stopped the test.