liquidctl/liquidctl #481: AttributeError: 'PosixPath' object has no attribute 'readlink' wit Commander Pro

maclarsson, 2022-07-23:
I tried to upgrade from liquidctl version 1.6.1 to 1.10.0 and get the following error when performing list / initialize commands:
```
$ liquidctl list
AttributeError: 'PosixPath' object has no attribute 'readlink'
```

The culprit seems to be with the 'Corsair Commander Pro' and the hwmon driver in my system (see debug output below), as another device (NZXT Kraken X3) can be initialized successfully.

```
$ liquidctl list --debug
[DEBUG] (cli) (main): liquidctl: 1.10.0
[DEBUG] (cli) (main): platform: Linux-5.16.15-051615-generic-x86_64-with-glibc2.29
[DEBUG] (cli) (_log_requirements): python: 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.0.2
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1050:0407 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1050:0407 (usage_page=0xf1d0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 0c45:5004 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0c45:5004 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0c45:5004 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 0c45:5004 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0c45:5004 (usage_page=0xff1c usage=0x0092)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b95:6801 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c10 (usage_page=0x0084 usage=0x0052)
Traceback (most recent call last):
  File "/home/maclarsson/.local/bin/liquidctl", line 8, in <module>
    sys.exit(main())
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/cli.py", line 413, in main
    selected = list(find_liquidctl_devices(**opts))
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/driver/__init__.py", line 60, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 572, in find_devices
    yield from drv.probe(handle, vendor=vendor, product=product, **kwargs)
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 113, in probe
    dev = cls(handle, desc, **consargs)
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/driver/commander_pro.py", line 134, in __init__
    super().__init__(device, description, **kwargs)
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 204, in __init__
    self._hwmon = HwmonDevice.from_hidraw(device.path)
  File "/home/maclarsson/.local/lib/python3.8/site-packages/liquidctl/driver/hwmon.py", line 66, in from_hidraw
    module = (sys_device / "driver" / "module").readlink().name
AttributeError: 'PosixPath' object has no attribute 'readlink'
```

---

Additional information about the system:

- Affected device(s): Corsair Commander Pro
- Operating system and version: Linux Mint 20.3 Cinnamon
- Liquidctl installation method: pip via PyPI
- Output of `liquidctl --version`: liquidctl v1.10.0 (Linux-5.16.15-051615-generic-x86_64-with-glibc2.29)

maclarsson, 2022-07-23:
I was able to correct the problem in the source code by changing line 66 in `/driver/hwmon.py` from 

`module = (sys_device / "driver" / "module").readlink().name`
to
`module = (sys_device / "driver" / "module").resolve().name`

Maybe it's related to the Python version (I am running 3.8.10 on this system)?

maclarsson, 2022-07-23:
Sorry for the spam :-)...

Just checked: according to https://docs.python.org/3.9/library/pathlib.html#pathlib.PosixPath the `.readlink()` function was added in Python 3.9 while `.resolve()` is available since at least 3.5.

I don't know if .resolve() is exactly the same as .readlink(), but changing it would improve compatibility with systems lagging behind with Python updates (like Linux Mint ;-) )

jonasmalacofilho, 2022-07-23:
Thanks for reporting and already finding the solution to this bug!

While they are indeed different...

_—`resolve()` will return an absolute path after following all symlinks, while `readlink()` just returns that (first) symlink's target—_

...we're only interested in the name component, and sysfs doesn't create the symlink in a particularly complex way.

In practice:

```txt
(Pdb) p (sys_device / "driver" / "module").readlink()
PosixPath('../../../../module/nzxt_grid3')
(Pdb) p (sys_device / "driver" / "module").resolve()
PosixPath('/sys/module/nzxt_grid3')
```

Do you want to submit a PR with the fix?

---

> Sorry for the spam :-)...

All of your messages were most definitively _not_ spam! : )

maclarsson, 2022-07-23:
Thank you for the feedback, I created the PR.

The fix seems to work fine with my own frontend, but how can I run the test suite?

jonasmalacofilho, 2022-07-24:
Thanks! Already reviewed and merged it.

> The fix seems to work fine with my own frontend, but how can I run the test suite?

The CI/online test suite requires the explicit approval on a maintainer for first-time contributors (in this case it's a GitHub thing, but other CI systems implemented similar restrictions because cryptomining abuse a while back).

The local suite can be executed with `python -m pytest`.

PR #483 (2022-07-24): Fix small bug in hwmon.py with Python <3.9
Change PosixPath.readlink() to PosixPath.resolve() for compatibility with Python <3.9

The .readlink() function is not available in the pathlib library prior to Python 3.9. Replacing it with a call to .resolve() for the same result.

Fixes: 
Closes: #481
Related:  

---

Checklist:

- [x] Adhere to the [development process]
- [x] Conform to the [style guide]
- [x] Verify that the changes work as expected on real hardware
- [ ] Add automated tests cases
- [ ] Verify that all (other) automated tests (still) pass
- [ ] Update the README and other applica [... cut]
commit 01f89996 (2022-07-24): hwmon: replace readlink() with resolve() for Python <3.9 compatibility (#483)

Fixes: #481
closed 2022-07-24 by jonasmalacofilho