liquidctl/liquidctl #485: NZXT Smart Device V2 (New Product ID 0x2019)

johandc, 2022-07-25:
Hello,

NZXT added another device with product id 0x2019

This is a 3 FAN and 6 LED driver. Apparantly now available also on their webpage:
https://nzxt.com/product/rgb-and-fan-controller

I tried to add a new entry to the driver, and it seems to be working:

```
liquidctl --verbose list
Device #2: NZXT RGB & Fan Controller
├── Vendor ID: 0x1e71
├── Product ID: 0x2019
├── Release number: 0x0106
├── Serial number: 0A00603BAA3584886082E26E
├── Bus: hid
├── Address: /dev/hidraw9
└── Driver: SmartDevice2
```

```
liquidctl -n 2 set fan1 speed 0
```

Works! :)

But...

```
 johan  ~  liquidctl -n 2 set fan1 speed 20 30  30 50  34 80  40 90  50 100
ERROR: operation not supported by NZXT RGB & Fan Controller
```

Does not seem to work...

Maybe it would work with Coolero managing the speed. But i think coolero bundles its own liquidctrl, so it's not very easy for me to test before we get an update of coolero too.

Finally, I can help with:

- testing changes on Linux: 
- attempting to capture USB traffic: 
- implementing the driver:

johandc, 2022-07-25:
[image]

It is confirmed to be working with coolero too.

jonasmalacofilho, 2022-07-26:
Hi!

> I tried to add a new entry to the driver, and it seems to be working:

Does the new entry look something like the following?

```py
        (0x1e71, 0x2019, None, 'NZXT RGB & Fan Controller (2022) (experimental)', {
            'speed_channel_count': 3,
            'color_channel_count': 6
        }),
```

Have you confirmed all 6 LED channels are working?

> ```
> johan  ~  liquidctl -n 2 set fan1 speed 20 30  30 50  34 80  40 90  50 100
> ERROR: operation not supported by NZXT RGB & Fan Controller
> ```
> Does not seem to work...

That's expected, as the device itself doesn't support fan profiles.  How about static fan speeds, are they working?

> Maybe it would work with Coolero managing the speed. But i think coolero bundles its own liquidctrl, so it's not very easy for me to test before we get an update of coolero too.

Yeah, Coolero implements those in software, on top of what the device (and liquidctl) supports.

Thanks!

johandc, 2022-07-28:
Thanks for the feedback! Yes your entry looks exactly like mine, and it works with static control perfectly thanks. Also i was able to fetch and build a coolero from source, and use it with the system version of liquidctl, and indeed i can now adjust the fan curves on this. It would be great for the next version of liquidctl to have that :)

P.s.:

Btw. i had a chat with NZXT as to why their minimum duty cycle is 20%, when for example their fans in reality operate down to as little as 8%. They seem to be too lazy to actually implement proper fan tuning in their CAM software. So instead they put the limit physically inside the Fan controller firmware. Which is just stupid. Because it prevents cool hackers, from doing what they were supposed to do in the first place. My asus freaking BIOS got that right. However, the asus can go down to 8%, it cannot go to 0% - which is just stupid. The other mistake the NZXT people as in their CAM software is the inability to control the fans based on the liquid temperature. Which is another shot in the foot to themselves.

The best possible setup i found, was to use Coolero with liquidctl. The pump controlled by CPU/Water delta temp, the first fan to turn on at water temp of 40 degrees and the seoncd to turn on at 45 degrees. Both with a slope to 100 percent at 50 and 55 respectively. What this does is it prevetns both fans from runnning the same RPM and causing annoying slow resonsance, when they are fading in an out of phase match. The second thingit does it being completely silent for around 1 hour of operation while water temp is slowly rising. At low or idle loads, the first fan would come on once per 10 minutes or so, and run for about 1 minue at the absolute lowest RPM.

P.p.s: Without any case airflow, the PSU does heat soak (like the water eventually), so the loudest component is now the stupid corsair SF600, which uses an agressive fan control going up and down in RPM every two seconds or so. Very very annoying. Ideally i would hook into coolero a temp sensor, and control the fan of the PSU too. I will also be tryuing out retrofitting it with some large passive heatsinks.

jonasmalacofilho, 2022-07-28:
> Thanks for the feedback! Yes your entry looks exactly like mine, and it works with static control perfectly thanks. Also i was able to fetch and build a coolero from source, and use it with the [...]

As long as we don't find a last minute major difference with these new models, it should be in the next release : )

Can you confirm whether you tested all six LED channels?
>> Have you confirmed all 6 LED channels are working?

---

> P.s.: [...]

Yeah, these arbitrary limitations are disappointing & frustrating.  And it's surprisingly not trivial to find controllers that just provide a "raw" interface for controlling fan speeds...

johandc, 2022-07-30:
> As long as we don't find a last minute major difference with these new models, it should be in the next release : )

Perfect, thank you very much.

> Can you confirm whether you tested all six LED channels?

I did not test any of it's LED features, no.

jonasmalacofilho, 2022-08-03:
> I did not test any of it's LED features, no.

In that case I'll need to hold off adding the change for now (ideally until others have a chance to test and confirm that they're working).  Otherwise we'll have something that we think is working but that might be terribly broken.

johandc, 2022-08-03:
Okay, but isn't that what experimental support is for? Otherwise, how are other people going to test it?

jonasmalacofilho, 2022-08-03:
> Okay, but isn't that what experimental support is for?

Experimental drivers should still have been tested by someone; the experimental "tag" is meant for the things the developer probably couldn't account for, like different firmware versions, all possible operating systems, weird use cases etc.  Plus plain simple bugs in the initial implementation...

> Otherwise, how are other people going to test it?

I agree we want people to test but, more importantly, we need their feedback.  Shipping a big untested feature is not a good way to get it:

- if it doesn't work, it will take a surprisingly long time for one of the affected users to report the severe bug: the change first has to wait for our next release (October), then the release has to trickle down to users with this device, and then someone has to report the bug (but many people don't do that);
- if it does work, we'll never hear about it...  after one or two releases (another 3-6 months) we'll just assume it does.

So by not doing that, at least for now, I'm letting people pop in here to ask that very question and, hopefully, give us some immediate feedback afterwards.

closed 2022-09-22 by jonasmalacofilho

johandc, 2022-10-22:
Thanks for the merge :-)

commit 3e7cf3b1 (2022-09-22): smart_device: register new Fan & RGB Controller PID

While RGB control hasn't been tested, it is assumed to behave as in the
previous variants.  This is supported by a similarly timed new variant
of the Kraken X3 being fully backward compatible (see #503).

Fixes: #485
Suggested-by: @johandc