liquidctl/liquidctl #487: Corsair AX Series AX1600i 1600W ATX 2.4

fabolous005, 2022-08-02:
Device type: PSU
Product page: https://www.corsair.com/us/en/Categories/Products/Power-Supply-Units/axi-series-config/p/CP-9020087-NA 
First-party software: iCue

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors: yes
 - monitoring of fan or pump speeds or duty cycles: tes
 - monitoring of voltages, currents or power: yes
 - configurable fan or pump speeds: yes
 - configurable voltages or current/power limits: yes
 - configurable lighting of embedded LEDs: yes
 - configurable lighting of accessories like RGB fans or LED strips: 

The device is physically connected to the host system using:

- USB: 
- PCI-E: 
- onboard the motherboard: micro-usb type b
- other: 

The device uses which protocols to communicate with the host system:

- HID: 
- USB:  micro-usb type b
- I²C: 
- SMBus: 
- other: 

Additional information about the device:

Finally, I can help with:

- testing changes on Linux: 
- testing changes on Windows: 
- testing changes on Mac OS: 
- attempting to capture USB traffic: 
- attempting to capture I²C/SMBus traffic: 
- analyzing traffic data: 
- documenting the protocol: 
- implementing the driver: 

assuming documentation and/or assistance will be provided.

Hey guys,

came across your project while searching for a new PSU.
I saw that there is support for the corsair 1200i but it's more expensive and "less powerful then the 1600i".
Are there any investigations on supporting this PSU?

I'am not really experienced in python (actually 5%), but I would buy this PSU if someone could help me to bring support to this PSU.

Great job so far

regards

Brummby, 2022-08-02:
G'day All,
I have one of these PSU's in my current computer if you need anything tried.

I run both Linux and Windows however, I only use liquidctl in Linux.

Note: I am rather new to python / programming.

jonasmalacofilho, 2022-08-03:
> Are there any investigations on supporting this PSU?

There was some initial work on supporting the AX1500i, and we think it and the AX1600i are _similar._  You can take a look at/follow issue #332.

(Unlike other device families, there are also some things that appear to be different between various AXi models).

fabolous005, 2022-08-03:
So how soon do you think there will be proper support for this PSU-(Family)?

jonasmalacofilho, 2022-08-03:
The best I can say is _not_ soon: I'm not currently working on the AX1500i, and IIRC we haven't even seen a traffic capture from a AX1600i yet (so its protocol, which we believe to be _similar,_ but not identical, to the AX1500i, could in fact be totally different).

Jon0, 2022-08-15:
I have a basic ax1600i driver working, enough to read temp and set the fan speed

https://github.com/Jon0/ax1600i

The protocol seems quite similar to the cpsumon code (https://github.com/ka87/cpsumon)
Main difference so far is an extra 0x12 byte is written after the main messages.

fabolous005, 2022-08-15:
Great work!!!
I like things being written in rust.
Thank you!

jonasmalacofilho, 2022-08-15:
@Jon0,

> https://github.com/Jon0/ax1600i

Cool!

> Main difference so far is an extra 0x12 byte is written after the main messages.

That's good to know, thanks!

gilvbp, 2022-11-10:
@jonasmalacofilho hi there, any news about the ax1600i support? thanks

fabolous005, 2022-11-11:
Check out @Jon0's [driver](https://github.com/Jon0/ax1600i). It is written in rust.

bra1nDeaD, 2025-03-23:
I've done a USB capture with wireshark and filtered the result to the AX1600i only. I hope this helps.

https://github.com/bra1nDeaD/ax1600i-usb-dump