liquidctl/liquidctl #488: liquidctl status --json filling up the RAM if liquictl initialize didn't run on startup

frisky5, 2022-08-04:
installed liquidctl with `python3 -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl`
everything works fine without issues, but once i reboot i found my RAM usage is increasing by time, looked at running processes and found more than 200 liquidctl processes with the following command `liquidctl status --json`, i opened a terminal and ran `liquidctl initialize` and all these processes starting to finish, so i did one more reboot and ran the `liquidctl status --json` command and it runs forever as long as liquictl is not initialized, so i added a cron job to run on boot and run liquidctl initialize.

I am not sure if I am doing something wrong or this is an issue, but just wanted to let you know about it.

jonasmalacofilho, 2022-08-04:
Hi!

Thanks for letting me know about this, that's quite unexpected (and unacceptable) behavior.

Can you run one of these faulty `status` commands but with an additional `--debug` option, and show me the output?

```
$ liquidctl status --json --debug
```

Now I suspect this has something to do with a device that misbehaves when not initialized, and initializing all devices after every boot (or resume from a suspended state) is indeed always recommended.  Does the documentation for the device(s) you have need to be fixed?

Regardless, this constant spawning of child processes shouldn't happen, even in this scenario; in fact, we don't do anything like it (when working correctly), so this bug is quite mystifying, at least for now.

Hopefully the debug output will help...

---

Another thing that might help, especially if the debug output doesn't, is to `strace` one of these stuck processes.  Or maybe two, the top most parent and one of the unexpected children.

jonasmalacofilho, 2022-08-04:
Note that bad error handling in a script and/or program that invokes `liquidctl status --json` could explain the endless spawning of liquidctl instances...  and the rest, i.e. getting stuck, could easily be the result of a device that simply doesn't respond when not yet initialized.

But even then we might look into adding some timeouts, or checking why any existing ones aren't being helpful in this scenario.

jonasmalacofilho, 2022-08-09:
As a complement or alternative to timeouts on individual operations or device reads/writes, I think we might want to add a global timeout for the entire CLI execution, configurable but defaulting to a reasonable value.

The reason for a default value, instead of being opt-in, is that existing tools (e.g. [freon](https://github.com/UshakovVasilii/gnome-shell-extension-freon/)) might be inadvertently set up to enable liquidctl support, without the user setting up proper initialization of the liquidctl devices.  (And it's not reasonable for some of these tools to take over the initialization, because of the need for user-supplied flags and robustness).  It would be complicated for these tools use a new option that's not necessarily supported by the version of liquidctl running on the system.

jonasmalacofilho, 2022-08-09:
I'm not sure how to implement a (real) global timeout without (directly or indirectly) running `cli.main` on a separate thread.[^1]  And that, unfortunately, has a **severe** (and unacceptable) performance impact...

Baseline (no timeout):

```
$ perf stat --repeat=3 python -m liquidctl status
[ ... ]
           0.37641 +- 0.00818 seconds time elapsed  ( +-  2.17% )
```

Timeout implemented with `async.to_thread` and `asyncio.wait`:

<details><summary>Patch</summary>

```diff
diff --git a/liquidctl/__main__.py b/liquidctl/__main__.py
index 6ba533045cd6..53622b446eb9 100644
--- a/liquidctl/__main__.py
+++ b/liquidctl/__main__.py
@@ -1,7 +1,21 @@
 # uses the psf/black style
 
+import asyncio
 import sys
-from liquidctl.cli import main
+import os
+from liquidctl import cli
+
+
+async def main():
+    tasks = {asyncio.create_task(asyncio.to_thread(cli.main))}
+    done, _ = await asyncio.wait(tasks, timeout=60)
+    try:
+        return done.pop().result()
+    except KeyError:
+        print("liquidctl.cli.main() timed out")
+        # force immediate exit, sys.exit would stall us anyway
+        os._exit(2)
+
 
 if __name__ == "__main__":
-    sys.exit(main())
+    sys.exit(asyncio.run(main()))
```

</details>

```
$ perf stat --repeat=3 python -m liquidctl status
[ ... ]
            1.4063 +- 0.0175 seconds time elapsed  ( +-  1.25% )
```

Timeout implemented with `threading.Thread.join`:

<details><summary>Patch</summary>

```diff
diff --git a/liquidctl/__main__.py b/liquidctl/__main__.py
index 6ba533045cd6..6fe8b0da91d7 100644
--- a/liquidctl/__main__.py
+++ b/liquidctl/__main__.py
@@ -1,7 +1,27 @@
 # uses the psf/black style
 
 import sys
-from liquidctl.cli import main
+import os
+from liquidctl import cli
+import threading
+
+
+exit_code = 0  # lazy, but this is just an experiment
+
+
+def main():
+    global exit_code
+    exit_code = cli.main()
+
 
 if __name__ == "__main__":
-    sys.exit(main())
+    t = threading.Thread(target=main)
+    t.start()
+    t.join(timeout=60)
+
+    if t.is_alive():
+        print("liquidctl.cli.main() timed out")
+        # force immediate exit, sys.exit would stall us anyway
+        os._exit(2)
+    else:
+        sys.exit(exit_code)
```

</details>

```
$ perf stat --repeat=3 python -m liquidctl status
[ ... ]
            1.3921 +- 0.0158 seconds time elapsed  ( +-  1.13% )
```

[^1]: And, because `sys.exit`, `SystemExit` and even `os._exit` only terminate the process if called from the main thread, it's not possible to shift the overhead to the timer/watchdog (by running _it_ on the separate thread).

commit f0aa68c0 (2022-09-23): usb: add default timeouts to all IO methods that support them

Fixes: #488
Related: #497, partially: hid.device.write doesn't support a timeout
Related: #505
Related: #506
closed 2022-09-23 by jonasmalacofilho