liquidctl/liquidctl #497: Hangs on FreeBSD

cartesius68, 2022-08-17:
Running commands hangs for many minutes without producing output.

`sudo liquidctl --match kraken set fan speed 80` works always

`sudo liquidctl --match kraken status` works after intialize just once

```
sudo liquidctl --match kraken initialize all
NZXT Kraken Z (Z53, Z63 or Z73)
└── Firmware version    5.7.0  

(env) jb@chrostik ~> sudo liquidctl --match kraken --debug status
[DEBUG] (cli) (main): liquidctl: 1.11.0.dev38+g9f352cb
[DEBUG] (cli) (main): platform: FreeBSD-13.1-RELEASE-p1-amd64-64bit-ELF
[DEBUG] (cli) (_log_requirements): python: 3.9.13 (main, Jul  3 2022, 01:17:15) 
[Clang 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: version n/a (smbus)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1e71:3008 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken Z (Z53, Z63 or Z73)
(env) jb@chrostik ~> sudo liquidctl --match kraken --debug status
[DEBUG] (cli) (main): liquidctl: 1.11.0.dev38+g9f352cb
[DEBUG] (cli) (main): platform: FreeBSD-13.1-RELEASE-p1-amd64-64bit-ELF
[DEBUG] (cli) (_log_requirements): python: 3.9.13 (main, Jul  3 2022, 01:17:15) 
[Clang 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: version n/a (smbus)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1e71:3008 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken Z (Z53, Z63 or Z73)
[DEBUG] (usb) (find_devices): HID device: 30be:0100 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 2516:001a (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 2516:001a (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 2516:001a
[DEBUG] (usb) (find_devices): USB device: 30be:0100
[DEBUG] (usb) (find_devices): USB device: 1e71:3008
[DEBUG] (usb) (find_devices): USB device: 1532:0078
[DEBUG] (usb) (find_devices): USB device: 0b05:18f3
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 0000:0000
[DEBUG] (usb) (find_devices): USB device: 0000:0000
[DEBUG] (usb) (find_devices): USB device: 0000:0000
[DEBUG] (cli) (main): device: NZXT Kraken Z (Z53, Z63 or Z73)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x74 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```

and hangs.

---

Additional information about the system:

- FreeBSD 13.1 RELEASE-p1
- Ryzen 5900X
- `Result #0: NZXT Kraken Z (Z53, Z63 or Z73)`

jonasmalacofilho, 2022-08-17:
Can you also reproduce this with the changes in PR #479?

That PR reworks a lot of the Kraken Z driver, so it would be a better starting point to fix this.  Shady (the PR author) might also be able to help us: I see that the `0x7401...` write also appears in their LCD screen code.

cartesius68, 2022-08-17:
> Can you also reproduce this with the changes in PR #479?

Unfortunately hangs as well. Some new LCD attributes displayed in the first successful run though but the second run for `status` hangs as well.

cartesius68, 2022-08-17:
Now getting non-sense data. Would be happy to help you with debugging that though.

```
sudo liquidctl --match kraken status
NZXT Kraken Z (Z53, Z63 or Z73)
├── Liquid temperature    2.0  °C
├── Pump speed              0  rpm
├── Pump duty               0  %
├── Fan speed               0  rpm
└── Fan duty                0  %
```

jonasmalacofilho, 2022-08-17:
> Would be happy to help you with debugging that though.

I think what's necessary here is to capture the traffic with CAM for sometime and when _only_ monitoring is enabled.  The best way to achieve that might be to set the fan, pump and screen set to fixed values[/image], and hope CAM doesn't periodically send update commands for those anyway (which, unfortunately, has happened before).

jonasmalacofilho, 2022-08-17:
@ShadyNawara,

Since you've recently figured out a lot more about the Kraken Z protocol, do you have any idea what we're missing here?

cartesius68, 2022-08-17:
@jonasmalacofilho I'm not sure if it helps, but this specific machine is dual boot FreeBSD / Linux and under Linux everything works smoothly. I mean `status`, `set fan`, `set pump`. That's all I use currently.

ShadyNawara, 2022-08-17:
if its a FreeBSD specific issue then it might be a problem with their HID driver. I will try to take a look and if thats the case I will test with the libusb driver and see if that fixes it

cartesius68, 2022-08-17:
Yes, with that very same NZXT hardware I've been successfully using `liquidctl` for months on Arch Linux. Great work guys anyway!

jonasmalacofilho, 2022-08-18:
> if its a FreeBSD specific issue then it might be a problem with their HID driver. I will try to take a look and if thats the case I will test with the libusb driver and see if that fixes it

IIRC, HIDAPI on FreeBSD uses a LibUSB-based implementation.  So it's more likely an issue there, or with an assumption in our code.  In the latter case, it has usually been related to report IDs and/or lengths.

This might be easier to test/compare on Linux, since there HIDAPI has both a HIDRAW and a LibUSB backend.  You just need to make sure that cython-hidapi has been compiled _with_ the LibUSB-backend (which isn't the case with ArchLinux's package, and possibly some other distros), and disable/bypass:

https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/usb.py#L66

---

Could either of you show me the HID descriptor of this device's HID interface?  There are many ways to do it, but I usually use `usbhid-dump` on Linux.

cartesius68, 2022-08-19:
@jonasmalacofilho  OK, I'll try later today.

Btw, what's interesting, that on FreeBSD the `status` command works perfectly and produces sane data for the very first time it's called. So it's not completely broken. But every next call hangs.

cartesius68, 2022-08-19:
@jonasmalacofilho 

```
sudo /usr/home/jb/env/bin/liquidctl list --verbose
Device #0: ASUS Aura LED Controller (experimental)
├── Vendor ID: 0x0b05
├── Product ID: 0x18f3
├── Release number: 0x0100
├── Serial number: 9876543210
├── Bus: hid
├── Address: 1-4:1.2
└── Driver: AuraLed

Device #1: NZXT Kraken Z (Z53, Z63 or Z73)
├── Vendor ID: 0x1e71
├── Product ID: 0x3008
├── Release number: 0x0507
├── Serial number: 205A316D3456
├── Bus: hid
├── Address: 1-5:1.1
└── Driver: KrakenZ3
```

cartesius68, 2022-08-19:
@jonasmalacofilho 

```
$ sudo usbconfig
ugen0.1: <AMD XHCI root HUB> at usbus0, cfg=0 md=HOST spd=SUPER (5.0Gbps) pwr=SAVE (0mA)
ugen2.1: <AMD XHCI root HUB> at usbus2, cfg=0 md=HOST spd=SUPER (5.0Gbps) pwr=SAVE (0mA)
ugen1.1: <AMD XHCI root HUB> at usbus1, cfg=0 md=HOST spd=SUPER (5.0Gbps) pwr=SAVE (0mA)
ugen0.2: <vendor 0x05e3 USB2.0 Hub> at usbus0, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=SAVE (100mA)
ugen1.2: <AsusTek Computer Inc. AURA LED Controller> at usbus1, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (16mA)
ugen2.2: <vendor 0x1532 product 0x0078> at usbus2, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)
ugen1.3: <NZXT Inc. NZXT KrakenZ Device> at usbus1, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=ON (100mA)
ugen2.3: <Schiit Audio Im Fulla Schiit> at usbus2, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=ON (0mA)
ugen2.4: <CM Storm Keyboard -- QuickFire XT> at usbus2, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (100mA)

$ sudo usbhid-dump -a 1:3
001:003:001:DESCRIPTOR         1660889802.021468
 06 00 FF 09 01 A1 01 85 10 09 01 15 00 26 FF 00
 75 08 95 3F B1 82 85 10 09 01 91 82 85 12 09 02
 15 00 26 FF 00 75 08 95 3F B1 82 85 12 09 02 91
 82 85 20 09 03 15 00 26 FF 00 75 08 95 3F B1 82
 85 20 09 03 91 82 85 22 09 04 15 00 26 FF 00 75
 08 95 3F B1 82 85 22 09 04 91 82 85 24 09 05 15
 00 26 FF 00 75 08 95 3F B1 82 85 24 09 05 91 82
 85 26 09 06 15 00 26 FF 00 75 08 95 3F B1 82 85
 26 09 06 91 82 85 28 09 07 15 00 26 FF 00 75 08
 95 3F B1 82 85 28 09 07 91 82 85 F2 09 08 15 00
 26 FF 00 75 08 95 3F B1 82 85 F2 09 08 91 82 85
 FE 09 09 15 00 26 FF 00 75 08 95 3F B1 82 85 FE
 09 09 91 82 85 70 09 0A 15 00 26 FF 00 75 08 95
 3F B1 82 85 70 09 0A 91 82 85 72 09 0B 15 00 26
 FF 00 75 08 95 3F B1 82 85 72 09 0B 91 82 85 73
 09 0C 15 00 26 FF 00 75 08 95 3F B1 82 85 73 09
 0C 91 82 85 74 09 0D 15 00 26 FF 00 75 08 95 3F
 B1 82 85 74 09 0D 91 82 85 2A 09 0E 15 00 26 FF
 00 75 08 95 3F B1 82 85 2A 09 0E 91 82 85 E2 09
 0F 15 00 26 FF 00 75 08 95 3F B1 82 85 E2 09 0F
 91 82 85 30 09 10 15 00 26 FF 00 75 08 95 3F B1
 82 85 30 09 10 91 82 85 32 09 11 15 00 26 FF 00
 75 08 95 3F B1 82 85 32 09 11 91 82 85 34 09 12
 15 00 26 FF 00 75 08 95 3F B1 82 85 34 09 12 91
 82 85 36 09 13 15 00 26 FF 00 75 08 95 3F B1 82
 85 36 09 13 91 82 85 38 09 14 15 00 26 FF 00 75
 08 95 3F B1 82 85 38 09 14 91 82 85 11 09 15 75
 08 95 3F 81 82 85 13 09 16 75 08 95 3F 81 82 85
 FF 09 17 75 08 95 3F 81 82 85 71 09 18 75 08 95
 3F 81 82 85 75 09 19 75 08 95 3F 81 82 85 21 09
 1A 75 08 95 3F 81 82 85 23 09 1B 75 08 95 3F 81
 82 85 25 09 1C 75 08 95 3F 81 82 85 27 09 1D 75
 08 95 3F 81 82 85 29 09 1E 75 08 95 3F 81 82 85
 2B 09 1F 75 08 95 3F 81 82 85 31 09 20 75 08 95
 3F 81 82 85 33 09 21 75 08 95 3F 81 82 85 35 09
 22 75 08 95 3F 81 82 85 37 09 23 75 08 95 3F 81
 82 85 39 09 24 75 08 95 3F 81 82 85 E3 09 25 75
 08 95 3F 81 82 C0
```

jonasmalacofilho, 2022-08-19:
Thanks!

Decoded (summarized):[^1]

```txt
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
[ ... ]
0x85, 0x74,        //   Report ID (116)
0x09, 0x0D,        //   Usage (0x0D)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x74,        //   Report ID (116)
0x09, 0x0D,        //   Usage (0x0D)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
[ ... ]
0x85, 0x75,        //   Report ID (117)
0x09, 0x19,        //   Usage (0x19)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
[ ... ]
0xC0,              // End Collection

// 598 bytes
```

<details><summary>In full (click to expand):</summary>

```txt
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x10,        //   Report ID (16)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x10,        //   Report ID (16)
0x09, 0x01,        //   Usage (0x01)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x12,        //   Report ID (18)
0x09, 0x02,        //   Usage (0x02)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x12,        //   Report ID (18)
0x09, 0x02,        //   Usage (0x02)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x20,        //   Report ID (32)
0x09, 0x03,        //   Usage (0x03)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x20,        //   Report ID (32)
0x09, 0x03,        //   Usage (0x03)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x22,        //   Report ID (34)
0x09, 0x04,        //   Usage (0x04)
0x15, 0x00,        //   Logical Minimum (0)
[... 240 lines cut ...]
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x39,        //   Report ID (57)
0x09, 0x24,        //   Usage (0x24)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0xE3,        //   Report ID (-29)
0x09, 0x25,        //   Usage (0x25)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 598 bytes
```
</details>

Looking at it, our code, and HIDAPI's LibUSB backend, it still isn't obvious to me what's wrong.  Based on your description of the issue, I suspected a subtle bug with the report lengths, but everything seems to make sense: for both reads and writes we expect 64 bytes, which is the report size*count + the report ID prefix; and (in this case, where report IDs are present) HIDAPI's LibUSB backend just forwards all this data from/to a LibUSB interrupt transfer.  (But please feel free to review this for yourselves, as I might have missed something).

Can you try setting `LIBUSB_DEBUG=4` (environment variable) and pasting the output from the first and second runs here?

And can you capture the USB traffic with the device (for, again, the first and second runs)?

[^1]: https://eleccelerator.com/usbdescreqparser/

ShadyNawara, 2022-08-22:
I just managed to get freebsd (GhostBSD) installed inside a VM after 3 days of trying and 3 different distros. Anyways, I am not experiencing the issues outlined here at all. I am able to run the status command as well as my lcd gif commands repeatedly without issues. Given my experience installing freebsd which was honestly not pleasant I would suggest reinstalling, trying a different distro and/or building hidapi and python-hidapi from source.

cartesius68, 2022-08-22:
@ShadyNawara  More on a funny note, I consider installing FreeBSD nowadays the most pleasant experience including Linux distros like Arch or Ubuntu :-) ...when using the auto partitioner it's just 5 enters or such. Anyway, will try the GhostBSD, and I've just reinstalled the FreeBSD 13.1. Everything clean & up-to-date and this is the second run:

```
sudo liquidctl --match kraken status
NZXT Kraken Z (Z53, Z63 or Z73)
├── Liquid temperature    1.0  °C
├── Pump speed              0  rpm
├── Pump duty               0  %
├── Fan speed               0  rpm
└── Fan duty                0  %
```

cartesius68, 2022-08-29:
@ShadyNawara @jonasmalacofilho 

This (of course) works each time on FreeBSD 13.1 **bare metal**:

```
sudo usbconfig -d /dev/ugen1.3 reset; sleep 1; sudo liquidctl initialize all; sleep 1; sudo liquidctl --match kraken status
```

Anyway:

## First run
```
sudo env LIBUSB_DEBUG=4 liquidctl status
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
ERROR: unexpected error with AsusTek Aura LED Controller (experimental): TypeError('list.append() takes exactly one argument (3 given)')
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
NZXT Kraken Z (Z53, Z63 or Z73)
├── Liquid temperature    34.4  °C
├── Pump speed            2764  rpm
├── Pump duty              100  %
├── Fan speed             1452  rpm
└── Fan duty                65  %
```

## Second run
```
sudo env LIBUSB_DEBUG=4 liquidctl status
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
ERROR: unexpected error with AsusTek Aura LED Controller (experimental): IndexError('list index out of range')
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
```
(this hangs).

jonasmalacofilho, 2022-08-30:
Is that the _complete_ stderr+stdout output?  If it is, wow, that's really different than what I'm used to seeing from libusb-1.0.

jonasmalacofilho, 2022-08-30:
I would also like to debug the issue with the Aura LED controller.  Can you post the `--debug` output for both first and second runs (since the exceptions are slightly different)?

cartesius68, 2022-08-31:
@jonasmalacofilho 
Just to be sure, another test:

## Packages
```
hidapi-0.12.0                  Library to access USB HID-class devices
py39-hidapi-0.11.0.p2          Cython interface to comms/hidapi
py39-pyusb-1.0.2               Python wrapper around libusb
```
yes, running via the system libraries.

## First run
```
sudo env LIBUSB_DEBUG=4 liquidctl status
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
AsusTek Aura LED Controller (experimental)
├── ARGB channels: 2      
└──  RGB channels: 1      

LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
NZXT Kraken Z (Z53, Z63 or Z73)
├── Liquid temperature    29.5  °C
├── Pump speed            2643  rpm
├── Pump duty               95  %
├── Fan speed              898  rpm
└── Fan duty                38  %
```

## Second run (hangs)
```
> sudo env LIBUSB_DEBUG=4 liquidctl status
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
ERROR: unexpected error with AsusTek Aura LED Controller (experimental): IndexError('list index out of range')
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
```

**That's all I see in the console.**

jonasmalacofilho, 2022-08-31:
> yes, running via the system libraries.

What is the libusb version?

---

> and full debug runs:

I wanted to see the (unrelated) issue with the Aura LED controller, so can you please provide these without filtering that device off?

cartesius68, 2022-08-31:
Sorry for misreading that request.

Regarding `libusb`, I don't have
```
linux_libusb-13.1.0            Linux-compatibility LibUSB
```
installed at all. Isn't that a problem? Does `liquidctl` somehow picks the appropriate backend?

Here's the full debug.
```
> sudo env LIBUSB_DEBUG=4 liquidctl status --debug
[DEBUG] (cli) (main): liquidctl: 1.10.0
[DEBUG] (cli) (main): platform: FreeBSD-13.1-RELEASE-p1-amd64-64bit-ELF
[DEBUG] (cli) (_log_requirements): python: 3.9.13 (main, Aug 11 2022, 01:13:34) 
[Clang 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.0.2
[DEBUG] (cli) (_log_requirements): smbus: version n/a (smbus)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
[DEBUG] (usb) (find_devices): HID device: 1c4f:0043 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1c4f:0043 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0078 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (probe): AuraLed identified: AsusTek Aura LED Controller (experimental)
[DEBUG] (usb) (find_devices): HID device: 30be:0100 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1e71:3008 (usage_page=0x0000 usage=0x0000)
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken Z (Z53, Z63 or Z73)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 0000:0000
[DEBUG] (usb) (find_devices): USB device: 0000:0000
[DEBUG] (usb) (find_devices): USB device: 0000:0000
[DEBUG] (usb) (find_devices): USB device: 1c4f:0043
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 1532:0078
[DEBUG] (usb) (find_devices): USB device: 0b05:18f3
[DEBUG] (usb) (find_devices): USB device: 30be:0100
[DEBUG] (usb) (find_devices): USB device: 1e71:3008
[DEBUG] (usb) (find_devices): USB device: 13fe:5500
[DEBUG] (cli) (main): device: AsusTek Aura LED Controller (experimental)
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
[DEBUG] (usb) (write): writting report 0xec with 64 bytes: b0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 32 bytes: ec:30:00:00:1e:9f:02:01:00:00:78:3c:00:01:00:00:78:3c:00:00:00:00:00:00:00:00:00:00:00:00:00:08
AsusTek Aura LED Controller (experimental)
├── ARGB channels: 2                                         
├──  RGB channels: 1                                         
├── Device Config: 1     0x1e, 0x9f, 0x02, 0x01, 0x00, 0x00  
├── Device Config: 2     0x78, 0x3c, 0x00, 0x01, 0x00, 0x00  
├── Device Config: 3     0x78, 0x3c, 0x00, 0x00, 0x00, 0x00  
├── Device Config: 4     0x00, 0x00, 0x00, 0x00, 0x00, 0x00  
├── Device Config: 5                 0x00, 0x00, 0x00, 0x08  
├── Device Config: 6                                         
├── Device Config: 7                                         
├── Device Config: 8                                         
├── Device Config: 9                                         
└── Device Config: 10                                        

[DEBUG] (cli) (main): device: NZXT Kraken Z (Z53, Z63 or Z73)
LIBUSB_DEBUG: bInterfaceNumber(2) is not sequential(1)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x74 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: ff:01:22:00:23:00:02:50:56:34:4b:31:37:20:72:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT Kraken Z (Z53, Z63 or Z73)
├── Liquid temperature    2.0  °C
├── Pump speed              0  rpm
├── Pump duty               0  %
├── Fan speed               0  rpm
└── Fan duty                0  %
```

@jonasmalacofilho

jonasmalacofilho, 2022-08-31:
> Regarding `libusb`, I **don't have**

That's odd, as both PyUSB and [cython-]hidapi require it (at least on FreeBSD, AFAIK).  In the latter case, it _could_ have been statically linked, but that's rare in Linux/BSD distributions.  And PyUSB is certainly finding _something..._ 

I can try to take a look a look at what might [cython-]hidapi use on FreeBSD, but it will likely take me a lot of time just to find the right places to look into.  So, in the meantime, I would appreciate if you (or someone else) still tried to figure out what specific LibUSB implementation and version is being used.

Note: PyUSB isn't important here, and it might even be using a different backend than [cython-]hidapi.

---

> Here's the full debug.

Thanks again!  Unfortunately that output doesn't manifest any of the two issues you showed before with the Aura LED Controller.  If they manifest again, I would still appreciate the information.

cartesius68, 2022-09-01:
`libusb` is inbuilt, source in `/usr/src/lib/libusb`. When I run:

```
>>> usb.version_info
(1, 0, 2)
```

but `man 3` page shows:

```
LIBUSB VERSION 0.1 COMPATIBILITY
     The library is also compliant with LibUSB version 0.1.12.
```

cartesius68, 2022-09-01:
> > Regarding `libusb`, I **don't have**

`libusb` is inbuilt, source in `/usr/src/lib/libusb`. When I run:

```
>>> usb.version_info
(1, 0, 2)
```

but `man 3` page shows:

```
LIBUSB VERSION 0.1 COMPATIBILITY
     The library is also compliant with LibUSB version 0.1.12.
```

cartesius68, 2022-09-01:
Added source codes for `libusb` on my machine.
[libusb-freebsd-13.1-RELEASE.zip](https://github.com/liquidctl/liquidctl/files/9468694/libusb-freebsd-13.1-RELEASE.zip)

jonasmalacofilho, 2022-09-01:
> `libusb` is inbuilt, source in `/usr/src/lib/libusb/`

Do you mean it's built into hidapi, cython-hidapi, or both?

> When I run python I get:

That's the PyUSB version, which isn't very relevant here.  And it has nothing to do with the backend being used by PyUSB, which may or _may not_ match the same library used or built into hidapi and cython-hidapi.

---

Regarding cython-hidapi, I think you need to figure out where the compiled cython-hidapi extension is, then `ldd` it.

Then do the same for the system hidapi.

swapfile-dev, 2022-09-01:
> Do you known where can I find the built "recipe" used by FreeBSD for `hidapi-0.12.0` and `py39-hidapi-0.11.0.p2`? Specifically, I want to see how FreeBSD built those packages, which options were used, etc.

I shall collect that info.