liquidctl/liquidctl #498: Corsair iCUE H150i ELITE RGB

sdns575, 2022-08-23:
Device type: AIO liquid cooler
Product page: https://www.corsair.com/it/it/Categorie/Prodotti/Sistemi-di-raffreddamento-a-liquido/Sistema-di-raffreddamento-a-liquido-per-CPU-iCUE-RGB-ELITE/p/CW-9060060-WW
First-party software: Corsair iCue

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors: yes
 - monitoring of fan or pump speeds or duty cycles:  yes
 - configurable fan or pump speeds: yes 
 - configurable lighting of embedded LEDs: yes  

The device is physically connected to the host system using:

- onboard the motherboard:  USB header

The device uses which protocols to communicate with the host system:

- USB: Type C

Additional information about the device:

Finally, I can help with:

- testing changes on Linux: 
- attempting to capture USB traffic:

jonasmalacofilho, 2022-08-23:
We already have an open issue for that device: #218.

Closing this one to keep the discussion in a single place.

jonasmalacofilho, 2022-08-23:
Oh, sorry.

jonasmalacofilho, 2022-08-23:
Can you post the output from `lsusb -v` for that device?

sdns575, 2022-08-24:
Bus 001 Device 004: ID 1b1c:0c37 Corsair H150iELITE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c37 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H150iELITE
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

jonasmalacofilho, 2022-08-25:
Thanks!  These USB descriptors suggest a similarity with the Platinum/Pro XT coolers.

Can you [capture some of the traffic](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md) between iCue and the device?

And can you also dump the HID descriptors with `usbhid-dump`?

sdns575, 2022-08-25:
This is the output from usbhid-dump -m 1b1c:0c37:
001:004:000:DESCRIPTOR         1661437902.164687
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

jonasmalacofilho, 2022-08-26:
Thanks!

Decoded:[^1]

```
0x01, 0x04,        // Unknown (bTag: 0x00, bType: 0x00)
0x00,              // Unknown (bTag: 0x00, bType: 0x00)
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0x09, 0x01,        //   Usage (0x01)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x40,        //   Report Count (64)
0x09, 0x01,        //   Usage (0x01)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 30 bytes
```

sdns575, 2022-08-26:
I have sent you via email with attached USB data dump. 
Have you received it?

jonasmalacofilho, 2022-08-26:
I can't find it.  Can you try sending it again?

I'm not sure what you used before, but my preferred address is `jonas@protocubo.io`.

sdns575, 2022-08-26:
Thank you for the address. I sent a copy of USB dump using wireshark.

Hope that helps.

jonasmalacofilho, 2022-08-27:
Got the email, will take a look when I find some time for it.

Thanks!

sdns575, 2022-08-28:
Thank you Jason, 
If  you need some test with liquidctl for this device I will help you.

Best regards

sdns575, 2022-09-26:
Hi Jason,
there are any news? 
I'm going crazy changing pump speed under load using VM.
How I can help you?

jonasmalacofilho, 2022-09-27:
No news from me so far, unfortunately.

> I'm going crazy changing pump speed under load using VM.

I sympathize.

> How I can help you?

Can you try to take a look at the captured traffic and see if it seems to match the PRO/Platinum protocol implemented in [`liquidctl.driver.hydro_platinum`](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/hydro_platinum.py)?

sdns575, 2022-09-27:
Hi Jason,
I tried reading hydro_platinum driver but I can't find any match with data in wireshark dump. Do you know if the pump speed state are equal for all Corsair AIOs? I installed liquidctl with pip and will try to add the model:

(0x1b1c, 0x0c37, None, 'Corsair iCUE H150i ELITE',
            {'fan_count': 3, 'fan_leds': 16}),

and try to see if at least pump could be managed.

On iCUE pump speed are: Silence, Balanced, Extreme, Variable. I expect that for the first three is the same as the previous cooler.

sdns575, 2022-09-27:
Hi Jonas, 
sorry for your name. My bad.

I tried adding only the model. Running liquidctl status I get:

Corsair iCUE H150i ELITE
├── Liquid temperature    27.5  °C
├── Fan 1 speed              0  rpm
├── Fan 1 duty              47  %
├── Fan 2 speed              0  rpm
├── Fan 2 duty              47  %
├── Fan 3 speed              0  rpm
├── Fan 3 duty              47  %
├── Pump speed            2042  rpm
└── Pump duty               55  %

Running liquidctl -n 1 --initialize --pump-mode extreme and running again status I get:

Corsair iCUE H150i ELITE
├── Liquid temperature    27.8  °C
├── Fan 1 speed              0  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed              0  rpm
├── Fan 2 duty             100  %
├── Fan 3 speed              0  rpm
├── Fan 3 duty             100  %
├── Pump speed            2991  rpm
└── Pump duty              100  %

Seems that the pump command works without any problem.

Running liquidctl -n 1 set led color fixed ff0000 it changes the color to red for all led but I got this error.
ERROR: unexpected OS error with Corsair iCUE H150i ELITE: OSError('Could not write to device')

I tried also attaching one fan to check if I can manage fans from liquidctl and running "liquidctl -n 1 set fan3 speed 80" and running liquidctl status I get:

Corsair iCUE H150i ELITE
├── Liquid temperature    27.1  °C
├── Fan 1 speed              0  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed              0  rpm
├── Fan 2 duty             100  %
├── Fan 3 speed           1575  rpm
├── Fan 3 duty              80  %
├── Pump speed            2990  rpm
└── Pump duty              100  %

I tried to check if zero RPM works but running liquidctl -n 1 set fan3 speed 0 it slow down up to 20% of duty cycle.

Hope that helps.

jonasmalacofilho, 2022-10-03:
> Hi Jonas, sorry for your name. My bad.

It's ok, and thanks.

> I tried adding only the model. Running liquidctl status I get:
> 
> [...]
> 
> Seems that the pump command works without any problem.

Good news indeed.

> Running liquidctl -n 1 set led color fixed ff0000 it changes the color to red for all led but I got this error. ERROR: unexpected OS error with Corsair iCUE H150i ELITE: OSError('Could not write to device')

Can you show us the complete `--debug` output?

> I tried also attaching one fan to check if I can manage fans from liquidctl and running "liquidctl -n 1 set fan3 speed 80" and running liquidctl status I get:
> 
> [...[

Yeah, that seems to work as well, which is great.
 
> I tried to check if zero RPM works but running liquidctl -n 1 set fan3 speed 0 it slow down up to 20% of duty cycle.

That suggests that zero RPM mode might require a different command, as (from what I can see) liquidctl would have passed the 0% duty value to the device unchanged (you should be able to check that in the output with `--verbose`).

Can you achieve zero RPM with iCue?  If so, can you try capturing that specific traffic?

st1vms, 2023-01-14:
I may want to gather opinion on this branch i proposed [branch #558](https://github.com/liquidctl/liquidctl/pull/558), it does include a `lsusb` wrapper, to use with `hydro_platinum` driver, which should auto fill the VID:PID in the `_MATCHES` parameter. It does lack auto udev-rule generation tho...

PR #559 (2023-05-11): Support for Corsair iCUE H150i RGB ELITE
Add support for the Corsair iCUE H150i RGB ELITE, based on #557.

While this has been attempted in the past, [unsuccessfully](https://github.com/liquidctl/liquidctl/issues/498#issuecomment-1259804020), we never knew exactly why that happened. It's possible that in the meantime the issue has been solved (by other changes to the driver). It is also possible that at that time the properties weren't filled in correctly (e.g. the value of `fan_leds`). So let's try it again.

@sdns575, can you give this a try?

Fixes: #498

---

Checklist:

- [x] Adhere to th [... cut]