liquidctl/liquidctl #503: NZXT Kraken X53 new device id 0x2014

MrDeepPurple, 2022-09-06:
Device type:  AIO liquid cooler
Product page: [NZXT Kraken X53](https://nzxt.com/it-IT/product/kraken-x53)
First-party software: NZXT CAM

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors: 
 - monitoring of pump speeds or duty cycles: 
 - configurable pump speeds: 
 - configurable lighting of embedded LEDs: 

The device is physically connected to the host system using:

- USB HID

Additional information about the device:

The device seems to be exactly the same as the previous NZXT Kraken x53 but the product ID of the USB interface seems to be changed. I manually edited the driver in the liquidctl Driver directory changing the productID and all seems to work correctly.

```
$ lsusb
Bus 001 Device 011: ID 1e71:2014 NZXT NZXT USB Device
```

After changing line 
```
SUPPORTED_DEVICES = [
        (0x1e71, 0x2014, None, 'NZXT Kraken X (X53, X63 or X73)', { # changed 0x2007 with 0x2014
            'speed_channels': _SPEED_CHANNELS_KRAKENX,
            'color_channels': _COLOR_CHANNELS_KRAKENX,
        })
    ]
```

liquidctl detects correctly the NZXT Kraken x53
```
liquidctl initialize --match kraken 
NZXT Kraken X (X53, X63 or X73)
├── Firmware version       2.1.0  
├── Pump Logo LEDs      detected  
└── Pump Ring LEDs      detected  
```

I also tried setting the pump speed successfully

Finally, I can help with:

- testing changes on Linux:

jonasmalacofilho, 2022-09-11:
> I also tried setting the pump speed successfully

Have you tried the LED controls, and are they also working?

MrDeepPurple, 2022-09-11:
> > I also tried setting the pump speed successfully
> 
> Have you tried the LED controls, and are they also working?

Yes, i was able to switch different effects and color both on ring and led. Everything seems to work as expected

commit 57231ba1 (2022-09-22): kraken3: register new Kraken X[567]3 PID

Fixes: #503
Suggested-by: @MrDeepPurple
closed 2022-09-22 by jonasmalacofilho