liquidctl/liquidctl #505: `liquidctl status` hangs every once in a while on an (up-to-date) Commander Core

intelfx, 2022-09-18:
I'm trying to use liquidctl (after applying #501) on a Commander Core device I have (as part of the H100i ELITE CAPELLIX CLC).
If I run `liquidctl status` in a loop each second, eventually an invocation hangs indefinitely and never exits.

According to strace, it hangs on a `read()` from a `/dev/hidrawN` node corresponding to the Commander Core.

* `watch -n1 liquidctl status`
* expected behavior: no hangs
* observed behavior: within 5-10 minutes an invocation of liquidctl would hang indefinitely

`liquidctl status --debug` output from an invocation reproducing the issue: [liquidctl-status-debug.txt](https://github.com/liquidctl/liquidctl/files/9592952/liquidctl-status-debug.txt)

Additional information about the system:

- Affected device(s): Corsair Commander Core
- Operating system and version: Arch Linux
- Liquidctl installation method: [`liquidctl-git`](https://aur.archlinux.org/packages/liquidctl-git) AUR package
- Output of `liquidctl --version`: `liquidctl v1.11.0.dev41+ga2b4391 (Linux-5.19.9-arch1pf4-1-x86_64-with-glibc2.36)`

jonasmalacofilho, 2022-09-18:
Cc: @Meriemi-git @ParkerMc

jonasmalacofilho, 2022-09-18:
Looking at your WIP proposal for a fix, it occurred to me: are you perhaps running some other instance of liquidctl (accessing that same device) while your `watch -n1 liquidctl status` loop is running?

Because what your fix effectively does is assume that the device will sometimes simply forget to reply (successfully) to `_CMD_OPEN_ENDPOINT`... but what could also happen, IIRC, is that the device did respond, but some other driver instance discarded the message by calling `clear_enqueued_reports()`.

(Specifically, IIRC, the report queue is global and, on Linux/hidraw, shared by all readers).

intelfx, 2022-09-18:
> Looking at your WIP proposal for a fix, it occurred to me: are you perhaps running some other instance of liquidctl (accessing that same device) while your watch -n1 liquidctl status loop is running?

No. Even if I did, on my system `liquidctl` is invoked via a custom wrapper for mutual exclusion (since the #298 discussion).

jonasmalacofilho, 2022-09-18:
> No. Even if I did, on my system `liquidctl` is invoked via a custom wrapper for mutual exclusion (since the #298 discussion), so this should not be happening.

Oh, right... by the way, to which file do you attach the lock for your wrapper?

> It does look suspicious, though. I'm not sure how I could verify that no one else is accessing the device concurrently. Any ideas?

Hm, maybe `lsof` is enough?  Or globally `strace`'ing (then filtering) the relevant syscalls?

Apart from those, there are is the kernel's tracing functionally...

jonasmalacofilho, 2022-09-18:
> Or globally `strace`'ing (then filtering) the relevant syscalls?

Hm, re-reading the man page for `strace` suggests that functionally doesn't actually exist.

I still vaguely remember reading about some some way to globally (and easily) trace all syscalls on a system...

intelfx, 2022-09-18:
> Oh, right... by the way, to which file do you attach the lock for your wrapper?

It doesn't really matter, so I just do it on the liquidctl executable itself:
```
$ cat /usr/local/bin/liquidctl                            
#!/bin/sh
exec flock -F /usr/bin/liquidctl /usr/bin/liquidctl "$@"
```

> 

I still vaguely remember reading about some some way to globally (and easily) trace all syscalls on a system...

There's bpftrace but I really hoped to avoid dealing with it :-)

jonasmalacofilho, 2022-09-18:
> It doesn't really matter, so I just do it on the liquidctl executable itself:

Yeah, as long as every instance agrees, it really doesn't matter.

By the way, are you sure `/usr/local/bin/liquidctl` is what's being called everywhere?  Presumably you could have scripts running with different environments, where `/usr/local/bin` isn't in the `PATH`.

> There's bpftrace (and friends) but I really hoped to avoid dealing with it :-)

Maybe that's what I was thinking of...

Anyway, if I'm reading the docs correctly, the `fanotify` APIs might have enough information (i.e. the PID that triggered the event), but it isn't currently reported by the `inotifywa*` CLI tools.

However, since you know the expected interval between `open` calls, maybe just looking if there are accesses that don't match that cadence, with

```
$ inotifywait -m --format '%T %w %e' --timefmt '%F %T' /dev/hidraw?
```

would be enough to at least see if further investigation is even necessary?

(Since the time resolution is rather limited, it might be good increase the `watch` loop interval).

intelfx, 2022-09-18:
> By the way, are you sure /usr/local/bin/liquidctl is what's being called everywhere? Presumably you could have scripts running with different environments, where /usr/local/bin isn't in the PATH.

Yes, pretty sure, $PATH is set globally via systemd:

```
# grep -z '^PATH=' /proc/*/environ | tr '\0' '\n' | grep -v '/usr/local/bin'
/proc/154661/environ:PATH=/bin:/usr/bin:/sbin:/usr/sbin  # this single process belongs to a container and is irrelevant
```

> Maybe that's what I was thinking of...

I have confirmed with bpftrace that there are indeed no other processes attempting to access the hidraw node.

Terminal 1:
```
# FOOBAR=1 watch -n1 liquidctl status
```

Terminal 2:
```
# sudo bpftrace -B line -e 'tracepoint:syscalls:sys_enter_open, tracepoint:syscalls:sys_enter_openat /str(args->filename) == "/dev/hidraw7"/ { printf("OUT: %d %s %s\n", pid, comm, str(args->filename)); }' | grep -E 'OUT:' --line-buffered | while read _ pid comm fn; do echo "$(grep -z 'FOOBAR=' /proc/$pid/environ) $pid $comm $fn"; done
stdin:1:95-97: WARNING: Addrspace mismatch                                                                                                                                                                                                                     
tracepoint:syscalls:sys_enter_open, tracepoint:syscalls:sys_enter_openat /str(args->filename) == "/dev/hidraw7"/ { printf("OUT %d %s %s\n", pid, comm, str(args->filename)); }                                                                                 
                                                                                              ~~                                                                                                                                                               

FOOBAR=1 <pid> liquidctl /dev/hidraw7
<many lines, each tagged with FOOBAR=1>
FOOBAR=1 <pid> liquidctl /dev/hidraw7
<hang, no more lines>
```

Meriemi-git, 2022-09-18:
I try to reproduce your problem i install the latest yay version for liquidctl-git and I'm on Arch too. 
For me there is no lock when looping on status :
`watch -n1 liquidctl status`
Every second my AIO blink and status is updated.

intelfx, 2022-09-19:
It does not necessarily happen right away. Might be tens of minutes. Try to leave it running for some time?

Otherwise I'm not sure what the problem is then. @jonasmalacofilho suggested I might have something concurrently accessing the hidraw node and breaking the protocol, but I have verified that it's not the case (at least in userspace). Any other ideas?

Grooty12, 2022-09-19:
Happens to me as well. Just that it happens immediately, I don't have to wait.

Meriemi-git, 2022-09-19:
After wait few minutes it stuck for me too.

ParkerMc, 2022-09-19:
Would any of y'all mind getting a capture in Wireshark of this issue happening?

Meriemi-git, 2022-09-19:
There is a Wireshark capture file, freeze appear at the end during few seconds before i stop the capture. 
I rename it to .txt to be able to upload it, the only filters i applied were idVendor IdProduct and device_address as i read in the documentation.
[commander_core_status_freeze.pcapng.txt](https://github.com/liquidctl/liquidctl/files/9601902/commander_core_status_freeze.pcapng.txt)

ParkerMc, 2022-09-20:
> I post a Wireshark capture file, freeze appears at the end during few seconds before i stop the capture.

Looking at that it appears to freeze on a read command `0x08`
@Meriemi-git Do you mind doing the same thing but running liquidctl with `--debug` and uploading the log and capture?

Meriemi-git, 2022-09-21:
Yes i will do it today

Meriemi-git, 2022-09-27:
Sorry for the delay, it tooks me a long time to reproduce the hang (between one and two hour) with Wireshark capture and --debug option.
To be able to capture the full log i tried with `tee` and the `watch -n 1` command but i wasn't able to write the entire debug info so i write a simple bash script to reproduce the watch behavior :
```
#!/bin/bash
i=0
while true
do
    liquidctl status --debug >> liquidctl_freeze.log 2>&1
    sleep 1
    ((i=i+1))
    echo "Duration : $i sec"
done
```
    
[liquidctl_freeze_.log](https://github.com/liquidctl/liquidctl/files/9659506/liquidctl_freeze_.log)
[liquidctl_freeze.pcapng.txt](https://github.com/liquidctl/liquidctl/files/9659509/liquidctl_freeze.pcapng.txt)

ParkerMc, 2022-09-28:
Thank you for doing that.
Try setting `_REPORT_LENGTH` and `_RESPONSE_LENGTH` to 96 and see if that fixes it.
https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/commander_core.py#L20-L21

Meriemi-git, 2022-09-28:
Tested during 5 hours without any freeze, but usually the freeze can happen after several hours so @intelfx can you test it ?

Meriemi-git, 2022-10-03:
[PR 513](https://github.com/liquidctl/liquidctl/pull/513)
Here is the pull request to apply @ParkerMc fix.

jonasmalacofilho, 2022-10-03:
@intelfx, can you please take a look at #513 and see if that also fixes things on your system?

intelfx, 2022-10-03:
Apologies for the delay. Testing now.

intelfx, 2022-10-04:
Yep, it looks like #513 works for me.

PR #513 (2022-10-04): Commander core fix hang
Reduce the _REPORT_LENGTH and _RESPONSE_LENGTH from 1024 to 96 to fix hang when loop over liquidctl status command according to @ParkerMc [comment](https://github.com/liquidctl/liquidctl/issues/505#issuecomment-1260311397) on issue #505 

Fixes:  #505

---

Checklist:

- [x] Adhere to the [development process]
- [x] Conform to the [style guide]
- [x] Verify that the changes work as expected on real hardware
- [ ] Add automated tests cases
- [x] Verify that all (other) automated tests (still) pass
- [ ] Update the README and other applicable documentation pages
- [ ] Upd [... cut]
commit ad801346 (2022-10-16): commander_core: fix report and response length (#513)

Prevents sporadic hangs reported in #505.

Fixes: #505
closed 2022-10-04 by jonasmalacofilho