liquidctl/liquidctl #511: CORSAIR iCUE Commander ST

zzjas, 2022-09-25:
Hi, I recently purchased a CORSAIR - iCUE H150i ELITE CAPELLIX CPU Cooler from Best Buy. Though in the description it says a "CORSAIR Commander CORE" will be included, I can only find a device called "CORSAIR iCUE Commander ST". Would it be possible to control the Commander ST with liquidctl? I'd be happy to assist if any help is needed. Thanks!

Here's the device information from `lsusb`.
```
$ lsusb -v -s 001:007
Bus 001 Device 007: ID 1b1c:0c32 Corsair CORSAIR iCUE Commander ST
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c32
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR iCUE Commander ST
  iSerial                 3 207832A43748
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0066
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

MarcinZablocki, 2022-11-06:
Adding USB ID to commander_core driver allows reading data from the device just fine. 

`> sudo liquidctl status
Corsair Commander Core ST (experimental)
├── Pump speed           2299  rpm
├── Fan speed 1           362  rpm
├── Fan speed 2           363  rpm
├── Fan speed 3           357  rpm
├── Fan speed 4             0  rpm
├── Fan speed 5             0  rpm
├── Fan speed 6           415  rpm
└── Water temperature    30.0  °C
`

`        
    (0x1b1c, 0x0c32, 'Corsair Commander Core ST (experimental)', {"has_pump": True}),
`

closed 2022-11-25 by jonasmalacofilho

paskalll, 2022-12-30:
CORSAIR iCUE Commander ST

This device is no core.
Can't find anything to this device it's like it don't exist.

jonasmalacofilho, 2023-01-02:
@paskalll 

The reason why we currently consider this to be a Commander Core ST comes from the initial description of the issue:

> Though **in the description it says a "CORSAIR Commander CORE" will be included,** I can only find a device called "CORSAIR iCUE Commander ST".

And the fact that both of the product descriptors bellow can be observed for the same VID:PID:

> ```
> Bus 001 Device 007: ID 1b1c:0c32 Corsair CORSAIR iCUE Commander ST
> Bus 002 Device 002: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core
> ```

Plus the apparent similarities in protocol further reinforces the idea that this is some type of Commander Core variant. Perhaps not that compatible with previous Commander Cores (Core and Core XT), hence the ST.

dlm21, 2023-02-03:
I've got one of these, if anyone needs someone to test stuff on I'm willing and able :)

From lsusb:
Bus 001 Device 009: ID 1b1c:0c39 Corsair LCD Cap for Elite Capellix coolers
Bus 001 Device 006: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core

❯ sudo liquidctl status
ASUS Aura LED Controller (experimental)
├── ARGB channels: 3      
└──  RGB channels: 1      

Corsair Commander ST (experimental)
├── Pump speed           2553  rpm
├── Fan speed 1          1074  rpm
├── Fan speed 2          1051  rpm
├── Fan speed 3          1054  rpm
├── Fan speed 4             0  rpm
├── Fan speed 5             0  rpm
├── Fan speed 6             0  rpm
└── Water temperature    34.8  °C

rwmjones, 2023-06-15:
Does anyone know how to turn the ludicrous LEDs off (permanently, without Windows) on the 1b1c:0c32 device?

commit c095479b (2022-11-25): commander_core: allow the driver to bind to a Commander ST

Newer H150i Elite Cappelix AIOs have been featuring a Commander ST,
instead of a Commander Core.

From preliminary information available so far, they appear to be very
similar, perhaps identical.

Fixes: #511
Related: https://github.com/liquidctl/liquidctl/issues/408#issuecomment-1326873959