liquidctl/liquidctl #515: Sudo and non sudo return different errors

igor17400, 2022-10-05:
When I execute the command `liquidctl list` it returns to me the following devices:

```
Device #0: AsusTek Aura LED Controller (experimental)
Device #1: Corsair Commander Core (experimental)
```

However, when I try to execute `sudo liquidctl list` it returns nothing.

And when I try to execute any other command such as `liquidctl status` it returns an error. And when the same command is executed with `sudo` it returns a different error `Error: no devices matches available drivers and selection criteria`. 

The image below shows such scenario.

[image]

Additional information about the system:

- Affected device(s): all devices
- Operating system and version: ubuntu 22.04.1 LTS
- Liquidctl installation method: `python -m pip install liquidctl`
- Output of `liquidctl --version`: `liquidctl v1.10.0 (Linux-5.15.0-48-generic-x86_64-with-glibc2.35)`

jonasmalacofilho, 2022-10-05:
Can you post the output from all of the following commands?

```
$ which liquidctl
$ liquidctl list --debug
$ sudo which liquidctl
$ sudo liquidctl list --debug
```

igor17400, 2022-10-05:
@jonasmalacofilho  The output for each command:

1. `which liquidctl`

```
/home/itachi/anaconda3/envs/igor174/bin/liquidctl
```

2. `liquidctl list --debug`

```
[DEBUG] (cli) (main): liquidctl: 1.10.0
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.13 | packaged by conda-forge | (main, May 27 2022, 16:58:50)
[GCC 10.3.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0037 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0053 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0051 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0036 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b05:19af (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (probe): AuraLed identified: AsusTek Aura LED Controller (experimental)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0xff00 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 413c:301d (usage_page=0xff01 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 174c:3074
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 174c:2074
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 413c:301d
[DEBUG] (usb) (find_devices): USB device: 0b05:19af
[DEBUG] (usb) (find_devices): USB device: 8087:0026
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
Device #0: AsusTek Aura LED Controller (experimental)
├── Vendor ID: 0x0b05
├── Product ID: 0x19af
├── Release number: 0x0100
├── Serial number: 9876543210
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: AuraLed
[DEBUG] (cli) (_list_devices_human): MRO: AuraLed, UsbHidDriver, BaseUsbDriver, BaseDriver, object

[... 32 lines cut ...]
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: ignoring 0-0036, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 0b05:19af
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 413c:301d
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1c
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1c
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 174c:3074
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 174c:2074
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c1c
[DEBUG] liquidctl.driver.usb: found USB device 413c:301d
[DEBUG] liquidctl.driver.usb: found USB device 0b05:19af
[DEBUG] liquidctl.driver.usb: found USB device 8087:0026
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
```

jonasmalacofilho, 2022-10-05:
You have two separate installations of liquidctl, and which one is getting picked depends on how `PATH` is set up in each case.

For your regular user:

```
/home/itachi/anaconda3/envs/igor174/bin/liquidctl
[DEBUG] (cli) (main): liquidctl: 1.10.0
```

With `sudo`:

```
/usr/bin/liquidctl
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
```

Adjustment your environment variables (and/or whether `sudo` inherits your user's environment), and/or remove one of the installations.

closed 2022-10-05 by jonasmalacofilho