liquidctl/liquidctl #516: Kraken Z73 not working console lsd command.

bismoney, 2022-10-05:
Hi guys!

Not working fo me :/

`❯ liquidctl --match kraken set lcd screen liquid
Usage:
  liquidctl [options] list
  liquidctl [options] initialize [all]
  liquidctl [options] status
  liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
  liquidctl [options] set <channel> speed <percentage>
  liquidctl [options] set <channel> color <mode> [<color>] ...
  liquidctl --help
  liquidctl --version
❯ liquidctl --match kraken set lcd screen orientation 90
Usage:
  liquidctl [options] list
  liquidctl [options] initialize [all]
  liquidctl [options] status
  liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
  liquidctl [options] set <channel> speed <percentage>
  liquidctl [options] set <channel> color <mode> [<color>] ...
  liquidctl --help
  liquidctl --version
❯ liquidctl list
Device #0: Corsair CMWX8GD3600C18W2D DIMM1
Device #1: Corsair CMWX8GD3600C18W2D DIMM2
Device #2: Corsair CMWX8GD3600C18W2D DIMM3
Device #3: Corsair CMWX8GD3600C18W2D DIMM4
Device #4: Gigabyte RGB Fusion 2.0 8297 Controller
Device #5: NZXT Kraken Z (Z53, Z63 or Z73)
`

`❯ liquidctl --version
liquidctl v1.10.0 (Linux-5.18.19-3-MANJARO-x86_64-with-glibc2.36)`

bismoney, 2022-10-05:
And Friemwork version Kraken
`NZXT Kraken Z (Z53, Z63 or Z73)
└── Firmware version    5.7.0  
`

bismoney, 2022-10-05:
debug log

> `sudo liquidctl --match kraken --debug status
[DEBUG] (cli) (main): liquidctl: 1.10.0
[DEBUG] (cli) (main): platform: Linux-5.18.19-3-MANJARO-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_requirements): python: 3.10.7 (main, Sep  6 2022, 21:22:27) [GCC 12.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0052 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0050 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0037 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0053 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0051 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0036 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0416:c345 (usage_page=0xff1b usage=0x0091)
[DEBUG] (usb) (find_devices): HID device: 1038:1724 (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:1724 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1038:1724 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:1724 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1038:1724 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 048d:8297 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:8297 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 3842:2002 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3842:2002 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3842:2002 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 3842:2002 (usage_page=0x0001 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1e71:3008 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken Z (Z53, Z63 or Z73)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 8087:0aaa
[DEBUG] (usb) (find_devices): USB device: 1e71:3008
[DEBUG] (usb) (find_devices): USB device: 3842:2002
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:8297
[DEBUG] (usb) (find_devices): USB device: 1038:1724
[DEBUG] (usb) (find_devices): USB device: 0416:c345
[DEBUG] (usb) (find_devices): USB device: 0bda:5411
[DEBUG] (usb) (find_devices): USB device: 0bda:5411
[DEBUG] (usb) (find_devices): USB device: 1a40:0101
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT Kraken Z (Z53, Z63 or Z73)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x74 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 75:01:42:00:41:00:04:50:53:43:48:32:37:20:01:1c:06:67:06:2e:2e:01:02:07:02:14:14:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT Kraken Z (Z53, Z63 or Z73)
├── Liquid temperature    28.6  °C
├── Pump speed            1639  rpm
├── Pump duty               46  %
├── Fan speed              519  rpm
└── Fan duty                20  %`

bismoney, 2022-10-06:
i'm idiots... working...

closed 2022-10-06 by bismoney

SpanishPear, 2022-10-06:
@bismoney what was your issue? I think I may also be doing something silly