liquidctl/liquidctl #517: Not list Corsair iCUE Elite Capellix H100i

KevinD777, 2022-10-08:
---

Additional information about the system:

- Affected device(s):  Corsair iCUE Elite Capellix H100i with Corsair Commander Core
- Operating system and version:  Ubuntu 22.04.1 LTS
- Liquidctl installation method: apt install
- Output of `liquidctl --version`: liquidctl v1.5.1

Hi,

the liquidctl won't list my device (Corsair iCUE Elite Capellix H100i with Corsair Commander Core). Here are the outputs of `list`and `debug` commands:

~$ liquidctl list
<code>Device ID 0: Gigabyte RGB Fusion 2.0 5702 Controller</code>

~$ lsusb -vt
<pre>
Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 5: Dev 2, If 0, Class=Wireless, Driver=btusb, 480M
        ID 0e8d:0608 MediaTek Inc. 
    |__ Port 5: Dev 2, If 1, Class=Wireless, Driver=btusb, 480M
        ID 0e8d:0608 MediaTek Inc. 
    |__ Port 5: Dev 2, If 2, Class=Wireless, Driver=, 480M
        ID 0e8d:0608 MediaTek Inc. 
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 0461:4e8d Primax Electronics, Ltd 
        |__ Port 2: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1ea7:0907 SHARKOON Technologies GmbH Keyboard
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1ea7:0907 SHARKOON Technologies GmbH Keyboard
        |__ Port 2: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1ea7:0907 SHARKOON Technologies GmbH Keyboard
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 048d:5702 Integrated Technology Express, Inc. 
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 480M
            ID 1b1c:0c1c Corsair 
        |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 480M
            ID 1b1c:0c1c Corsair 
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 20000M/x2
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
</pre>

~$ liquidctl list --debug
<pre>
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 048d:5702
[DEBUG] liquidctl.driver.usb: found HID device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 5702 Controller
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1c
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1c
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found HID device 0461:4e8d
[DEBUG] liquidctl.driver.usb: found HID device 0461:4e8d
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 0461:4e8d
[DEBUG] liquidctl.driver.usb: found USB device 1ea7:0907
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 0e8d:0608
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c1c
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 048d:5702
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
Device ID 0: Gigabyte RGB Fusion 2.0 5702 Controller
├── Vendor ID: 0x048d
├── Product ID: 0x5702
├── Release number: 0x0001
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: RgbFusion2
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver
</pre>

However, I installed Coolero (GUI based on liquidctl) and it shows my Corsair Commander Core. 
[image]

Coolero can only be accessed via a screen, but I also want to monitor and control the fans via SSH CLI, so I want to make liquidctl work as well. Thanks!

jonasmalacofilho, 2022-10-08:
> * Output of `liquidctl --version`: liquidctl v1.5.1

You need a more recent version of liquidctl to work with that device.  Preferably 1.10.0 (or, beginning Tuesday, 1.11.0) or the latest code in the `main` branch.

jonasmalacofilho, 2022-10-08:
P.S. Thanks for the detailed report.

closed 2022-10-08 by jonasmalacofilho

KevinD777, 2022-10-09:
> > * Output of `liquidctl --version`: liquidctl v1.5.1
> 
> You need a more recent version of liquidctl to work with that device. Preferably 1.10.0 (or, beginning Tuesday, 1.11.0) or the latest code in the `main` branch.

Thanks!