liquidctl/liquidctl #520: Corsair Commander Core Error: Incorrect Data Type (Even with the newest codes)

KevinD777, 2022-10-09:
---

Additional information about the system:

- Affected device(s): Corsair Commander Core
- Operating system and version: Ubuntu 22.04
- Liquidctl installation method: Pip
- Output of `liquidctl --version`: v1.10.0

I met the same issue with [#455](https://github.com/liquidctl/liquidctl/issues/455), and I produced some logs according to your suggestions in that post. 

I also tried using the codes in the main branch, but the error still exists.

Could you please help with that? Thanks!

Here's the debug log:
<pre>
~$ liquidctl set -d 1 fan1 speed 10 --debug
[DEBUG] (cli) (main): liquidctl: 1.10.0
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[WARNING] (cli) (main): -d/--device is deprecated, prefer --match or other selection options
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5702 Controller
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 1024 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 768 bytes: 00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00: [... cut ...]
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 120, in set_fixed_speed
    res = self._read_data(_MODE_HW_SPEED_MODE, _DATA_TYPE_HW_SPEED_MODE)
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 181, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: device returned incorrect data type
[ERROR] (cli) (log): unexpected error with Corsair Commander Core (experimental): ExpectationNotMet('device returned incorrect data type')
</pre>

KevinD777, 2022-10-09:
Here's the output of <code>liquidctl --match core status --debug</code>
<pre>
$ liquidctl --match core status --debug
[DEBUG] (cli) (main): liquidctl: 1.10.0
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 1024 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 [... cut ...] 0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
</pre>

jonasmalacofilho, 2022-10-09:
> I also tried using the codes in the main branch, but the error still exists.

Can you show that `--debug` output?

---

And I'm sorry for failing to account for the new firmware before, when I recommended liquidctl 1.10.0 for this device.  But I'm glad you tried the code in `main` anyway.

KevinD777, 2022-10-09:
> 

Thanks for your reply and sure, here it is.

<pre>
$ liquidctl set -d 1 fan1 speed 10 --debug
[DEBUG] (cli) (main): script: /home/jiangshd/miniconda3/bin/liquidctl
[DEBUG] (cli) (main): version: 0.0.0-unknown
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[WARNING] (cli) (main): -d/--device is deprecated, prefer --match or other selection options
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5702 Controller
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:0d:00:60:6d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:0d:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 146, in set_fixed_speed
    res = self._read_data(_MODE_HW_SPEED_MODE, _DATA_TYPE_HW_SPEED_MODE)
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 206, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: device returned incorrect data type
[ERROR] (cli) (log): unexpected error with Corsair Commander Core (experimental): ExpectationNotMet('device returned incorrect data type')
<pre>

KevinD777, 2022-10-09:
Here's the output of <code>liquidctl --match core status --debug</code> with main branch codes.
<pre>
$ liquidctl --match core status --debug
[DEBUG] (cli) (main): script: /home/jiangshd/miniconda3/bin/liquidctl
[DEBUG] (cli) (main): version: 0.0.0-unknown
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:0d:00:17:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:0d:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:08:00:06:00:07:dc:08:b3:02:66:02:48:04:5c:04:a7:02:cb:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[... cut ...]
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
Corsair Commander Core (experimental)
├── Pump speed           2268  rpm
├── Fan speed 1           691  rpm
├── Fan speed 2           614  rpm
├── Fan speed 3          1096  rpm
├── Fan speed 4          1116  rpm
├── Fan speed 5           679  rpm
├── Fan speed 6           715  rpm
└── Water temperature    29.8  °C
</pre>

jonasmalacofilho, 2022-10-09:
Thanks!  But it seems that you're not running as recent code as you thought.

Based on what information is logged, the available drivers, and registered dependencies, it seems that you're running code that hit the `main` branch between 20 and 70 days ago.  Which makes it possible/likely that you're still missing the important fix for 2.x Commander Core firmware.  And you're certainly missing a later, smaller but related, fix.

KevinD777, 2022-10-09:
> Thanks! But it seems that you're not running as recent code as you thought.
>
> Based on what information is logged, the available drivers, and registered dependencies, it seems that you're running [...]

Sorry, my bad. I re-installed with git and here's the debug output, is it the latest version now? 
<pre>
$ liquidctl set -d 1 fan1 speed 10 --debug
[DEBUG] (cli) (main): script: /home/jiangshd/miniconda3/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.0.dev88+ge08eebe
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 8.2.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[WARNING] (cli) (main): -d/--device is deprecated, prefer --match or other selection options
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5702 Controller
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:00:2b:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:0d:00:60:6d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:0d:00:10:00:02:00:2b:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 [... cut ...]
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 146, in set_fixed_speed
    res = self._read_data(_MODE_HW_SPEED_MODE, _DATA_TYPE_HW_SPEED_MODE)
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 206, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: device returned incorrect data type
[ERROR] (cli) (log): unexpected error with Corsair Commander Core (experimental): ExpectationNotMet('device returned incorrect data type')

</pre>

KevinD777, 2022-10-09:
Here's the latest output of <code>liquidctl --match core status --debug</code>
<pre>
$ liquidctl --match core status --debug
[DEBUG] (cli) (main): script: /home/jiangshd/miniconda3/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.0.dev88+ge08eebe
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 8.2.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander Core (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:0d:00:17:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:0d:00:10:00:02:00:2a:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:08:00:06:00:07:c5:08:cf:02:71:02:4b:04:66:04:ac:02:e1:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:00:29:01:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
Corsair Commander Core (experimental)
├── Pump speed           2245  rpm
├── Fan speed 1           719  rpm
├── Fan speed 2           625  rpm
├── Fan speed 3          1099  rpm
├── Fan speed 4          1126  rpm
├── Fan speed 5           684  rpm
├── Fan speed 6           737  rpm
└── Water temperature    29.7  °C

</pre>

jonasmalacofilho, 2022-10-09:
@KevinD777 ,

> Sorry, my bad. I re-installed with git and here's the debug output, is it the latest version now?

Yes, thanks!

---

@ParkerMc and @Meriemi-git,

It seems that we're sometimes still getting unexpected responses from the Commander Core, despite the recent fixes.

Can you take a look at this, when you have some time?

Thanks!

ParkerMc, 2022-10-09:
Hi @KevinD777
If I am reading this right, the issue appeared with the older version and then continued latest new versions. Is that correct?
That would make sense because the fix just prevents the device from ending up in the broken state.
These instructions will get you into a good state, and from there, you can use any version with the fix. https://github.com/liquidctl/liquidctl/issues/455#issuecomment-1121434961

ParkerMc, 2022-10-09:
@jonasmalacofilho, if you want, we can ignore that safety check when the datatype is equal to `00:00` as that is what happens when a write fails(the cause of this bug.) From there, liquidctl will automatically bring the device into a good state.

jonasmalacofilho, 2022-10-10:
@ParkerMc,

Thanks for the quick response.

Maybe we could turn it into a warning (if the data type is `00:00`)?  This way the user is still notified, and will likely report to us if, for some reason, the warning keeps coming back.

KevinD777, 2022-10-10:
> Hi @KevinD777 If I am reading this right, the issue appeared with the older version and then continued latest new versions. Is that correct? That would make sense because the fix just prevents the device from ending up in the broken state. These instructions will get you into a good state, and from there, you can use any version with the fix. [#455 (comment)](https://github.com/liquidctl/liquidctl/issues/455#issuecomment-1121434961)

Hi @ParkerMc ,

Thanks for you help and here is my detailed situation:

I firstly used Coolero to manage my fan's speed, and it worked fine. After that, I installed <code>liquidctl</code> via <code>apt install</code> since I wanted to check fan speeds in the terminal, and it was an old version (v1.5.1) and thus could not recognize the Commander Core so I posted #517. Then after a few hours and multiple reboots, I noticed that the Coolero stopped working due to the "incorrect data type" error (I am not sure if Coolero stopped working immediately after I installed the old version <code>liquidctl</code>). 
So I uninstalled the old version <code>liquidct</code>, updated it, and then met the issue I posted here.

I initialized devices again and noticed that my Commander Core firmware version is 1.7.142, see below:
<pre>
Corsair Commander Core (experimental)
├── Firmware version            1.7.142  
├── AIO LED count                    29  
├── RGB port 1 LED count              8  
├── RGB port 2 LED count              8  
├── RGB port 3 LED count            N/A  
├── RGB port 4 LED count            N/A  
├── RGB port 5 LED count            N/A  
├── RGB port 6 LED count            N/A  
├── AIO port connected              Yes  
├── Fan port 1 connected            Yes  
├── Fan port 2 connected            Yes  
├── Fan port 3 connected            Yes  
├── Fan port 4 connected            Yes  
├── Fan port 5 connected            Yes  
├── Fan port 6 connected            Yes  
├── Water temperature sensor        Yes  
└── Temperature sensor 1             No  
</pre>

Also, I tried the instructions here ([#455 (comment)](https://github.com/liquidctl/liquidctl/issues/455#issuecomment-1121434961)), commented out the exception but it gave me a new error <code>IndexError: bytearray index out of range
</code>, here's the debug output.
<pre>
$ liquidctl --match core set fan1 speed 30 --debug
[DEBUG] (cli) (main): script: /home/jiangshd/miniconda3/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.0.dev88+ge08eebe
[DEBUG] (cli) (main): platform: Linux-5.15.0-48-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.9.12 (main, Apr  5 2022, 06:56:58) 
[GCC 7.5.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 8.2.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0461:4e8d (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0461:4e8d (usage_page=0x0001 usage=0x0001)
[... 22 lines cut ...]
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 96 bytes: 08:0d:00:60:6d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/cli.py", line 291, in _device_set_speed
    dev.set_fixed_speed(args['<channel>'].lower(), int(args['<percentage>'][0]), **opts)
  File "/home/jiangshd/miniconda3/lib/python3.9/site-packages/liquidctl/driver/commander_core.py", line 151, in set_fixed_speed
    data[chan + 1] = 0x00  # Set the device's hardware mode to fixed percent
IndexError: bytearray index out of range
[ERROR] (cli) (log): unexpected error with Corsair Commander Core (experimental): IndexError('bytearray index out of range')

</pre>

jonasmalacofilho, 2023-06-10:
Someone should take a fresh look at implementing automatic correction of corrupted device states. See #522 for a first attempt that, sadly, failed.

ProjectSynchro, 2024-01-27:
I tried out the above PR, and it does at least report the issue when using a Core XT in a corrupted state, but as stated, reinitializing the device doesn't appear to work.

Would love to be able to help but I am pretty inexperienced with reverse engineering protocols. 

Would wireshark captures or something similar help someone with more knowledge? 

I have noticed that when "plugging" in the corrupted Core XT into a Windows VM running iCue, iCue will either reflash the device if it is super fucked (no idea how it determines that, or if it is a scorched earth fix), or it will reset the device.. 

Maybe capturing that/those reset command(s) could be helpful?

jonasmalacofilho, 2024-01-28:
@ProjectSynchro,

> I tried out the above PR, and it does at least report the issue when using a Core XT in a corrupted state, but as stated, reinitializing the device doesn't appear to work.

Thanks for the feedback.

> Maybe capturing that/those reset command(s) could be helpful?

Yes, it will likely help.

jonasmalacofilho, 2024-07-14:
Take a look at [Capturing USB traffic](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md). Hopefully it will help you get started...

jonasmalacofilho, 2024-12-16:
Merged with other similar issues into #753.

closed 2024-12-16 by jonasmalacofilho