liquidctl/liquidctl #526: NZXT RGB & Fan Controller Timeout

codifryed, 2022-10-18:
Hi :wave: 

When the hwmon kernel driver is not present or used, the NZXT RGB & Fan Controller needs longer than 2 seconds to complete a read during initialization, causing a Timeout error with liquidctl 1.11.0. This is repeatable and confirmed by another user as well.

```
liquidctl --version              
liquidctl v1.11.0 (Linux-6.0.2-zen1-1-zen-x86_64-with-glibc2.36)

liquidctl -m "rgb" initialize
NZXT RGB & Fan Controller
└── Firmware version    1.13.0

liquidctl -m "rgb" initialize --direct-access
WARNING: forcing re-initialization despite nzxt-smart2 kernel driver
ERROR: unexpected error with NZXT RGB & Fan Controller: Operation timed out
```

When running the same command with 1.10.0:
```
liquidctl --version
liquidctl v1.10.0 (Linux-6.0.2-zen1-1-zen-x86_64-with-glibc2.36)

liquidctl -m "rgb" initialize --direct-access
WARNING: forcing re-initialization despite nzxt_smart2 kernel driver
NZXT RGB & Fan Controller
└── Firmware version    1.13.0 
```

Using Coolero I was able to determine the approximate time needed for initialization with Liquidctl 1.10.0 
```
2022-10-18 21:05:16,260 DEBUG: liquidctl.driver.usb - discarded 0 previously enqueued reports
2022-10-18 21:05:16,260 WARNING: liquidctl.driver.smart_device - forcing re-initialization despite nzxt_smart2 kernel driver
2022-10-18 21:05:16,260 DEBUG: liquidctl.driver.usb - writting report 0x60 with 63 bytes: 02:01:e8:01:01:e8:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
2022-10-18 21:05:16,261 DEBUG: liquidctl.driver.usb - writting report 0x60 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
2022-10-18 21:05:16,262 DEBUG: liquidctl.driver.usb - writting report 0x10 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
2022-10-18 21:05:16,263 DEBUG: liquidctl.driver.usb - writting report 0x20 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
2022-10-18 21:05:16,264 DEBUG: liquidctl.driver.usb - read 64 bytes: 11:01:37:00:1d:00:19:51:31:31:32:33:31:37:10:20:01:01:0d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
2022-10-18 21:05:18,670 DEBUG: liquidctl.driver.usb - read 64 bytes: 21:03:37:00:1d:00:19:51:31:31:32:33:31:37:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
2022-10-18 21:05:18,670 DEBUG: coolero.repositories.liquidctl_repo - Liquidctl device initialization response: [('Firmware version', '1.13.0', '')]
``` 

I repeated this several times with the following read times:
(seconds)
2.457
2.459
2.316
2.293
2.406

This came to my attention from an issue submitted by a Coolero user, who doesn't have the kernel driver installed: https://gitlab.com/coolero/coolero/-/issues/74

Liquidctl Debug output:

1.11.0
```
liquidctl -m "rgb" initialize --direct-access --debug
[DEBUG] (cli) (main): script: /usr/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.0
[DEBUG] (cli) (main): platform: Linux-6.0.2-zen1-1-zen-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_requirements): python: 3.10.8 (main, Oct 11 2022, 20:04:56) [GCC 12.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.2.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[... 161 lines cut ...]
[DEBUG] (usb) (find_devices): USB device: 046d:0843
[DEBUG] (usb) (find_devices): USB device: 1852:7022
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT RGB & Fan Controller
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[WARNING] (smart_device) (initialize): forcing re-initialization despite nzxt_smart2 kernel driver
[DEBUG] (usb) (write): writting report 0x60 with 63 bytes: 02:01:e8:01:01:e8:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writting report 0x60 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writting report 0x10 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writting report 0x20 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 11:01:37:00:1d:00:19:51:31:31:32:33:31:37:10:20:01:01:0d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 61:03:37:00:1d:00:19:51:31:31:32:33:31:37:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 21:03:37:00:1d:00:19:51:31:31:32:33:31:37:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT RGB & Fan Controller
└── Firmware version    1.13.0 
```

---

Additional information about the system:

- Affected device(s): NZXT RGB & Fan Controller
- Operating system and version: Arch Linux
- Liquidctl installation method: Pip, Arch Linux
- Output of `liquidctl --version`: liquidctl v1.11.0 (Linux-6.0.2-zen1-1-zen-x86_64-with-glibc2.36)

jonasmalacofilho, 2022-10-18:
Hey!  First, sorry about the breakage...

Next, when I first read this report I was worried that there might be something wrong the `update_interval` on those devices.  But that doesn't seem to be the case, instead, it seems that only this specific response (lighting infos) takes a bit longer, and that makes sense.

While we could just have this specific driver use a longer timeout, I think it's also reasonable to increase the default timeout, as 2 seconds was indeed maybe a bit too optimistic: for reading periodic responses 2 seconds is probably fine, but certain initialization commands (and other writes) can take a bit longer.

Can you try 066488c767c3e5489235fa52b7d06b5f4b6d87f7 and confirm that it fixes the issue?

codifryed, 2022-10-19:
Thanks @jonasmalacofilho,
All good, I know it's pretty hard to know max response times for all these devices. Sometimes we start out optimistic. :wink: 

Fix tested successfully: (total command completion time was always < 3 seconds)
```
liquidctl --version
liquidctl v1.12.0.dev15+g066488c (Linux-6.0.2-zen1-1-zen-x86_64-with-glibc2.36)

liquidctl -m "rgb" initialize --direct-access        
WARNING: forcing re-initialization despite nzxt-smart2 kernel driver
NZXT RGB & Fan Controller
└── Firmware version    1.13.0
```

jonasmalacofilho, 2022-10-19:
@codifryed ,

Thanks!  Patch release will almost certainly go out later today.

> (total command completion time was always < 3 seconds)

Yeah, I really couldn't see any reason why the one digit fix would not work.  But maybe something else was also broken; better to wait for the confirmation than to make another broken release.

closed 2022-10-19 by jonasmalacofilho

jonasmalacofilho, 2022-10-20:
Fixed in [liquidctl 1.11.1](https://github.com/liquidctl/liquidctl/releases/tag/v1.11.1).

commit 805c323a (2022-10-19): usb: increase default timeout to 5 s to account for init messages

While 2 s was probably fine for all recurring reads, reads during
initialization can take a bit longer, even on devices that normally
respond much quicker than that.

One such example are SmartDevice2 devices, which take 2.5 seconds to
reply with the lighting info. As that's the maximum known response time,
set the default timeout value to double that value.

Only increasing the timeout locally, on the affected call, was also
considered: however, it's likely that other devices will also need
slightly longer timeouts, either dur [... cut]
commit 2d57d4b7 (2022-10-19): usb: increase default timeout to 5 s to account for init messages

While 2 s was probably fine for all recurring reads, reads during
initialization can take a bit longer, even on devices that normally
respond much quicker than that.

One such example are SmartDevice2 devices, which take 2.5 seconds to
reply with the lighting info. As that's the maximum known response time,
set the default timeout value to double that value.

Only increasing the timeout locally, on the affected call, was also
considered: however, it's likely that other devices will also need
slightly longer timeouts, either dur [... cut]