liquidctl/liquidctl #531: Corsair H115i Pro RGB

yw4z, 2022-10-28:
this one is a nice find for me but it didnt work

corsair link not working great and cue is heavy on resources

thanks for your effort on this on look like very useful,

i can give any further info if needed

my motherboard model Asus P9x79 Pro

liquidctl.exe list
Device #0: Corsair Hydro H115i Pro

liquidctl.exe status
ERROR: Corsair Hydro H115i Pro: unexpected error: NotImplementedError('Operation not supported or unimplemented on this platform')

liquidctl.exe list --debug
[DEBUG] (cli) (_log_env_infos): script: C:\Users\xxx\Downloads\FanCtrl_v1.5.1\plugin\liquidctl.exe
[DEBUG] (cli) (_log_env_infos): version: 0.0.0-unknown
[DEBUG] (cli) (_log_env_infos): platform: Windows-10-10.0.19044-SP0
[DEBUG] (cli) (_log_env_infos): python: 3.10.8 (tags/v3.10.8:aaaf517, Oct 11 2022, 16:50:30) [MSC v.1933 64 bit (AMD64)]
[DEBUG] (cli) (_log_env_infos): encoding: cp1252 default, cp1252 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: version n/a (No package metadata was found for colorlog)
[DEBUG] (cli) (_log_env_infos): with crcmod: version n/a (No package metadata was found for crcmod)
[DEBUG] (cli) (_log_env_infos): with docopt: version n/a (No package metadata was found for docopt)
[DEBUG] (cli) (_log_env_infos): with hidapi: version n/a (No package metadata was found for hidapi)
[DEBUG] (cli) (_log_env_infos): with pyusb: version n/a (No package metadata was found for pyusb)
[DEBUG] (cli) (_log_env_infos): with pillow: version n/a (No package metadata was found for pillow)
[DEBUG] (cli) (_log_env_infos): with smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xffbc usage=0x0088)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
[DEBUG] (usb) (find_devices): USB device: 8086:1d26
[DEBUG] (usb) (find_devices): USB device: 8087:0024
[DEBUG] (usb) (find_devices): USB device: 1b21:1042
[DEBUG] (usb) (find_devices): USB device: 1b21:1042
[DEBUG] (usb) (find_devices): USB device: 8086:1d2d
[DEBUG] (usb) (find_devices): USB device: 046d:c52b
[DEBUG] (usb) (find_devices): USB device: 0b05:179c
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 2109:0810
[DEBUG] (usb) (find_devices): USB device: 8087:0024
[DEBUG] (usb) (find_devices): USB device: 2109:3431
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c13
[DEBUG] (usb) (probe): HydroPro identified: Corsair Hydro H115i Pro
Device #0: Corsair Hydro H115i Pro
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c13
├── Release number: 0x0100
[ERROR] (cli) (_list_devices_human): Could not read the serial number (device possibly requires a kernel driver)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 201, in _list_devices_human
  File "liquidctl\driver\usb.py", line 160, in serial_number
  File "liquidctl\driver\usb.py", line 369, in serial_number
  File "usb\core.py", line 864, in serial_number
  File "usb\util.py", line 311, in get_string
  File "usb\core.py", line 851, in langids
  File "usb\util.py", line 266, in get_langids
  File "usb\control.py", line 172, in get_descriptor
  File "usb\core.py", line 1071, in ctrl_transfer
  File "usb\core.py", line 113, in wrapper
  File "usb\core.py", line 131, in managed_open
  File "usb\backend\libusb1.py", line 804, in open_device
  File "usb\backend\libusb1.py", line 652, in __init__
  File "usb\backend\libusb1.py", line 600, in _check
NotImplementedError: Operation not supported or unimplemented on this platform
├── Bus: usb3
├── Address: 2
├── Port: 1.4
└── Driver: HydroPro
[DEBUG] (cli) (_list_devices_human): MRO: HydroPro, _Base690Lc, UsbDriver, BaseUsbDriver, BaseDriver, object

jonasmalacofilho, 2022-10-29:
That can be caused by not replacing the Windows device driver, which is necessary with that device.

Did you follow the procedure, explained in the README, to replace the driver with WinUSB, using the Zadig tool?

yw4z, 2022-10-29:
i look for readme but it looks a bit complicated at there
if i understand correctly LibUSB bundled with latest versions so i dont have to install python and libusb
i downloaded zadig, is this only thing i have to do

[image]

thanks for your support

jonasmalacofilho, 2022-10-30:
You need to select the correct device, and (only) then click "replace driver". Make sure that new driver is WinUSB.

yw4z, 2022-10-30:
thanks again
all works great
only device visible as removable now, i send a feature request to zadiq about this

few suggestions on liquidctl 
multiple commands on single line would be great eg.
`liquidctl set fan speed 30 25 70 100 set pump mode performance`

i can get pump mode on powershell with
```
$lct="C:\ProgramData\Apps\System\Scripts\CPU\liquidctl.exe"
foreach($v in ($(& $lct status --json)|ConvertFrom-Json).status){if($v.key -eq "Pump mode"){return $v.value}}
```
cleaner way would be great or let me know if there is another way

for other h115i users my presets
Zero RPM
liquidctl initialize --pump-mode=quiet
liquidctl set fan speed 40 0 80 100

Quite
liquidctl initialize --pump-mode=quiet
liquidctl set fan speed 35 5 80 100

Balanced
liquidctl initialize --pump-mode=balanced
liquidctl set fan speed 30 15 75 100

Performance
liquidctl initialize --pump-mode=performance
liquidctl set fan speed 30 25 70 100

jonasmalacofilho, 2022-11-16:
> all works great

Great, I'll close this issue then.

---

> only device visible as removable now, i send a feature request to zadiq about this

That's normal (even if slightly unfortunate).

> multiple commands on single line would be great eg.

That's a bit out of sync with how our CLI is (intentionally) designed. But thanks anyway for the suggestion.

> i can get pump mode on powershell with [...]

I'm not a big PowerShell user. The most I can suggest is to also look at tools like `jq`, maybe they can make things easier for you. Alternatively, perhaps a different shell, like Nushell?

closed 2022-11-16 by jonasmalacofilho