liquidctl/liquidctl #532: OSError('read error') when NZXT CAM is running

nmarinel, 2022-10-31:
The latest version of NZXT CAM (4.40.0) appears to interfere with liquidctl. When CAM is running, I get an error trying to run any command, but liquidctl works fine once I close CAM. The error reads: 
```
OSError: read error
[ERROR] (cli) (log): unexpected OS error with NZXT Kraken X (X53, X63 or X73): OSError('read error')
```

`status` and `initialize` fail with the aforementioned error when CAM is running, but `list` works fine. I haven't tried any others but am happy to test as needed.

```
PS C:\Users\marin> liquidctl.exe status --debug
[DEBUG] (cli) (main): script: C:\Users\marin\AppData\Local\Programs\Python\Python311\Scripts\liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Windows-10-10.0.22621-SP0 /* this isn't right, see version below */
[DEBUG] (cli) (_log_requirements): python: 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.2.0
[DEBUG] (cli) (_log_requirements): smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_requirements): winusbcdc: 1.4
[DEBUG] (cli) (_log_requirements): libusb-package: 1.0.26.0
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0xff01 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a14 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0xff41 usage=0xff41)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xffbc usage=0x0088)
[DEBUG] (usb) (find_devices): HID device: 26ce:01a2 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0xff02 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a14 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1e71:2007 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): KrakenX3 identified: NZXT Kraken X (X53, X63 or X73)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a14 (usage_page=0xffc5 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
[DEBUG] (usb) (find_devices): USB device: 25a7:fa70
[DEBUG] (usb) (find_devices): USB device: 1e71:2007
[DEBUG] (usb) (find_devices): USB device: 174c:2074
[DEBUG] (usb) (find_devices): USB device: 8087:0029
[DEBUG] (usb) (find_devices): USB device: 174c:3074
[DEBUG] (usb) (find_devices): USB device: 046d:c53f
[DEBUG] (usb) (find_devices): USB device: 1022:149c
[DEBUG] (usb) (find_devices): USB device: 1b1c:0a14
[DEBUG] (usb) (find_devices): USB device: 26ce:01a2
[DEBUG] (usb) (find_devices): USB device: 062a:38b3
[DEBUG] (usb) (find_devices): USB device: 1022:43ee
[DEBUG] (cli) (main): device: NZXT Kraken X (X53, X63 or X73)
[INFO] (cli) (log): detailed error: unexpected OS error with NZXT Kraken X (X53, X63 or X73): OSError('read error')
Traceback (most recent call last):
[... 73 lines cut ...]
[DEBUG] (usb) (find_devices): USB device: 1022:43ee
[DEBUG] (cli) (main): device: NZXT Kraken X (X53, X63 or X73)
[INFO] (cli) (log): detailed error: unexpected OS error with NZXT Kraken X (X53, X63 or X73): OSError('read error')
Traceback (most recent call last):
  File "C:\Users\marin\AppData\Local\Programs\Python\Python311\Lib\site-packages\liquidctl\cli.py", line 460, in main
    status = dev.initialize(**opts)
             ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marin\AppData\Local\Programs\Python\Python311\Lib\site-packages\liquidctl\driver\kraken3.py", line 236, in initialize
    self.device.clear_enqueued_reports()
  File "C:\Users\marin\AppData\Local\Programs\Python\Python311\Lib\site-packages\liquidctl\driver\usb.py", line 437, in clear_enqueued_reports
    while self.hiddev.read(max_length=1, timeout_ms=timeout_ms):
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "hid.pyx", line 219, in hid.device.read
OSError: read error
[ERROR] (cli) (log): unexpected OS error with NZXT Kraken X (X53, X63 or X73): OSError('read error')
```

---

Additional information about the system:

- Affected device(s): NZXT Kraken X53 
- Operating system and version: Windows 11 22H2 22621.674 
- Liquidctl installation method: py -m pip install liquidctl 
- Output of `liquidctl --version`: `liquidctl v1.11.1 (Windows-10-10.0.22621-SP0)` Note: it's actually Windows 11.

CriticalPoint, 2023-08-10:
As you've likely worked out by now, the world of RGB and the associated software can be a bit of a nightmare. With that being said, and in the defence of this being a real bug - I wouldn't expect liquidctl to work with CAM running as CAM will hold onto the driver, making the device unavailable.
What you should do is disable the startup of CAM and then reboot, so that you can then go on to access your Kraken X, and apply whatever you need to your device.

nmarinel, 2023-08-10:
Indeed my solution was to uninstall CAM and forget it existed. I think this issue could be closed if there was a friendlier error message or something in the docs that says liquidctl doesn't support running side-by-side with CAM and similar.

PR #622 (2023-08-11): docs: note that CAM must be closed for kraken drivers to work
Inspired by the linked issue. Add a note to Kraken docs that CAM can't be running if you want liquidctl to work.

Fixes: #532
closed 2023-08-11 by jonasmalacofilho