liquidctl/liquidctl #533: "usb.core.NoBackendError: No backend available" when installing on Python 3.11

nmarinel, 2022-10-31:
There seems to be an issue with the libusb package not properly adding its dll when installing liquidctl with pip on Python 3.11. As discussed in Discord, this is probably because libusb-package hasn't been updated yet with Python 3.11 wheels. So not really a liquidctl bug but maybe worth documenting here if indeed this is the problem. 

pip install log: [pip_log.txt](https://github.com/liquidctl/liquidctl/files/9897385/pip_log.txt)
``` bash
marin@Barnacle_Boy MINGW64 ~/OneDrive/Desktop/pythonVenv/Scripts
$ ./liquidctl.exe status --debug
[DEBUG] (cli) (main): script: C:\Users\marin\OneDrive\Desktop\pythonVenv\Scripts\liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Windows-10-10.0.22621-SP0
[DEBUG] (cli) (_log_requirements): python: 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.3.0
[DEBUG] (cli) (_log_requirements): smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_requirements): winusbcdc: 1.4
[DEBUG] (cli) (_log_requirements): libusb-package: 1.0.26.0
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0xff01 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a14 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0xff41 usage=0xff41)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xffbc usage=0x0088)
[DEBUG] (usb) (find_devices): HID device: 26ce:01a2 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 062a:38b3 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0xff02 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a14 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1e71:2007 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): KrakenX3 identified: NZXT Kraken X (X53, X63 or X73)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa70 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a14 (usage_page=0xffc5 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Scripts\liquidctl.exe\__main__.py", line 7, in <module>
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Lib\site-packages\liquidctl\cli.py", line 417, in main
    selected = list(find_liquidctl_devices(**opts))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Lib\site-packages\liquidctl\driver\__init__.py", line 61, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Lib\site-packages\liquidctl\driver\usb.py", line 615, in find_devices
    for handle in PyUsbDevice.enumerate(vendor, product):
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Lib\site-packages\liquidctl\driver\usb.py", line 352, in enumerate
    for handle in find(find_all=True, **args):
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Lib\site-packages\libusb_package\__init__.py", line 108, in find
    return usb.core.find(*args, backend=backend, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marin\OneDrive\Desktop\pythonVenv\Lib\site-packages\usb\core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

nmarinel, 2022-10-31:
And if anybody comes across this looking for a solution, it is to copy the [libusb dll](https://github.com/libusb/libusb/releases) to your C:\Windows\System32

jonasmalacofilho, 2022-12-19:
libusb-package wheels for Python 3.11 are now available.

closed 2022-12-19 by jonasmalacofilho