liquidctl/liquidctl #536: Corsair Hydro H115i Pro failed to get fan speed

jnuernberg, 2022-11-21:
When requesting the status of my H115i Pro, liquidctl fails to read the fan speed of the two attached fans. However setting the pump profile, fan speeds, even led colors works perfectly fine. I am running a Dual-Boot setup (Linux Mint/Windows) and in iCue under Windows both fan speeds are read perfectly. Thus I exclude a hardware issue as the cause. 

Here is the result of the status command; I would expect to read fan speeds and potentially duty cycles in the output.
```
jacob@Linux-Desktop ~ % sudo liquidctl --match Hydro status
WARNING: failed to get current speed of fan 0
WARNING: failed to get current speed of fan 1
Corsair Hydro H115i Pro (experimental)
├── Liquid temperature       30.7  °C
├── Pump mode               quiet  
├── Pump speed               1020  rpm
└── Firmware version      1.0.4.0  
```

I understand, that support for the H115i Pro is experimental. But I would like to track the pump and fan speeds during some rendering tasks. Having that output would be amazing. 

I appreciate any help. 

Best regards, Jacob

---

Here is the output of the same command with the debug flag.
```
jacob@Linux-Desktop ~ % sudo liquidctl --match Hydro --debug status
[DEBUG] liquidctl.cli: running liquidctl v1.6.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: ignoring 0-0008, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:4074
[DEBUG] liquidctl.driver.usb: found HID device 046d:4074
[DEBUG] liquidctl.driver.usb: found HID device 046d:4074
[DEBUG] liquidctl.driver.usb: found HID device 046d:4074
[DEBUG] liquidctl.driver.usb: found HID device 046d:4074
[DEBUG] liquidctl.driver.usb: found HID device 046d:4074
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 046d:c53f
[DEBUG] liquidctl.driver.usb: found HID device 0451:82ff
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1c0b
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver, CorsairAsetekProDriver)
[DEBUG] liquidctl.driver.usb: found USB device 174c:3074
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 0a12:0001
[DEBUG] liquidctl.driver.usb: found USB device 0424:2514
[DEBUG] liquidctl.driver.usb: found USB device 174c:2074
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 0451:8440
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 26ce:01a2
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c13
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair Hydro H115i Pro (experimental)
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:1c0b
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 0451:82ee
[DEBUG] liquidctl.driver.usb: found USB device 0451:82ff
[DEBUG] liquidctl.driver.usb: found USB device 046d:c53f
[DEBUG] liquidctl.driver.usb: found USB device 0451:8442
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair Hydro H115i Pro (experimental)
[DEBUG] liquidctl.driver.usb: selected interface: 0
[DEBUG] liquidctl.driver.asetek: set clear to send = True
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 2), {}
[DEBUG] liquidctl.driver.usb: writting 1 bytes: a9
[DEBUG] liquidctl.driver.usb: read 5 bytes: a9:00:02:21:01
[DEBUG] liquidctl.driver.usb: writting 2 bytes: 41:00
[DEBUG] liquidctl.driver.usb: read 6 bytes: 41:00:02:00:00:95
[WARNING] liquidctl.driver.asetek_pro: failed to get current speed of fan 0
[DEBUG] liquidctl.driver.usb: writting 2 bytes: 41:01
[DEBUG] liquidctl.driver.usb: read 6 bytes: 41:00:02:01:00:9e
[WARNING] liquidctl.driver.asetek_pro: failed to get current speed of fan 1
[DEBUG] liquidctl.driver.usb: writting 1 bytes: 33
[DEBUG] liquidctl.driver.usb: read 4 bytes: 33:00:02:00
[DEBUG] liquidctl.driver.usb: writting 1 bytes: 31
[DEBUG] liquidctl.driver.usb: read 5 bytes: 31:00:02:04:1a
[DEBUG] liquidctl.driver.usb: writting 1 bytes: aa
[DEBUG] liquidctl.driver.usb: read 7 bytes: aa:00:02:01:00:04:00
[DEBUG] liquidctl.driver.usb: explicitly release interface
Corsair Hydro H115i Pro (experimental)
├── Liquid temperature       33.1  °C
├── Pump mode               quiet  
├── Pump speed               1050  rpm
└── Firmware version      1.0.4.0  

[DEBUG] liquidctl.driver.usb: explicitly release interface

```

---

Additional information about the system:

- Affected device(s): Corsair Hydro H115i Pro
- Operating system and version: Linux Minx 20.3 (Una)
- Liquidctl installation method: Pip, i.e. `pip install liquidctl`
- Output of `liquidctl --version`: `liquidctl v1.11.1 (Linux-5.4.0-132-generic-x86_64-with-glibc2.29)`

jonasmalacofilho, 2022-11-21:
Would 149 rpm and 158 rpm make sense for the fans speeds during the second execution (where: liquid @ 33.1°C, pump @ quiet/1050 rpm)?

jnuernberg, 2022-11-21:
Hi @jonasmalacofilho,

Yes, it makes an awefull lot of sense.
Based on your response, I started a small experiment: In both, Linux and Windows, I set the fan speed to a fixed duty cycle and looked at the fan speeds. In Windows, I simply used iCue for that. In linux, I set the speeds using `liquidctl set fan1 speed 25` and observed the readback lines in the debug log of `liquidctl --debug status` converting the last bytes from hex to decimal, i.e. these lines
```
[DEBUG] liquidctl.driver.usb: writting 2 bytes: 41:00
[DEBUG] liquidctl.driver.usb: read 6 bytes: 41:00:02:00:00:95
[WARNING] liquidctl.driver.asetek_pro: failed to get current speed of fan 0
[DEBUG] liquidctl.driver.usb: writting 2 bytes: 41:01
[DEBUG] liquidctl.driver.usb: read 6 bytes: 41:00:02:01:00:9e
[WARNING] liquidctl.driver.asetek_pro: failed to get current speed of fan 1
```

And yes, these values are identical for the speeds in iCue and liquidctl. I have seen your commit. That reduced expectation should fix it I and may get around testing it soon. 

Cheers and Thanks, 

Jacob

P.S.: These were the (rought) fan speed readings (on both, linux and windows)

| Duty Cycle | Fan 1 | Fan 2 |
| --------------: | --------| --------|
|  25 %  | 240 | 240 |
|  50 %  | 600 | 600 |
|  75 %  | 880  | 880 |
|  100 %  | 1140 | 1140 |

jonasmalacofilho, 2022-11-21:
Glad to hear that, thanks!

And when you can, please do test the `jonas-staging-area` branch, as there are a couple of other (unrelated) changes to the asetek_pro driver where it would be nice to have some real world testing.

jnuernberg, 2022-11-22:
Hi Jonas,

So, I got around testing your branch. Here are the results:

```
jacob@Linux-Desktop ~/liquidctl
 % python -m liquidctl --match Hydro initialize --pump-mode quiet
Corsair Hydro H115i Pro
└── Firmware version    1.0.4.0  

jacob@Linux-Desktop ~//liquidctl
 % python -m liquidctl --match Hydro status                      
Corsair Hydro H115i Pro
├── Liquid temperature     39.1  °C
├── Fan 1 speed             425  rpm
├── Fan 2 speed             429  rpm
├── Pump mode             quiet  
└── Pump speed             2070  rpm
```

As you can see, it is running just fine. Would you like me to pay some attention to certain things?
Otherwise, we can close this issue and Ill wait untill this branch is merged into the main branch. Until then, Ill use a manual install.

Thanks a lot for your support. 

Jacob

jonasmalacofilho, 2022-11-23:
Yeah, everything looks ok from here too.

Thanks!

closed 2022-11-23 by jonasmalacofilho

jonasmalacofilho, 2022-11-23:
Fixed it, thanks!

commit ab044281 (2022-11-23): asetek_pro: reduce expectations on _CMD_READ_FAN_SPEED responses

Fixes: #536