liquidctl/liquidctl #538: Corsair Commander Pro no firmware without iCue running

xuuk3r, 2022-11-24:
Hi, I'm trying to get my system to run without Corsair's iCue bloatware. When trying to set up Liquidctl without running iCue however, my Commander Pro seems to lose its firmware driver and I'm no longer able to detect or configure the connected fans through Liquidctl. The problem only occurs while running Liquidctl after a fresh boot without iCue loaded, once I open iCue and force update the Commander, Liquidctl recognizes the firmware and I'm again able to control it. 
The reason I don't want iCue running is because I also run OpenRGB and these conflict together. 

I've been searching for similar issues but can't find any. I'm no developer whatsoever and I don't really know if this has anything to do with Liquidctl, but I'm fairly certain I won't get any help from Corsair with this so I thought I'd start here. Maybe some of you know anything more :-) 

**Now list which commands were executed, what was the expected behavior, and what was observed instead.**
```
liquidctl list 
Device #0: Corsair Commander Pro
Device #1: Corsair Hydro H100i v2

liquidctl status 
Corsair Commander Pro
├── Temperature 2      0.0  °C
├── +12V rail         0.00  V
├── +5V rail         11.98  V
└── +3.3V rail        4.92  V

Corsair Hydro H100i v2
├── Liquid temperature       28.0  °C
├── Fan speed                 840  rpm
├── Pump speed               1920  rpm
└── Firmware version      2.8.0.0

liquidctl -m commander initialize 
Corsair Commander Pro
├── Firmware version       0.0.0
├── Bootloader version       0.9
├── Temperature probe 1       No
├── Temperature probe 2      Yes
├── Temperature probe 3       No
├── Temperature probe 4       No
├── Fan 1 control mode       N/A
├── Fan 2 control mode       N/A
├── Fan 3 control mode       N/A
├── Fan 4 control mode       N/A
├── Fan 5 control mode       N/A
└── Fan 6 control mode       N/A
```
As you can see, Liquidctl detects the Commander Pro, but it has no firmware and the fans are not recognized. 

**Re-run all commands with an additional --debug flag, and include the complete output.  Use code blocks, delimited by lines with three backticks (```), to the show program invocation and output.**

```
C:\Users\frede>liquidctl list -g
[DEBUG] (cli) (main): script: C:\Users\frede\AppData\Local\Programs\Python\Python311\Scripts\liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Windows-10-10.0.19044-SP0
[DEBUG] (cli) (_log_requirements): python: 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.3.0
[DEBUG] (cli) (_log_requirements): smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_requirements): winusbcdc: 1.5
[DEBUG] (cli) (_log_requirements): libusb-package: 1.0.26.0
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:c07e (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:1227 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c07e (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1baf (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c07e (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1baf (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c10 (usage_page=0x0084 usage=0x0052)
[DEBUG] (usb) (probe): CommanderPro identified: Corsair Commander Pro
[DEBUG] (usb) (find_devices): HID device: 1b1c:1baf (usage_page=0x0001 usage=0x0006)
[... 186 lines cut ...]
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 20:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): wrote 64 total bytes, expected 65
[DEBUG] (usb) (read): read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (keyval) (store): stored fan_modes=b'\x00\x00\x00\x00\x00\x00' (in C:\Users\frede\AppData\Local\Temp\liquidctl\vid1b1c_pid0c10\loc1_1_0_10_6_1978198_0_0000_4_1_55_2_16_11_88_001111000030\fan_modes)
Corsair Commander Pro
├── Firmware version       0.0.0
├── Bootloader version       0.9
├── Temperature probe 1       No
├── Temperature probe 2      Yes
├── Temperature probe 3       No
├── Temperature probe 4       No
├── Fan 1 control mode       N/A
├── Fan 2 control mode       N/A
├── Fan 3 control mode       N/A
├── Fan 4 control mode       N/A
├── Fan 5 control mode       N/A
└── Fan 6 control mode       N/A
```

Additional information about the system:

- Affected device(s):  Corsair Commander Pro
- Operating system and version:  Windows 10
- Liquidctl installation method:  Python -m pip install Liquidctl
- Output of `liquidctl --version`:  liquidctl v1.11.1 (Windows-10-10.0.19044-SP0)

jonasmalacofilho, 2022-11-24:
Hey @MarshallAsch, can you take a look at this one?

MarshallAsch, 2022-11-24:
What version of icue are you using and what firmware version does it report the commander pro as running?

xuuk3r, 2022-11-24:
> 

I'm running 4.30.162, latest normally. Same for the Commander Pro, when running iCue the firmware is 0.9.212.
I just don't understand that even when forcing firmware update in iCue, after a reboot it's just gone until I launch iCue and force update again..