liquidctl/liquidctl #539: H100i RGB Pro XT no longer listed after kernel update

jareddr, 2022-11-27:
Hello,

Thank you for making this great software!  I've been using it for a couple months and it's been working perfectly.

Recently I did a system update on EndeavorOS which updated my kernel to version `6.0.9-arch1-1`  and after rebooting my cooler was no longer listed.
```
❯ liquidctl --version
liquidctl v1.11.1 (Linux-6.0.9-arch1-1-x86_64-with-glibc2.36)
❯ liquidctl list
❯ liquidctl status
ERROR: no device matches available drivers and selection criteria
```

I read through the 'device not listed' trouble shooting section and I don't see any references to the H100i in the lsusb command.

```
❯ lsusb -vt
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Audio, Driver=snd-usb-audio, 480M
        ID 30be:0106  
    |__ Port 1: Dev 2, If 1, Class=Audio, Driver=snd-usb-audio, 480M
        ID 30be:0106  
    |__ Port 1: Dev 2, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        ID 30be:0106  
    |__ Port 1: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 480M
        ID 30be:0106  
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 308f:0013 Input Club Kira
    |__ Port 1: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 308f:0013 Input Club Kira
    |__ Port 1: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        ID 308f:0013 Input Club Kira
    |__ Port 1: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 12M
        ID 308f:0013 Input Club Kira
    |__ Port 1: Dev 2, If 4, Class=Human Interface Device, Driver=usbhid, 12M
        ID 308f:0013 Input Club Kira
    |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c06b Logitech, Inc. G700 Wireless Gaming Mouse
    |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c06b Logitech, Inc. G700 Wireless Gaming Mouse
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
        |__ Port 5: Dev 3, If 0, Class=Audio, Driver=snd-usb-audio, 480M
            ID 046d:0821 Logitech, Inc. HD Webcam C910
        |__ Port 5: Dev 3, If 1, Class=Audio, Driver=snd-usb-audio, 480M
            ID 046d:0821 Logitech, Inc. HD Webcam C910
        |__ Port 5: Dev 3, If 2, Class=Video, Driver=uvcvideo, 480M
            ID 046d:0821 Logitech, Inc. HD Webcam C910
        |__ Port 5: Dev 3, If 3, Class=Video, Driver=uvcvideo, 480M
            ID 046d:0821 Logitech, Inc. HD Webcam C910
        |__ Port 6: Dev 4, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
            ID 1a86:7523 QinHeng Electronics CH340 serial converter
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
        |__ Port 1: Dev 3, If 2, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
        |__ Port 1: Dev 3, If 3, Class=Application Specific Interface, Driver=, 12M
            ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
```

This seems more like an issue with my system not detecting my cooler, but I haven't been able to think of the correct search terms to uncover anything useful.  Is there anything I can try or check to get back up and running?

Thanks much!

jonasmalacofilho, 2022-11-27:
> [...] after rebooting I get a systemd error and my cooler was no longer listed. [...]

Is the error in that system log from Oct 22 what you meant by a "systemd error"?

And can you confirm the device model and, if possible, SKU number (something that looks like `CW-0000000-XX`)?
(The log mentions a H100i Platinum SE, not Pro XT).

---

> This seems more like an issue with my system not detecting my cooler, but I haven't been able to think of the correct search terms to uncover anything useful. Is there anything I can try or check to get back up and running?

The `lsusb` output suggests that the device is no longer visible to the kernel.

I would check the kernel log (with `dmesg` or `journalctl -k`) to see if the device was detected at all.

And if there's not mention of it, then you might need to do the same for whatever hub (or root hub) it's connected to, as that may have stopped working after the kernel update.

You may also want to try different ports.

jareddr, 2022-11-27:
> > [...] after rebooting I get a systemd error and my cooler was no longer listed. [...]
> 
> Is the error in that system log from Oct 22 what you meant by a "systemd error"?

Yeah, I see it print out while the machine is booting up.  By 'systemd error' i meant that the liquidctl systemd startup script was displaying the error.  Created systemd script following these instructions https://github.com/liquidctl/liquidctl#set-up-linux-using-systemd

> And can you confirm the device model and, if possible, SKU number (something that looks like `CW-0000000-XX`)? (The log mentions a H100i Platinum SE, not Pro XT).

Hmmm yeah that's weird.  I found the h100i pro XT in my purchase history / email but liquidctl automatically identified it as the h100i platinum when everything was working.

> The `lsusb` output suggests that the device is no longer visible to the kernel.
>
> I would check the kernel log (with `dmesg` or `journalctl -k`) to see if the device was detected at all.

I grepped dmesg and journalctl for any mention of 100i or corsair and didn't find anything.

> And if there's not mention of it, then you might need to do the same for whatever hub (or root hub) it's connected to, as that may have stopped working after the kernel update.
> 
> You may also want to try different ports.

Ok thanks for the tips, I'll start digging into the USB hub.  I'm worried this is going to be a pain in the butt as my motherboard is pretty old and not super common (asus sabertooth x79).

Cheers for taking the time to respond!

jonasmalacofilho, 2022-11-27:
> Yeah, I see it print out while the machine is booting up. By 'systemd error' i meant that the liquidctl systemd startup script was displaying the error. Created systemd script following these instructions https://github.com/liquidctl/liquidctl#set-up-linux-using-systemd

But at that point liquidctl was able to detect the cooler; the error is related to a invalid channel specification.

That may suggest that the kernel is able to detect the device, but is dropping it after a while. I've seen it happen before with a faulty physical connection. Maybe the new kernel is a bit more strict regarding physical layer errors.

---

> Hmmm yeah that's weird. I found the h100i pro XT in my purchase history / email but liquidctl automatically identified it as the h100i platinum when everything was working.

It's possible we got the name wrong, or that Corsair is reusing the VID:PID pair. 

But, also, are the included fans RGB?

---

> I grepped dmesg and journalctl for any mention of 100i or corsair and didn't find anything.

Try also grepping for the VID (1b1c) and PID (0c19), ignoring case.

---

Can you show us the output of `liquidctl --version` and `liquidctl list --debug`?

jareddr, 2022-11-28:
> It's possible we got the name wrong, or that Corsair is reusing the VID:PID pair.
> 
> But, also, are the included fans RGB?

yes to RGB fans and liquidctl was able to change set the color correctly
 
> > I grepped dmesg and journalctl for any mention of 100i or corsair and didn't find anything.
> 
> Try also grepping for the VID (1b1c) and PID (0c19), ignoring case.

```
❯ sudo dmesg | grep -i 1b1c
❯ sudo dmesg | grep -i 0c19
❯ journalctl -k | grep -i 1b1c
❯ journalctl -k | grep -i 0c19
```

No results for any of those.

> Can you also link (or email me) the complete kernel log

Here is the full output of `dmesg` https://pastebin.com/6CHXJkXE

> Can you show us the output of `liquidctl --version` and `liquidctl list --debug`?

```
❯ liquidctl --version
liquidctl v1.11.1 (Linux-6.0.9-arch1-1-x86_64-with-glibc2.36)
❯ liquidctl list --debug
[DEBUG] (cli) (main): script: /usr/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Linux-6.0.9-arch1-1-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_requirements): python: 3.10.8 (main, Nov  1 2022, 14:18:21) [GCC 12.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.3.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 308f:0013 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 308f:0013 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 308f:0013 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 308f:0013 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 308f:0013 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 308f:0013 (usage_page=0xff1c usage=0x1100)
[DEBUG] (usb) (find_devices): HID device: 046d:c06b (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c06b (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c06b (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c06b (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c06b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c06b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 30be:0106 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1a86:7523
[DEBUG] (usb) (find_devices): USB device: 046d:0821
[DEBUG] (usb) (find_devices): USB device: 8087:0024
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 30be:0106
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 046d:c06b
[DEBUG] (usb) (find_devices): USB device: 308f:0013
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 0a5c:21e8
[DEBUG] (usb) (find_devices): USB device: 8087:0024
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
```

jonasmalacofilho, 2022-11-28:
> Here is the full output of `dmesg` https://pastebin.com/6CHXJkXE

I think this one is the cooler:

```
$ grep -n 1-1.4 6CHXJkXE.txt
751:[    1.519251] usb 1-1.4: new full-speed USB device number 4 using ehci-pci
752:[    1.592584] usb 1-1.4: device descriptor read/64, error -32
759:[    1.772582] usb 1-1.4: device descriptor read/64, error -32
788:[    1.955917] usb 1-1.4: new full-speed USB device number 5 using ehci-pci
819:[    2.029251] usb 1-1.4: device descriptor read/64, error -32
831:[    2.209279] usb 1-1.4: device descriptor read/64, error -32
867:[    2.912620] usb 1-1.4: new full-speed USB device number 6 using ehci-pci
868:[    3.325943] usb 1-1.4: device not accepting address 6, error -32
869:[    3.399278] usb 1-1.4: new full-speed USB device number 7 using ehci-pci
870:[    3.812611] usb 1-1.4: device not accepting address 7, error -32
```

Try re-seating all physical connections (USB and power) to the cooler, on both ends.

Also try a different internal port, if you have one to spare (at least for a test). At the very least it would allow you to confirm that is indeed the right device.

And if none of it fixes the issue, then it's probably worth it to report it as a kernel bug: the device used to work, and it doesn't anymore. Just re-check with an older kernel first, to make sure that by some coincidence the device hasn't just died.

jareddr, 2022-11-28:
Thanks, I'll try all of this tomorrow.  

With regard to the cooler 'being dead', that's the weird thing--it's on and running its on-device profile.  The lights cycle through a range of colors and it does seem to have some fan curves active as it gets louder if I load a 3d app.

I'll boot into windows 10 and see if it's addressable via the corsair software as well.

jareddr, 2022-11-28:
Aha.  Thank you so much for all your help and sorry to bother you with an unrelated issue.

By some coincidence the USB connection to my cooler seems to have died.  

The cooler is not present in windows10 either where it also used to work.

I also tried plugging it in to a different USB header on my motherboard and it still didn't work.  

Also tried reseating the USB connection on the cooler itself, no luck.

jonasmalacofilho, 2022-11-29:
Thanks for the update, and sorry to hear it's the cooler.

closed 2022-11-29 by jonasmalacofilho