liquidctl/liquidctl #541: New NZXT RGB & Fan Controller variant

feiticeir0, 2022-11-30:
### Device type

NZXT NZXT RGB & Fan Controller

### Product page

https://nzxt.com/product/rgb-and-fan-controller

### First-party software

NZXT

### What monitoring functionality does the device support?

Fan/pump speeds or duty cycles

### What configuration functionality does the device support?

Lighting of accessories like RGB fans or LED strips

### System connection

USB

### System protocol

USB

### Additional information about the device

`Bus 001 Device 005: ID 1e71:2011 NZXT NZXT RGB & Fan Controller`

liquidctl does not detect this device. 

I'm using Arch Linux with kernel 6.0.10-arch2-1
I have the module loaded nzxt-smart2 if it helps... 

Thank you

My case is a NZXT H7 Elite .
Thank you
[lsusb-devices.txt](https://github.com/liquidctl/liquidctl/files/10118458/lsusb-devices.txt)

### Can you help with the implementation or testing of the new device?

testing changes on Linux, attempting to capture I²C/SMBus traffic

jonasmalacofilho, 2022-11-30:
Thanks for reporting it.

Still 3 fan channels and 2 lighting channels?

feiticeir0, 2022-11-30:
> Thanks for reporting it.
> 
> Still 3 fan channels and 2 lighting channels?

Hi @jonasmalacofilho . Thank you for the reply
No, It has 4 FANS and 6 RGB channels. 
Here's a photo
[image]

jonasmalacofilho, 2022-11-30:
I can only see 3 fan channels in that picture, what am I missing?

feiticeir0, 2022-11-30:
> I can only see 3 fan channels in that picture, what am I missing?

I'm sorry. You're right. I've counted the fans. 🤦

jonasmalacofilho, 2022-11-30:
Can you test the code in [9-channel-nzxt-controller](https://github.com/liquidctl/liquidctl/tree/9-channel-nzxt-controller), especially the new LED channels (3–6)?

feiticeir0, 2022-11-30:
> Can you test the code in [9-channel-nzxt-controller](https://github.com/liquidctl/liquidctl/tree/9-channel-nzxt-controller), especially the new LED channels (3–6)?

Yes ! I will. Let me just get home and will try it ! 
Thank you

feiticeir0, 2022-11-30:
I've tried and no luck... 
It only detects the Asus AURA
Here's the screenshot
[image]

jonasmalacofilho, 2022-12-01:
Can you show me the output from

```
$ python -m liquidctl list --debug
```

jonasmalacofilho, 2022-12-01:
Oh, nevermind, the output from `liquidctl --version` shows that you're running the code from the wrong branch: you're on main @ 45901948de70, instead of 9-channel-nzxt-controller.

feiticeir0, 2022-12-01:
Now I'm controlling. 
The FANs I can control them - speeds . Tried all the 3 channels and all worked.  :+1: 

LEDs don't work...

Here's the initialize all:

`python -m liquidctl initialize all
ASUS Aura LED Controller (experimental)
└── Firmware version    AULA3-AR42-0207  

NZXT RGB & Fan Controller (3+6 channels)
├── Firmware version       1.7.0  
├── LED 1 accessory 1    Unknown  
├── LED 2 accessory 1    Unknown  
├── LED 3 accessory 1    Unknown  
├── LED 4 accessory 1    Unknown  
├── LED 5 accessory 1    Unknown  
└── LED 6 accessory 1    Unknown  
`

and debug:
`python -m liquidctl list --debug
[DEBUG] (cli) (_log_env_infos): script: /home/feiticeir0/Downloads/builds/liquidctl/liquidctl/__main__.py
[DEBUG] (cli) (_log_env_infos): version: 1.12.0.dev52+g15efe4d
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.0.10-gentoo-x86_64-AMD_Ryzen_7_5700G_with_Radeon_Graphics-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.10.8 (main, Nov 30 2022, 22:28:51) [GCC 11.3.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): not installed, package metadata not available
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b05:1939 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (probe): AuraLed identified: ASUS Aura LED Controller (experimental)
[DEBUG] (usb) (find_devices): HID device: 1e71:2011 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): SmartDevice2 identified: NZXT RGB & Fan Controller (3+6 channels)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0xff00 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0xff01 usage=0x0001)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 413c:4503
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 046d:08e5
[DEBUG] (usb) (find_devices): USB device: 1e71:2011
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0b05:1939
[DEBUG] (usb) (find_devices): USB device: 0b05:190e
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
Device #0: ASUS Aura LED Controller (experimental)
├── Vendor ID: 0x0b05
├── Product ID: 0x1939
├── Release number: 0x0100
├── Serial number: 9876543210
├── Bus: hid
├── Address: /dev/hidraw3
└── Driver: AuraLed
[DEBUG] (cli) (_list_devices_human): MRO: AuraLed, UsbHidDriver, BaseUsbDriver, BaseDriver, object

Device #1: NZXT RGB & Fan Controller (3+6 channels)
├── Vendor ID: 0x1e71
├── Product ID: 0x2011
├── Release number: 0x0107
├── Serial number: 0A038012AB7664C15F049B40
├── Bus: hid
├── Address: /dev/hidraw4
└── Driver: SmartDevice2
[DEBUG] (cli) (_list_devices_human): MRO: SmartDevice2, _BaseSmartDevice, UsbHidDriver, BaseUsbDriver, BaseDriver, object`

Thank you

jonasmalacofilho, 2022-12-01:
Can you now show me the output from both commands bellow?

```
$ python -m liquidctl --match nzxt initialize --debug
$ python -m liquidctl --match nzxt set led6 color fixed abcdef --debug
```

feiticeir0, 2022-12-01:
`python -m liquidctl --match nzxt initialize --debug
[DEBUG] (cli) (_log_env_infos): script: /home/feiticeir0/Downloads/builds/liquidctl/liquidctl/__main__.py
[DEBUG] (cli) (_log_env_infos): version: 1.12.0.dev52+g15efe4d
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.0.10-gentoo-x86_64-AMD_Ryzen_7_5700G_with_Radeon_Graphics-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.10.8 (main, Nov 30 2022, 22:28:51) [GCC 11.3.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): not installed, package metadata not available
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b05:1939 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): HID device: 1e71:2011 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): SmartDevice2 identified: NZXT RGB & Fan Controller (3+6 channels)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0xff00 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 413c:4503 (usage_page=0xff01 usage=0x0001)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 413c:4503
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 04e8:61f5
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 046d:08e5
[DEBUG] (usb) (find_devices): USB device: 1e71:2011
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0b05:1939
[DEBUG] (usb) (find_devices): USB device: 0b05:190e
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT RGB & Fan Controller (3+6 channels)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writing report 0x60 with 63 bytes: 02:01:e8:01:01:e8:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writing report 0x60 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writing report 0x10 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writing report 0x20 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 11:01:12:80:03:0a:c1:64:76:ab:40:9b:04:5f:11:20:01:01:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 21:03:12:80:03:0a:c1:64:76:ab:40:9b:04:5f:06:14:00:00:00:00:00:14:00:00:00:00:00:14:00:00:00:00:00:13:00:00:00:00:00:13:00:00:00:00:00:13:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT RGB & Fan Controller (3+6 channels)
├── Firmware version       1.7.0  
├── LED 1 accessory 1    Unknown  
├── LED 2 accessory 1    Unknown  
├── LED 3 accessory 1    Unknown  
├── LED 4 accessory 1    Unknown  
├── LED 5 accessory 1    Unknown  
└── LED 6 accessory 1    Unknown`

and 

`python -m liquidctl --match nzxt set led6 color fixed abcdef --debug
[DEBUG] (cli) (_log_env_infos): script: /home/feiticeir0/Downloads/builds/liquidctl/liquidctl/__main__.py
[DEBUG] (cli) (_log_env_infos): version: 1.12.0.dev52+g15efe4d
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.0.10-gentoo-x86_64-AMD_Ryzen_7_5700G_with_Radeon_Graphics-with-glibc2.36
[... cut ...]
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0b05:1939
[DEBUG] (usb) (find_devices): USB device: 0b05:190e
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT RGB & Fan Controller (3+6 channels)
[DEBUG] (usb) (write): writing report 0x28 with 63 bytes: 03:20:00:00:02:00:00:01:00:cd:ab:ef:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00`

jonasmalacofilho, 2022-12-01:
It seems that you have some new accessory IDs: 0x14 and 0x13. Maybe NZXT changed them for some reason... Can you tell me what type of accessory is connected to each physical LED port?

---

Did the `set color` command change anything visually on the respective accessory?  Can you try other colors and modes, and report if/which combinations produce some visual change?

(No need for the full `--debug` logs, as they won't show anything useful at this point).

feiticeir0, 2022-12-01:
> It seems that you have some new accessory IDs: 0x14 and 0x13. Maybe NZXT changed them for some reason... Can you tell me what type of accessory is connected to each physical LED port?
> 
To each LED port is a RGB fan connected, from NZXT . F-Series, like this ones:
https://nzxt.com/product/f120-rgb

> Did the `set color` command change anything visually on the respective accessory? Can you try other colors and modes, and report if/which combinations produce some visual change?
> The set color didn't change anything...

> (No need for the full `--debug` logs, as they won't show anything useful at this point).
> 
> P.S. `abcdef` wasn't the best choice of test color on my part. While it allows to me to check if the colors are being correctly encoded, visually it might a bit close to white, which IIRC is also the default color used by NZXT accessories.

this command:
python -m liquidctl --match NZXT set led4 color super-fixed ffffff 
changed ONE led from fan4 to white... (or the color specified or the fan/led specified. all worked)

This one: 
python -m liquidctl --match NZXT set led5 color super-fixed ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff 

change all the 18 LEDS from the LED5 (fan5) to white...

This one also works - breathing from 000000 to ffffff 
python -m liquidctl --match NZXT set led6 color super-breathing ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff ffffff

all the others don't work... :+1:

jonasmalacofilho, 2022-12-01:
> To each LED port is a RGB fan connected, from NZXT . F-Series, like this ones:
> https://nzxt.com/product/f120-rgb

But are all 6 completely identical? Maybe there's a difference in number of LEDs or something, or even just in complete model number?

I ask because the IDs appear to be different and, presumably, that was necessary for some reason.

---

> all the others don't work... +1

Oh, that's unfortunate.

Any chance you can work with us in capturing the CAM<->device traffic for each lighting mode and then figure out what needs to change in our driver and docs?

It's generally not very hard to do, but it's a bit of work, and it can be tedious to try all combinations...

feiticeir0, 2022-12-01:
Hi @jonasmalacofilho ! Thank you for all the replies and your trouble. 

> > To each LED port is a RGB fan connected, from NZXT . F-Series, like this ones:
> > https://nzxt.com/product/f120-rgb
> 
> But are all 6 completely identical? Maybe there's a difference in number of LEDs or something, or even just in complete model number?
They are all identical ! The case comes with 3x 140mm RGB fans on the front and a 140mm fan in the back of the case NOT RGB- but I've changed it to a RGB - but it's not connected - no room in the controller. 

I've bought 3x 120mm RGB fans from NZXT - the ones in the link from previous posts that are mounted on the top of the case and are in channels 4 to 6 - the 140mm on the front are in channels 1 to 3 . 
> I ask because the IDs appear to be different and, presumably, that was necessary for some reason.
>
>
> Oh, that's unfortunate.
>
> Any chance you can work with us in capturing the CAM<->device traffic for [...]

Yes please ! Anything I can do to help.  Just point me to the DOCS to try all the combinations and how do you want me to capture the traffic. I'm curious.  
I've attached some photos of the case with the fans. 

[image]
[image]

Thank you

jonasmalacofilho, 2022-12-01:
> They are all identical ! The case comes with 3x 140mm RGB fans on the front and a 140mm fan in the back of the case NOT RGB- but I've changed it to a RGB - but it's not connected - no room in the [...]

I think the there being both 120 mm and 140 mm fans explains the different IDs. While this doesn't seem important if they have the same number of LEDs, is probably needed for the CAM UI.

I'll update the driver to recognize these new IDs in `status`.

---

> Yes please ! Anything I can do to help. Just point me to the DOCS to try all the combinations and how do you want me to capture the traffic. I'm curious.

Great, thanks!

You can probably start by taking a look at our developer docs: https://github.com/liquidctl/liquidctl/tree/main/docs#documentation-for-developers. There are documents specifically on traffic capture and protocol analysis.

Also, try to familiarize yourself with the `smart_device.py` module, especially the `SmartDevice2` class.

feiticeir0, 2022-12-01:
Do you want me to capture it using windows or Linux ?  I'm referring to windows because since the CAM software works on windows you may want to capture the traffic working.... :)

jonasmalacofilho, 2022-12-02:
Personally, I prefer to capture the traffic on Linux, passing the USB device through to a Windows VM where the vendor-provided tool is installed/running.

feiticeir0, 2022-12-03:
Here's the traffic captured from wireshark while messing around with CAM on the Windows VM. 
[traffic.zip](https://github.com/liquidctl/liquidctl/files/10146123/traffic.zip)

I'm hopping I didn't get the device wrong :)

ultamatum, 2022-12-12:
Hi,

Is there any update on when this will be working as I would like to control the fan colours on my new pc.

Thanks

eggznog, 2022-12-12:
I managed to get a temporary solution. I installed CAM in a Windows VM, and passed through the fan controller. This stays applied until power is lost.

This is a solution for anyone waiting for this issue to be fixed. :)

jonasmalacofilho, 2023-06-16:
Possibly related to #606 (and #550), and the use of slightly different protocol.