liquidctl/liquidctl #542: Corsair iCUE Elite Capellix H170i

paskalll, 2022-12-08:
### Device type

AIO

### Product page

https://www.corsair.com/de/de/Kategorien/Produkte/CORSAIR-iCUE/Flüssigkühlung/iCUE-ELITE-CAPELLIX-Liquid-CPU-Cooler/p/CW-9060055-WW

### First-party software

Corsair ICUE

### What monitoring functionality does the device support?

fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of accessories like RGB fans or LED strips

### Physical connection

USB Header

### Additional information about the device

Bus 001 Device 011: ID 1b1c:0c39 Corsair LCD Cap for Elite Capellix coolers

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

paskalll, 2022-12-14:
Bus 002 Device 002: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core
Bus 001 Device 002: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT
Bus 001 Device 004: ID 1b1c:0c39 Corsair LCD Cap for Elite Capellix coolers

liquidctl list
Device #0: Corsair Commander Core XT (experimental)
so 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core don't list.

paskalll, 2022-12-14:
ok seems like 0x1b1c, 0x0c32 is a commander core in my case and not a core st. 
Changed the code and I can list the device now. 
I try to change some speeds now. we will see

jonasmalacofilho, 2022-12-20:
Can you share the output from

```
$ lsusb -v d1b1c:
```

? I would like to take a look at the strings descriptors returned from each device, as well as if the interfaces and endpoints match what we've seen before.

SteveBronder, 2022-12-21:
Hi @jonasmalacofilho see below, I believe I have the same cpu cooler and this might explain why I'm getting an odd error when trying to use liquidctl with the Commander Core and the H170i

```
Bus 001 Device 003: ID 1b1c:0c1c Corsair CORSAIR iCUE Commander CORE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c1c 
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR iCUE Commander CORE
  iSerial                 3 920730402e81e3aafd131a066091005f
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
    Interface Descriptor:
[... 178 lines cut ...]
   Port 9: 0000.0100 power
   Port 10: 0000.0100 power
   Port 11: 0000.0103 power enable connect
   Port 12: 0000.0503 highspeed power enable connect
   Port 13: 0000.0100 power
   Port 14: 0000.0100 power
Device Status:     0x0001
  Self Powered
```

I've attached the debug info for both running the two following commands below

```
liquidctl initialize --debug
liquidctl set fan1 speed 75 --debug
```

https://gist.github.com/SteveBronder/049a364f908b48b97002e9c914898937 

liquidctl version below

```
liquidctl --version
liquidctl v1.12.0.dev63+g2ca87e2 (Linux-5.15.0-56-generic-x86_64-with-glibc2.35)
```

SteveBronder, 2022-12-21:
Actually sorry I think [this](https://github.com/liquidctl/liquidctl/issues/455#issuecomment-1121434961) comment helped me with my issue

paskalll, 2022-12-23:
> Can you share the output from
> 
> ```
> $ lsusb -v d1b1c:
> ```
> 
> ? I would like to take a look at the strings descriptors returned from each device, as well as if the interfaces and endpoints match what we've seen before.
lsusb -v d1b1c:
Usage: lsusb [options]...
List USB devices
  -v, --verbose
      Increase verbosity (show descriptors)
  -s [[bus]:][devnum]
      Show only devices with specified device and/or
      bus numbers (in decimal)
  -d vendor:[product]
      Show only devices with the specified vendor and
      product ID numbers (in hexadecimal)
  -D device
      Selects which device lsusb will examine
  -t, --tree
      Dump the physical USB device hierarchy as a tree
  -V, --version
      Show version of program
  -h, --help
      Show usage and help

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.

closed 2025-11-15 by jonasmalacofilho