liquidctl/liquidctl #543: yoda having ValueErrors

basking-in-the-sun2000, 2022-12-11:
### Describe the bug

Got an error when doing a status call. Have yoda running on the background.

### Commands executed

I'm using this command to launch yoda

`script='/home/user/yoda.py'`                                                                                                  
`/usr/bin/python3 $script control fan with '(30,25),(40,50),(50,70),(60,90)' on _internal.vrm &`

### Output of all relevant commands with `--debug` flag

unexpected error with Corsair HX850i: ValueError('204 is not a valid OCPMode',)

### Affected device

corsair hx850i

### Operating system and version

Linux hive 5.10.0-hiveos #110.hiveos.220411 SMP Mon Apr 11 06:15:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Installation method

pip3

### Version of liquidctl

liquidctl v1.11.1 (Linux-5.10.0-hiveos-x86_64-with-Ubuntu-18.04-bionic)

jonasmalacofilho, 2022-12-11:
That device requires (many) writes for each status call. By having yoda running in the background, your status call and it are racing and trampling over each other when writing to and reading back from the device, and that probably explains (what's likely just) an incorrect read.

Still, since that PSU supports changing some settings (like the OCP mode), that situation is far from ideal. If possible, try to have yoda ignore that device (I assume you're using it to control some other device, not the PSU fan).

---

Implementing synchronized access to devices is in our to-do list (#298), so I'll close this issue. But, admittedly, work on that feature has pretty much stalled...

closed 2022-12-11 by jonasmalacofilho

basking-in-the-sun2000, 2022-12-11:
thanks. Only got the psu. Will increase yoda interval, hoping that helps