liquidctl/liquidctl #546: ERROR: unexpected OS error with Corsair HYdro H115i Platinum: OSError('Could not write to device')

HeathenHacks, 2022-12-18:
### Describe the bug

I'm using Coolero to control my AIO and fan's speed and liquidctl is one of it's dependencies.

I'm using an Arch-derivative distro called "ArcoLinux" with KDE Plasma as my WM.

Earlier today, after I booted up my PC and logged in, I received a notification from Coolero that's says:

### Device Communication Issue

**Liquidctl** has detected a communication problem with your device.

This is most likely due to permissions issue when accessing devices **not** as root.

To give your user access to the system's USB devices you can apply some udev rules.

[image]

-------------------------------------------------------------------------------------------------------------------------------------------------------------

So, I clicked on **Do it**, then restarted my PC.

But, after logging in, I received the same notification.

So, I checked on liquidctl via the terminal.

I entered `liquidctl status`

Then, it showed the unexpected OS error.

[image]
-------------------------------------------------------------------------------------------------------------------------------------------------------------

It can detect my Commander Pro just fine, but it's having a problem with my AIO.

Just happened today as well.

Thanks!

### Commands executed

`$ liquidctl status`
`$ liquidctl initialize all`

### Output of all relevant commands with `--debug` flag

$ liquidctl status --debug
[DEBUG] (cli) (main): script: /usr/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Linux-6.0.12-arch1-1-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_requirements): python: 3.10.8 (main, Nov  1 2022, 14:18:21) [GCC 12.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.3.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0008 (skipped, not a bus or unsupported adapter)
[... 219 lines cut ...]
├── Temperature probe 2    Yes
├── Temperature probe 3    Yes
├── Temperature probe 4    Yes
├── Fan 1 control mode     PWM
├── Fan 2 control mode     PWM
├── Fan 3 control mode     PWM
├── Fan 4 control mode     PWM
├── Fan 5 control mode     N/A
└── Fan 6 control mode     N/A

### Affected device

Corsair Hydro H115i RGB Platinum

### Operating system and version

Arch Linux

### Installation method

AUR

### Version of liquidctl

liquidctl v1.11.1 (Linux-6.0.12-arch1-1-x86_64-with-glibc2.36)

HeathenHacks, 2022-12-18:
Not sure if I should mark this as closed or not, but after numerous reboots (because of multi-monitor problems), the issue with my AIO seemed to have been fixed.

closed 2022-12-18 by HeathenHacks

jonasmalacofilho, 2022-12-20:
That's odd. Please let me know if the issue comes back.