liquidctl/liquidctl #547: TEAMGROUP Team T-Force Delta RGB DDR4 2x16GB

feiticeir0, 2022-12-20:
### Device type

RAM

### Product page

https://www.teamgroupinc.com/en/product/delta-rgb-ddr4

### First-party software

It fully supports Asus AURA SYNC

### What configuration functionality does the device support?

lighting of embedded LEDs

### Physical connection

motherboard memory slots

### Additional information about the device

I don't know how the communication is handled, but here's the memory info. 

`sudo lshw -C memory
  *-firmware                
       description: BIOS
       vendor: American Megatrends Inc.
       physical id: 0
       version: 2403
       date: 06/16/2021
       size: 64KiB
       capacity: 16MiB
       capabilities: pci apm upgrade shadowing cdboot bootselect socketedrom edd int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer acpi usb biosbootspecification uefi
  *-memory
       description: System Memory
       physical id: 31
       slot: System board or motherboard
       size: 32GiB
     *-bank:0
          description: [empty]
          product: Unknown
          vendor: Unknown
          physical id: 0
          serial: Unknown
          slot: DIMM_A1
     *-bank:1
          description: DIMM DDR4 Synchronous Unbuffered (Unregistered) 3200 MHz (0.3 ns)
          product: TEAMGROUP-UD4-3200
          vendor: Team Group Inc.
          physical id: 1
          serial: 0104D648
          slot: DIMM_A2
          size: 16GiB
          width: 64 bits
          clock: 3200MHz (0.3ns)
     *-bank:2
          description: [empty]
          product: Unknown
          vendor: Unknown
          physical id: 2
          serial: Unknown
          slot: DIMM_B1
     *-bank:3
          description: DIMM DDR4 Synchronous Unbuffered (Unregistered) 3200 MHz (0.3 ns)
          product: TEAMGROUP-UD4-3200
          vendor: Team Group Inc.
          physical id: 3
          serial: 0104D652
          slot: DIMM_B2
          size: 16GiB
          width: 64 bits
          clock: 3200MHz (0.3ns)
  *-cache:0
       description: L1 cache
       physical id: 33
       slot: L1 - Cache
       size: 512KiB
       capacity: 512KiB
       clock: 1GHz (1.0ns)
       capabilities: pipeline-burst internal write-back unified
       configuration: level=1
  *-cache:1
       description: L2 cache
       physical id: 34
       slot: L2 - Cache
       size: 4MiB
       capacity: 4MiB
       clock: 1GHz (1.0ns)
       capabilities: pipeline-burst internal write-back unified
       configuration: level=2
  *-cache:2
       description: L3 cache
       physical id: 35
       slot: L3 - Cache
       size: 16MiB
       capacity: 16MiB
       clock: 1GHz (1.0ns)
       capabilities: pipeline-burst internal write-back unified
       configuration: level=3
`

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic, attempting to capture I²C/SMBus traffic, analyzing traffic data

jonasmalacofilho, 2022-12-21:
Can you provide the output from `decode-dimms`, as root?

You can usually find `decode-dimms` as part of the i2c-tools package.

---

Can you also tell me what platform you're running on (Intel/AMD, which chipset)? We'll need some features that are only (easily) available, at the kernel level, for some of these platforms.

feiticeir0, 2022-12-21:
> Can you provide the output from `decode-dimms`, as root?
> 
> You can usually find `decode-dimms` as part of the i2c-tools package.
> 
> Can you also tell me what platform you're running on (Intel/AMD, which chipset)? We'd need some features that are only (easily) available, at the kernel level, for some of these platforms.

Hi @jonasmalacofilho  ! Thank you for the reply. 
I can gladly provide what you want:
I've struggled with decode-dimms, because at first it wasn't providing any response:
`No EEPROM found, try loading the eeprom, at24 or ee1004 module`

After a bit of googling, I found this page:
`https://www.spinics.net/lists/linux-i2c/msg32331.html`

And after following some instructions:

echo ee1004 0x52 > /sys/bus/i2c/devices/i2c-0/new_device 
ee1004 0x53 > /sys/bus/i2c/devices/i2c-0/new_device

and the decode-dimms finally worked:
`[root@nightcrawler ~]# decode-dimms 
# decode-dimms version 4.3

Memory Serial Presence Detect Decoder
By Philip Edelbrock, Christian Zuckschwerdt, Burkart Lingner,
Jean Delvare, Trent Piepho and others

Decoding EEPROM: /sys/bus/i2c/drivers/ee1004/0-0052
Guessing DIMM is in                              bank 3
Kernel driver used                               ee1004

---=== SPD EEPROM Information ===---
EEPROM CRC of bytes 0-125                        OK (0x3B9F)
# of bytes written to SDRAM EEPROM               384
Total number of bytes in EEPROM                  512
Fundamental Memory type                          DDR4 SDRAM
SPD Revision                                     1.1
Module Type                                      UDIMM
EEPROM CRC of bytes 128-253                      OK (0x9464)

---=== Memory Characteristics ===---
Maximum module speed                             2400 MT/s (PC4-19200)
Size                                             16384 MB
Banks x Rows x Columns x Bits                    16 x 16 x 10 x 64
SDRAM Device Width                               8 bits
Ranks                                            2
Rank Mix                                         Symmetrical
Primary Bus Width                                64 bits
AA-RCD-RP-RAS (cycles)                           16-16-16-39
Supported CAS Latencies                          18T, 17T, 16T, 15T, 14T, 13T, 12T, 11T, 10T

---=== Timings at Standard Speeds ===---
AA-RCD-RP-RAS (cycles) as DDR4-2400              16-16-16-39
AA-RCD-RP-RAS (cycles) as DDR4-2133              15-15-15-35
AA-RCD-RP-RAS (cycles) as DDR4-1866              13-13-13-30
AA-RCD-RP-RAS (cycles) as DDR4-1600              11-11-11-26

---=== Timing Parameters ===---
Minimum Cycle Time (tCKmin)                      0.833 ns
Maximum Cycle Time (tCKmax)                      1.600 ns
Minimum CAS Latency Time (tAA)                   13.320 ns
Minimum RAS to CAS Delay (tRCD)                  13.320 ns
Minimum Row Precharge Delay (tRP)                13.320 ns
Minimum Active to Precharge Delay (tRAS)         32.000 ns
Minimum Active to Auto-Refresh Delay (tRC)       45.320 ns
Minimum Recovery Delay (tRFC1)                   350.000 ns
Minimum Recovery Delay (tRFC2)                   260.000 ns
Minimum Recovery Delay (tRFC4)                   160.000 ns
Minimum Four Activate Window Delay (tFAW)        21.000 ns
Minimum Row Active to Row Active Delay (tRRD_S)  3.299 ns
[... 75 lines cut ...]
---=== Other Information ===---
Package Type                                     Monolithic
Maximum Activate Count (MAC)                     Unlimited
Post Package Repair                              One row per bank group
Soft PPR                                         Supported
Module Nominal Voltage                           1.2 V
Thermal Sensor                                   No

---=== Physical Characteristics ===---
Module Height                                    32 mm
Module Thickness                                 2 mm front, 2 mm back
Module Reference Card                            A revision 2

---=== Manufacturer Data ===---
Module Manufacturer                              Team Group Inc.
DRAM Manufacturer                                Samsung
Manufacturing Date                               2022-W31
Assembly Serial Number                           0x0104D652
Part Number                                      TEAMGROUP-UD4-3200  

Number of SDRAM DIMMs detected and decoded: 2
`
I'm running AMD platform - AMD Ryzen 7 5700G on a ASUS ROG STRIX B550-f gamming .  The chipset is B550 .

Thank you.

jonasmalacofilho, 2022-12-21:
Thanks for the infos!

So, in order to support these modules, it'll be necessary to:

- implement auto-detection of DDR4 EEPROMs for Ryzen, either in liquidctl or, preferably, the kernel (for Intel that's done in the `i2c-i801` kernel driver, IIRC);
- capture and reverse engineer the SMBus traffic between Aura on Windows and the modules;[^1]
- and, finally, implement a new driver in liquidctl.

I can help them, but someone else will need to do the bulk of the work. Any chance it's something you're interested in working on?

[^1]: Capturing SMBus traffic can sometimes be a bit more complicated than just opening a tool like Wireshark. There are software tools, but they might not work in all cases. Luckily, capturing SMBus traffic with a hardware tool is fairly easy and inexpensive: in the past I got everything I needed – a cheap logic analyzer and a blank or adapter DIMM – under 30 USD (and this in Brazil, where things tend to more expensive than US/EU). Since the SMBus is exposed through the DIMM socket, there's no need to attach/solder things directly to your motherboard.

jonasmalacofilho, 2022-12-21:
For more information on capturing SMBus traffic, take a look at https://github.com/liquidctl/liquidctl/issues/291#issuecomment-789025503 and subsequent comments. For Ryzen the corresponding OpenRGB sniffer tool is i2c_sniffer_piix4: [OpenRGB's wiki page linking to a pre-built binary](https://gitlab.com/CalcProgrammer1/OpenRGB/-/wikis/Reverse-Engineering) and [source](https://gitlab.com/CalcProgrammer1/OpenRGB/-/tree/i2c_sniffer_piix4).

The diagrams from OpenRGB's [SMBus Interface Details](https://gitlab.com/CalcProgrammer1/OpenRGB/-/wikis/SMBus-Interface-Details) wiki page are also useful.

feiticeir0, 2022-12-21:
> Thanks for the infos!
> 
> So, in order to support these modules, it'll be necessary to:
> 
>     * implement auto-detection of DDR4 EEPROMs for Ryzen, either in liquidctl or, preferably, the kernel (for Intel that's done in the `i2c-i801` kernel driver, IIRC);
> 
>     * capture and reverse engineer the SMBus traffic between Aura on Windows and the modules;[1](#user-content-fn-1-6e2d6796872974cdea44df41c377ab75)
> 
>     * and, finally, implement a new driver in liquidctl.
> 
> 
> I can help them, but someone else will need to do the bulk of the work. Any chance it's something you're interested in working on?
> ## Footnotes
> 
>     1. Capturing SMBus traffic can sometimes be a bit more complicated than just opening a tool like Wireshark. There are software tools, but they might not work in all cases. Luckily, capturing SMBus traffic with a hardware tool is fairly easy and inexpensive: in the past I got everything I needed – a cheap logic analyzer and a blank or adapter DIMM – under 30 USD (and this in Brazil, where things tend to more expensive than US/EU). Since the SMBus is exposed through the DIMM socket, there's no need to attach/solder things directly to your motherboard. [leftwards_arrow_with_hook](#user-content-fnref-1-6e2d6796872974cdea44df41c377ab75)

Hi ! Thank you for the reply. 
Yes, I'm guessing capturing the traffic will be no problem, but will have to explore the wiki page . I will have no trouble using hardware, especially if I don't have to mess around the motherboard . 

If I can, I will ! 
Will explore the OpenRGB sniffer and I'll let you know

feiticeir0, 2022-12-24:
Hi ! 
I've captured several packages using OpenRGB's sniffers. 
Only got something using the bus0 sniffer, for the PIIX4 - AMD
_PIIX4 Sniffer Bus 0_

I have 3 files, comprising of 3 light modes - static, breathing and strobing . 
**static**
Here are the colors used:
R | G | B
170 | 0 | 255

255 | 0 | 0

255 | 255 | 0 

0 | 255 | 255

**breathing**
R | G | B 
170 | 0 | 255  - Speed 1 (from 0 to 4)
Speed 3
Speed 0

0 | 0 | 255

**strobing**
R | G | B 

112 | 255 | 0 Speed 2

speed 0

255 | 0 | 0 

[bus0_strobing.txt](https://github.com/liquidctl/liquidctl/files/10298686/bus0_strobing.txt)
[bus0_breathing.txt](https://github.com/liquidctl/liquidctl/files/10298687/bus0_breathing.txt)
[bus0_static.txt](https://github.com/liquidctl/liquidctl/files/10298688/bus0_static.txt)

If more needed, I'm here .

feiticeir0, 2022-12-26:
Not putting any pressure, but yesterday I was trying openRGB and the memory is already supported... I can change the lighting colors and effects. 
Here's a screenshot:
[image]

jonasmalacofilho, 2023-01-02:
That's good no know, thanks! And it means that porting the OpenRGB controller to liquidctl is another way to approach this.

But like I mentioned before, someone else needs to do the bulk of the work. I can't do it myself right now (or in the foreseeable future).

feiticeir0, 2023-01-02:
> That's good no know, thanks! And it means that porting the OpenRGB controller to liquidctl is another way to approach this.
> 
> But like I mentioned before, someone else needs to do the bulk of the work. I can't do it myself right now (or in the foreseeable future).

I will capture the traffic. I've already done that, but it's not well done. I've been reading OpenRGB wiki and they explain a better way to do this. 
Next weekend will do it. I already have the sniffers .