liquidctl/liquidctl #548: Asus ROG STRIX LC

mingfai, 2022-12-23:
### Device type

AIO liquid cooler

### Product page

https://www.asus.com/hk/motherboards-components/cooling/rog-republic-of-gamers/filter?SubSeries=ROG-Strix-LC

### First-party software

Armoury Crate for win10/11 (https://www.asus.com/us/supportonly/armoury%20crate/helpdesk_download/)

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles

### Physical connection

onboard (detected as USB)

mingfai, 2022-12-23:
FanXpert, Armoury Crate
[image]

jonasmalacofilho, 2023-01-16:
I'll probably wont work on this myself. But, I'll certainly accept a PR adding a driver for this device (with or without support for the screen).