liquidctl/liquidctl #549: Corsair iCUE Commander Core not listed by default

kozlov-ma, 2022-12-26:
### Describe the bug

Corsair iCUE Commander Core (from Corsair iCUE h150i Elite Capellix) is matched by ID `1b1c:0c1c` in 'commander_core.py' driver. However, in my `lsusb` output it is has another id, so after I modified the `commander_core.py` on my computer with the respective values everything started working as expected.

### Commands executed

lsusb output for my device:
`Bus 005 Device 006: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core`

Expected lsusb output, according to the original commander_core.py file:
`Bus XXX Device XXX: ID 1b1c:0c1c Corsair CORSAIR iCUE COMMANDER Core`

### Output of all relevant commands with `--debug` flag

I ran this command after modifying the commander_core.py file
```
$ sudo liquidctl status --debug
[DEBUG] (cli) (main): script: /usr/bin/liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Linux-6.1.1-arch1-1-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_requirements): python: 3.10.9 (main, Dec 19 2022, 17:35:49) [GCC 12.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.3.0
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-CYP0002:00 (skipped, invalid literal for int() with base 10: 'CYP0002:00')
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-MSFT8000:00 (skipped, invalid literal for int() with base 10: 'MSFT8000:00')
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 3434:0100 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3434:0100 (usage_page=0xff60 usage=0x0061)
[DEBUG] (usb) (find_devices): HID device: 3434:0100 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 3434:0100 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3434:0100 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0b05:1a53 (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 14ed:1012 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 14ed:1012 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4082 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (probe): AuraLed identified: ASUS Aura LED Controller (experimental)
[DEBUG] (usb) (find_devices): HID device: 056a:0378 (usage_page=0xff0d usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 056a:0378 (usage_page=0xff0d usage=0x0020)
[DEBUG] (usb) (find_devices): HID device: 056a:0378 (usage_page=0xff0d usage=0x0039)
[DEBUG] (usb) (find_devices): HID device: 056a:0378 (usage_page=0xff0d usage=0x1013)
[DEBUG] (usb) (find_devices): HID device: 056a:0378 (usage_page=0xff0d usage=0x000e)
[... 76 lines cut ...]
├── Pump speed           2477  rpm
├── Fan speed 1             0  rpm
├── Fan speed 2             0  rpm
├── Fan speed 3             0  rpm
├── Fan speed 4           699  rpm
├── Fan speed 5           732  rpm
├── Fan speed 6           726  rpm
└── Water temperature    34.8  °C
```

### Affected device

Corsair iCUE h150i Elite Capellix

### Operating system and version

Linux 6.1

### Installation method

Official Arch & Fedora repos

### Version of liquidctl

liquidctl v1.11.1 (Linux-6.1.1-arch1-1-x86_64-with-glibc2.36)

jonasmalacofilho, 2022-12-27:
Related to (conflicting) degrees: #408, #511 and #542.

paskalll, 2022-12-30:
> Related and/or conflicting with: #408, #511 and #542.

The commander core st is not a core .....
It's just a Commander ST and we are mismatching it with a commander core here. But the ST is not a Core when you watch the name

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.