liquidctl/liquidctl #550: NZXT RGB Controller AC-CRFR0-B1 (Included in the NZXT H5 Elite case)

Luisj5231, 2022-12-27:
### Device type

NZXT

### Product page

https://nzxt.com/product/h5-elite

### First-party software

NZXT CAM

### What configuration functionality does the device support?

lighting of accessories like RGB fans or LED strips

### Physical connection

USB 2.0 Header

### Additional information about the device

I got the NZXT H5 Elite case, that includes an RGB Controller (), this device only have 3 connectors for controlling the RGB with the use of the NZXT CAM software. I'm only using two connection ports.

I added the new device to the smart_device.py file.

```python
class SmartDevice2(_BaseSmartDevice):
    """NZXT HUE 2 lighting and, optionally, fan controller."""

    # support for hwmon: nzxt-smart2, Linux 5.17

    _MATCHES = [
        (0x1e71, 0x2006, 'NZXT Smart Device V2', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x200d, 'NZXT Smart Device V2', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x200f, 'NZXT Smart Device V2', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x2001, 'NZXT HUE 2', {
            'speed_channel_count': 0,
            'color_channel_count': 4
        }),
        (0x1e71, 0x2002, 'NZXT HUE 2 Ambient', {
            'speed_channel_count': 0,
            'color_channel_count': 2
        }),
        (0x1e71, 0x2009, 'NZXT RGB & Fan Controller', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x200e, 'NZXT RGB & Fan Controller', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x2010, 'NZXT RGB & Fan Controller', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x2019, 'NZXT RGB & Fan Controller', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x2012, "NZXT RGB Controller", { # Here is the new device
[... 358 lines cut ...]
     * . . .  . . . .  . . . .  . . . .  . . . .  . . . .  . . . .  . . . .
b6ed63c0 0.455284 C Io:1:006:1 0:1 64 >
8a163080 0.457288 C Ii:1:006:1 -2:1 0
8a1638c0 0.518413 S Ii:1:007:1 -:1 64 <
8a1638c0 0.519282 C Ii:1:007:1 0:1 64 =
    ff031610 031503d5 70ab2320 af5e0c0a 00000000 00000000 00000000 00000000
     . . . .  . . . .  p . #    . ^ . .  . . . .  . . . .  . . . .  . . . .
8a1638c0 0.519284 S Ii:1:007:1 -:1 64 <
b6ed6180 0.570540 S Io:1:007:2 -:1 64 =
    28030100 00020000 01000000 00000000 00000000 00000000 00000000 00000000
     ( . . .  . . . .  . . . .  . . . .  . . . .  . . . .  . . . .  . . . .
b6ed6180 0.571283 C Io:1:007:2 0:1 64 >
8a1638c0 0.573416 C Ii:1:007:1 -2:1 0
8a1638c0 0.635586 S Ii:1:007:1 -:1 64 <
47daefc0 0.687479 S Io:1:007:2 -:1 64 =
    28030100 02020000 01000000 00000000 00000000 00000000 00000000 00000000
     ( . . .  . . . .  . . . .  . . . .  . . . .  . . . .  . . . .  . . . .
47daefc0 0.688274 C Io:1:007:2 0:1 64 >
8a1638c0 0.690406 C Ii:1:007:1 -2:1 0

```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic, implementing the changes

jonasmalacofilho, 2023-01-16:
Sorry for the long response time.

---

Odd, the protocol appears  to have some similarity with the HUE2 and RGB & Fan Controller, but clearly something is different in how LED colors are passed to the device.

Can you try capturing some traffic between CAM and the device?

(By the way, there usually isn't a lot of value in capturing the traffic between liquidctl and the device: we already have quite a bit of info available with `--debug`, and we can also just hack the sources and add more traces... or even use an interactive debugger).

Luisj5231, 2023-01-17:
Hello Jonas.

Sure. just let me setup a Windows machine and I will check it but, how could I do this? I'm not too skilled with Windows.

jonasmalacofilho, 2023-01-17:
You can still capture the traffic on the Linux host (using [usbmon + Wireshark](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md#capturing-usb-traffic-on-linux)), as any USB device forwarded to the Windows VM will still go throw the host's USB stack.

---

P.S. The Wireshark part is arguably unnecessary, but I find the format a lot easier to work than what usbmon natively outputs.

nemat-rakhmatov, 2023-02-27:
I've got the same NZXT case and attaching some traffic from CAM on virtualbox windows
[usbmon-capture.zip](https://github.com/liquidctl/liquidctl/files/10836824/usbmon-capture.zip)

Please have a look and let me know if you need something else.
BTW, on the first install CAM didn't work, and after it updated RGB controller firmware it started working

nemat-rakhmatov, 2023-03-17:
@jonasmalacofilho 
Hi.
No time to look into this issue? Maybe you can give me some clue to try to fix this myself?

jonasmalacofilho, 2023-03-17:
@nemat-rakhmatov,

> No time to look into this issue?

Yeah... (and also #569).

> Maybe you can give me some clue to try to fix this myself?

Thanks!

My strategy would be to start by comparing what our driver does and what we see in the captures from CAM, for a few specific (but arbitrarily chosen) cases of modes & colors.

PauloMPereira, 2023-03-30:
Hello,

Any news about this topic?
I'm also have the same issue :/

Mindgamesnl, 2023-04-14:
I started working on a small implementation while trying to reverse engineer some other stuff, currently have barebones rgb working in the h5 elite (https://github.com/Mindgamesnl/nzxt-linux-driver). I'll contribute here once I have most stuff implemented and figured out.