liquidctl/liquidctl #553: [FR] A support for buffering LED color setting.

ErrorBot1122, 2023-01-02:
For my **Feature Request** I want support for **Buffering** when setting the colors of the LEDs. 

## The Problem

_Currently_, you have to set each color group of LEDs **separately**. _This is a problem_ as it takes about a **half second** for it to update the LEDs, so it can take a while if you want, for instance, an **_alternating strip of colors_**.

## The Solution

_Add support for_ **_buffering_** inputs, **meaning**, setting all the colors _beforehand_ and changing them **all at once**. 

## The Benefits

_By doing this,_ it will a lot smother to make custom **animations** with out it being so _flicker-y_.

## The Downsides

_One of the biggest downsides_ would be the **programing**, as I personally don't know how this program works _internally_ so I don't know it would be **possible** to do this. **Especially** for all devices.

jonasmalacofilho, 2023-01-03:
In many cases almost all of the time is spent waiting for I/O from/to the device. What devices are we talking about here? 

Additionally, what commands (with arguments) do you execute that take half a second? And can you provide their complete outputs when running with `--debug`?

ErrorBot1122, 2023-01-04:
> What devices are we talking about here?

The [Corsair Lighting Node Pro](https://github.com/liquidctl/liquidctl/blob/main/docs/corsair-commander-guide.md)

I'm actually using python for this, running this command:
```python
subprocess.check_output(f"liquidctl --address \"{device_address}\" set sync color fixed ff0000 --start-led {led} --maximum-leds {led_amount} --json", shell=True)
```

 - `device_address` is the address part of the **Corsair Lighting Node.** _(from the `list` command)_
 - `led` is the starting index of the first _LED_ I want to change. _(e.g. `1`)_
 - `led_amount` is the **AMOUNT** of _LEDs_ I want to set. _(e.g. `24` for my 3 8-LED fans)_

``

> And can you provide their complete outputs when running with `--debug`?

```shell
Errorbot@Errors-Beafy-PC MINGW64 ~/Documents/Stuff/Coding/Python/Custom-Light-Test
$ liquidctl --address '\\?\HID#VID_1B1C&PID_0C1A#7&dd063cf&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}' set sync color fixed ff0000 --start-led 1 --maximum-leds 24 --debug
[DEBUG] (cli) (main): script: C:\Users\Errorbot\Documents\Stuff\Coding\Python\Custom-Light-Test\venv\Scripts\liquidctl
[DEBUG] (cli) (main): version: 1.11.1
[DEBUG] (cli) (main): platform: Windows-10-10.0.19045-SP0
[DEBUG] (cli) (_log_requirements): python: 3.10.9 (tags/v3.10.9:1dd9be6, Dec  6 2022, 20:01:21) [MSC v.1934 64 bit (AMD64)]
[DEBUG] (cli) (_log_requirements): colorlog: 6.7.0
[DEBUG] (cli) (_log_requirements): crcmod: 1.7
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.12.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): pillow: 9.3.0
[DEBUG] (cli) (_log_requirements): smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_requirements): winusbcdc: 1.5
[DEBUG] (cli) (_log_requirements): libusb-package: 1.0.26.1
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1a (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (probe): CommanderPro identified: Corsair Lighting Node Core
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
[DEBUG] (cli) (main): device: Corsair Lighting Node Core
[DEBUG] (keyval) (__init__): data in C:\Users\Errorbot\AppData\Local\Temp\liquidctl\vid1b1c_pid0c1a\loc1_1_0_1_7_063_0_0000_4_1_55_2_16_11_88_001111000030
[DEBUG] (keyval) (load): loaded saved_effects=[{'channel': 0, 'start_led': 0, 'num_leds': 8, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 
8, 'num_leds': 8, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 16, 'num_leds': 8, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 
0, 'colors': [255, 0, 0]}] (from C:\Users\Errorbot\AppData\Local\Temp\liquidctl\vid1b1c_pid0c1a\loc1_1_0_1_7_063_0_0000_4_1_55_2_16_11_88_001111000030\saved_effects)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 37:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 3 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 34:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 3 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 38:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (keyval) (store): stored saved_effects=[{'channel': 0, 'start_led': 0, 'num_leds': 8, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 8, 'num_leds': 8, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 16, 'num_leds': 8, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 24, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}] (in C:\Users\Errorbot\AppData\Local\Temp\liquidctl\vid1b1c_pid0c1a\loc1_1_0_1_7_063_0_0000_4_1_55_2_16_11_88_001111000030\saved_effects)
[DEBUG] (usb) (clear_enqueued_reports): discarded 3 previously enqueued reports
[... cut ...]
[DEBUG] (usb) (read): read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 3 previously enqueued reports
[DEBUG] (usb) (write): writting report 0x00 with 64 bytes: 33:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```

jonasmalacofilho, 2023-01-04:
Using the `set_color` API directly, instead of going through the CLI and a separate Python subprocess, might improve things slightly. But I suspect a significant amount of time will still be necessary for I/O alone.

```
$ pydoc liquidctl.driver.commander_pro.CommanderPro.set_color
Help on function set_color in liquidctl.driver.commander_pro.CommanderPro:

liquidctl.driver.commander_pro.CommanderPro.set_color = set_color(self, channel, mode, colors, direction='forward', speed='medium', start_led=1, maximum_leds=204, **kwargs)
    Set the color of each LED.
    
    The table bellow summarizes the available channels, modes, and their
    associated maximum number of colors for each device family.
    
    | Channel  | Mode        | Num colors |
    | -------- | ----------- | ---------- |
    | led      | off         |          0 |
    | led      | fixed       |          1 |
    | led      | color_shift |          2 |
    | led      | color_pulse |          2 |
    | led      | color_wave  |          2 |
    | led      | visor       |          2 |
    | led      | blink       |          2 |
    | led      | marquee     |          1 |
    | led      | sequential  |          1 |
    | led      | rainbow     |          0 |
    | led      | rainbow2    |          0 |
```

Additionally, what you described as a generic (as in driver-agnostic) feature for liquidctl wouldn't really help here. What would be more likely to help is to have our API specification more closely match how that particular device protocol works.  But that would obviously introduce issues with the other devices we support, and is also outside the scope of liquidctl.

Anyway, thanks for the feature request, and please let me know if using the API directly results in any significant improvement.

---

Closing: outside the scope.

closed 2023-01-04 by jonasmalacofilho