liquidctl/liquidctl #556: Support for Corsair iCUE H100i RGB ELITE

st1vms, 2023-01-12:
### Device type

Corsair iCUE H100i RGB ELITE 

### Product page

https://www.corsair.com/es/it/Categorie/Prodotti/Sistemi-di-raffreddamento-a-liquido/Sistema-di-raffreddamento-a-liquido-per-CPU-iCUE-RGB-ELITE/p/CW-9060058-WW

### First-party software

Corsair iCue

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs

### Physical connection

USB header

### Connection protocol

USB/HID

### Additional information about the device

Content of `/etc/udev/rules.d/71-liquidctl.rules` :

```
# Copied from official liquidctl repo: https://github.com/liquidctl/liquidctl/blob/main/extra/linux/71-liquidctl.rules

# Rules that grant unprivileged access to devices supported by liquidctl.
#
# Users and distros are encouraged to use these if they want liquidctl to work
# without requiring root privileges (e.g. without the use of `sudo`).
#
# In the case of I²C/SMBus devices, these rules also cause the loading of the
# `i2c-dev` kernel module.  This module is required for access to I²C/SMBus
# devices from userspace, and manually loading kernel modules is in itself a
# privileged operation.
#
# Distros will likely want to place this file in `/usr/lib/udev/rules.d/`, while
# users installing this manually SHOULD use `/etc/udev/rules.d/` instead.
#
# The suggested name for this file is `71-liquidctl.rules`.  Whatever name is
# used, it MUST lexically appear before 73-seat-late.rules.  The suggested name
# was chosen so that it is also lexically after systemd-provided 70-uaccess.rules.
#
# Once installed, reload and trigger the new rules with:
#
#   # udevadm control --reload
#   # udevadm trigger
#
# Note that this will not change the mode of `/dev/hidraw*` devices that have
# already been created.  In practice, this means that HIDs may continue to require
# privileged access until, either, they are rebound to their kernel drivers, or
# the system is rebooted.
#
# These rules assume a system with modern versions of systemd/udev, that support
# the `uaccess` tag.  On older systems the rules can be changed to instead set
# GROUP="plugdev" and MODE="0660"; other groups and modes may also be used.
#
# The use of the `uaccess` mechanism assumes that only physical sessions (or
# "seats") need unprivileged access to the devices.[^1][^2]  In case headless
# sessions are also expected to interactively run liquidctl, GROUP and MODE should
# also be set, as a fallback.
#
# Finally, this file was automatically generated.  To update it, from a Linux
# shell and the current directory, execute:
#
#     $ python generate-uaccess-udev-rules.py > 71-liquidctl.rules
#
# [^1]: https://github.com/systemd/systemd/issues/4288
# [^2]: https://wiki.archlinux.org/title/Users_and_groups#Pre-systemd_groups

# Section: special cases

# Host SMBus on Intel mainstream/HEDT platforms
KERNEL=="i2c-*", DRIVERS=="i801_smbus", TAG+="uaccess", \
    RUN{builtin}="kmod load i2c-dev"

# Section: USB devices and USB HIDs

# ASUS Aura LED Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="19af", TAG+="uaccess"

# ASUS Aura LED Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="1939", TAG+="uaccess"

# ASUS Aura LED Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="18f3", TAG+="uaccess"

# Corsair Commander Core
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c1c", TAG+="uaccess"

# Corsair Commander Core XT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c2a", TAG+="uaccess"

# Corsair Commander Pro
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c10", TAG+="uaccess"

# Corsair HX1000i
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1c07", TAG+="uaccess"

# Corsair Hydro H100i GTX
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c03", TAG+="uaccess"

# Corsair Hydro H100i Platinum
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c18", TAG+="uaccess"

# Corsair Hydro H100i Platinum SE
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c19", TAG+="uaccess"

# Corsair Hydro H100i Pro
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c15", TAG+="uaccess"

# Corsair Hydro H100i Pro XT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c20", TAG+="uaccess"

# Corsair Hydro H100i v2
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c09", TAG+="uaccess"

# Corsair Hydro H110i GTX
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c07", TAG+="uaccess"

# Corsair Hydro H115i
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c0a", TAG+="uaccess"

# Corsair Hydro H115i Platinum
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c17", TAG+="uaccess"

# Corsair Hydro H115i Pro
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c13", TAG+="uaccess"

# Corsair Hydro H115i Pro XT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c21", TAG+="uaccess"

# Corsair Hydro H150i Pro
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c12", TAG+="uaccess"

# Corsair Hydro H150i Pro XT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c22", TAG+="uaccess"

# Corsair Hydro H60i Pro XT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c29", TAG+="uaccess"

# Corsair Hydro H80i GT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c02", TAG+="uaccess"

# Corsair Hydro H80i v2
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c08", TAG+="uaccess"

# Corsair Lighting Node Core
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c1a", TAG+="uaccess"

# Corsair Lighting Node Pro
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c0b", TAG+="uaccess"

# Corsair Obsidian 1000D
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1d00", TAG+="uaccess"
```

Corsair related `lsusb -v` output:

```
Bus 003 Device 003: ID 1b1c:0c35 Corsair H100iELITE Device Descriptor:
  bLength 18 bDescriptorType 1 bcdUSB 2.00 bDeviceClass 0
  bDeviceSubClass 0 bDeviceProtocol 0 bMaxPacketSize0 64 idVendor 0x1b1c
  Corsair idProduct 0x0c35 bcdDevice 1.00 iManufacturer 1 Corsair Inc.
  iProduct 2 H100iELITE iSerial 0 bNumConfigurations 1 Configuration
  Descriptor:
    bLength 9 bDescriptorType 2 wTotalLength 0x0022 bNumInterfaces 1
    bConfigurationValue 1 iConfiguration 0 bmAttributes 0xc0
      Self Powered MaxPower 100mA Interface Descriptor: bLength 9
      bDescriptorType 4 bInterfaceNumber 0 bAlternateSetting 0
      bNumEndpoints 1 bInterfaceClass 3 Human Interface Device
      bInterfaceSubClass 0 bInterfaceProtocol 1 Keyboard iInterface 0
        HID Device Descriptor: bLength 9 bDescriptorType 33 bcdHID 1.11
          bCountryCode 0 Not supported bNumDescriptors 1 bDescriptorType
          34 Report wDescriptorLength 27
         Report Descriptors: ** UNAVAILABLE ** Endpoint Descriptor:
        bLength 7 bDescriptorType 5 bEndpointAddress 0x81 EP 1 IN
        bmAttributes 3
          Transfer Type Interrupt Synch Type None Usage Type Data
        wMaxPacketSize 0x0040 1x 64 bytes bInterval 7
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic, attempting to capture I²C/SMBus traffic, analyzing traffic data, documenting the protocol, implementing the changes

st1vms, 2023-01-12:
These are the filters I used to capture usb traffic on all the `usbmon*` devices:

`usb.device_address == 3`
`usb.idVendor == 0x1b1c && usb.idProduct == 0x0c35`

jonasmalacofilho, 2023-01-12:
Related: #498.

---

I wont be able to look into the captures right now, but from #498 it seems that the RGB Elite family uses a protocol that's quite similar to the Platinum/Pro XT family.

So what I can recommend is that you start by looking into:

- the issue mentioned above,
- our [`hydro_platinum`](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/hydro_platinum.py) module,
- past collected data for the Platinum/Pro XT family in our [collected-device-data](https://github.com/liquidctl/collected-device-data) repository,
- and the corresponding [traffic analysis script](https://github.com/liquidctl/collected-device-data/blob/master/Corsair%20H115i%20RGB%20Platinum/analyze.py).

(Important: take care to not mix Pro XT and Pro coolers from Corsair, as those are _very different_ families and protocols).

st1vms, 2023-01-13:
I'm proposing this [fork](https://github.com/st1vms/liquidctl/tree/icue-rgb-elite),
having pushed a working driver for this device.

Since this is a totally different cooler family, i thought of adding an adapted copy of [hydro_platinum](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/hydro_platinum.py) as [icue_rgb_elite](https://github.com/st1vms/liquidctl/blob/icue-rgb-elite/liquidctl/driver/icue_rgb_elite.py), in case anyone would like to publish their specific product info. But it may also be merged to hydro, since they do differs on a few changes.

Tested on this version:
`~$ liquidctl --version`
`liquidctl v1.13.0.dev6+g4c98127.d20230113 (Linux-5.15.0-58-generic-x86_64-with-glibc2.35)`

```
~$ liquidctl status
Corsair iCUE H100i RGB ELITE
├── Liquid temperature    29.7  °C
├── Fan 1 speed            708  rpm
├── Fan 1 duty              39  %
├── Fan 2 speed            698  rpm
├── Fan 2 duty              39  %
├── Pump speed            2005  rpm
└── Pump duty               55  %
```

By adding the approriate [udev rule](https://github.com/st1vms/liquidctl/blob/91c40e418eb4917a4d2822e10d701f028f5f4aca/extra/linux/71-liquidctl.rules#L365)
and installing from source using sudo, i was able to set up fan duties by temperature steps, pump-mode and color using a systemd .service file saved as `/etc/systemd/system/liquidctl.service` :

```
[Unit]
Description=liquidctl service

[Service]
Type=oneshot
ExecStart=liquidctl initialize all
#Corsair iCUE H100i RGB ELITE
ExecStart=liquidctl -m corsair initialize --pump-mode quiet
# 20°C -> 20% 
# 30°C -> 40%
# 40°C -> 60% 
# 60°C -> 80%
# 80°C -> 100%
ExecStart=liquidctl -m corsair set fan1 speed 20 20 30 40 40 60 60 80 80 100
ExecStart=liquidctl -m corsair set fan2 speed 20 20 30 40 40 60 60 80 80 100
ExecStart=liquidctl -m corsair set led color fixed 4200ff

[Install]
WantedBy=default.target
```

st1vms, 2023-01-13:
Opened a [pull request](https://github.com/liquidctl/liquidctl/pull/557#issue-1532318599)

closed 2023-01-14 by jonasmalacofilho