liquidctl/liquidctl #560: Add Asus Rog Ryuo

rjousse18, 2023-01-16:
### Device type

Liquid cooler

### Product page

https://rog.asus.com/fr/cooling/cpu-liquid-coolers/rog-ryuo/rog-ryuo-240-model/

### First-party software

Armoury crate

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs

### Physical connection

USB header

### Connection protocol

Hid (i think)

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

jonasmalacofilho, 2023-01-16:
I'll probably wont work on this myself. But, I'll certainly accept a PR adding a driver for this device (with or without support for the screen).

rjousse18, 2023-01-17:
is there any documentation for how to write driver in python ? I saw that the drivers are on python and I can try to write it myself

rjousse18, 2023-01-17:
ok i will take a look

rjousse18, 2023-01-18:
It seems Armoury Crate can't detect correctly my watercooling through a Windows VM, and I don't have full windows installed on my PC so... Waiting for support I guess

jonasmalacofilho, 2023-01-18:
> It seems Armoury Crate can't detect correctly my watercooling through a Windows VM

Perhaps a hypervisor bug? And/or maybe more than one USB interface is provided by the device, and without passing through all of them, Armory Crate can't quite work out what's wrong?

Either way, unless Armoury Crate is being _actively_ hostile (by, for example, trying to detect it's running inside a VM), a successfully passed through USB device should be indistinguishable to it from a "real" USB device.

jonasmalacofilho, 2023-01-18:
Can you also show the complete Linux output from:

```
$ lsusb -tv
$ lsusb -v
```

jonasmalacofilho, 2023-01-19:
I edited your comment to strip out unrelated devices, but thanks for sending me everything as I asked; I really wanted to check that there wasn't a second, perhaps non-obvious, USB device exposed by the cooler.

---

There's a single configuration and a single interface. So it's weird that the VM had any issues with passing it through. By the way, what VM software are you using?

And were you able to confirm that the device had been correctly passed through to the Windows client, at least as far as Windows is concerned (i.e. through Device Manager)?

rjousse18, 2023-01-19:
I use qemu/virt-manager and yup the device is correctly passthrough.

Armoury Crate detect the device but, when I want to access to parameters, AC crash

rjousse18, 2023-01-21:
VirtualBox not working too

jonasmalacofilho, 2023-01-21:
That's annoying. Maybe Armory Crate tries to access something not well emulated (like I/O ports), or maybe it's indeed being actively hostile. Either way, I guess it will be necessary to connect the cooler to a Windows box after all...

(At least the Windows box doesn't necessarily needs to be the machine the cooler is installed into... maybe you can borrow a Windows laptop?)

rjousse18, 2023-01-26:
Mmmmh I don't have any Windows box in my house lol