liquidctl/liquidctl #568: Handle different psf/black versions and styles

jonasmalacofilho, 2023-02-17:
Black 23 introduces a new 2023 stable style. Unfortunately, many systems are still running Black 22 (or older), and cannot conveniently/locally fix issues arising from the new style.

At a glance, there doesn't seem to exist a configuration option to make Black 23 stick to the older 22 style. The only option I could find is to require a specific major Black version, but that doesn't really solve the issue: regardless of which version we pick (22 or 23), there will be some users running the other one.

jonasmalacofilho, 2023-02-17:
Related: https://github.com/psf/black/issues/3539

amezin, 2023-02-23:
> regardless of which version we pick (22 or 23), there will be some users running the other one.

You already have `tox.ini` in the repository. Why not run black through tox - both in CI and on developer's machines?

Another option is [pre-commit](https://pre-commit.com/) - look at pytest projects for examples of [configuration](https://github.com/pytest-dev/pytest/blob/main/.pre-commit-config.yaml) and [ci in action](https://results.pre-commit.ci/run/github/37489525/1677075016.CHaQnIJjRviHRQxwcpJutw). Although you'll likely still need tox to [manage pre-commit installation](https://github.com/pytest-dev/pytest/blob/9ccae9a8e3c7bb532eb7dc225f4c68f7dd5a8534/tox.ini#L69-L73)

jonasmalacofilho, 2023-02-23:
Can you look into it? If so, pick whatever you think is more reliable and ergonomic.

(The fact that we have a minimal tox config should not be taken into consideration. That's currently only a convenience for those that prefer to run all tests through tox. When deciding what approach to take, pretend like it's not there).

closed 2023-06-16 by jonasmalacofilho