liquidctl/liquidctl #569: Open invitation for new team members

jonasmalacofilho, 2023-02-17:
Hi all,

In the almost 5 years since its creation, the liquidctl project has been maintained mostly by me.

I've had some help, sure, from @MarshallAsch, @aleksamagicka, @amezin, and other driver authors and contributors. But so far it has remained, in essence, a one-person project.[^1] And I think it's time to change that.

All areas of the project could use new team members:

- issue triage and follow-up;
- reviewing of pull requests;
- bug fixes;
- new features;
- new drivers;
- refactoring and polishing of the codebase.

_(There's also an idea for a liquidctl 2.0: a different architecture – possibly daemon based – and implementation – possibly in Rust[^2])._

So please comment here, or write me an [email](mailto:jonas@protocubo.io), if you would like to join the team. Of course, for most of the areas mentioned above, you can just get started right now, and we'll make it official later.

_(A way to get reasonably up to speed is probably to read the open issues and pull requests [and their comments], check out the developer documentation in `docs/developer`, and familiarize yourself the codebase. Once you get started, make sure to also check out relevant old issues and pull requests, as well the commit history of relevant files)._

In the meantime, I ask users and contributors to accept a higher delay in all activities of the project, but especially in the triage of new issues and review of non-trivial pull requests. Further, please consider the next planned[^3] releases as postponed.

Thank you,
Jonas

[^1]: This doesn't apply to our sibling project [liquidtux](https://github.com/liquidctl/liquidtux), which is much more decentralized. But new contributors are welcome there as well.
[^2]: Although @amezin doesn't (or didn't) agree on that point.
[^3]: We usually do time-based minor releases every 13 weeks.

amezin, 2023-02-17:
> As for liquidctl 2.0, if it's in Rust I'm probably not going to be of much help - unless my future workplace requires, I don't plan on tackling it

Same. Rust is a purpose-built language - when you need high performance and reliability at the same time. RGB/Fan control software needs neither of these IMO (it just shouldn't be annoyingly inefficient). And reliability has its cost in development time and effort (Rust basically forces you to write highly reliable code). In practice, nothing bad would (or at least should) happen even if a RGB/fan controlling daemon restarts.

So yes, I'm still opposing Rust for this particular project. There are other reasons, but I believe this one should be enough.

And Python should allow transitioning existing drivers to the new daemon architecture. Yes, I believe (based on nzxt smart device driver) that existing liquidctl code needs a refactoring, but not a complete rewrite.

I've experimented with GLib/GObject and C - because there is pygobject, and because GLib's event loop can be used for HID devices on Windows and Linux as is.

But a pure-Python (+ctypes, asyncio) implementation is possible too. Rust implementation would require a lot more time and effort IMO.

BTW, if you need any help with existing liquidctl - especially in areas of CI, test automation - let me know.

jonasmalacofilho, 2023-02-17:
@aleksamagicka ,

> As for liquidctl 2.0, if it's in Rust I'm probably not going to be of much help - unless my future workplace requires, I don't plan on tackling it (famous last words).

For now a 2.0 is just an idea, and it being Rust is not even that.

I feel that _if_ we decide on a big rewrite (not for the language, but for the new architecture), then we might also want to reconsider the language (and Rust might be one contender, besides keeping it in Python, or using something else).

Even then a big factor would be what languages the team members are comfortable with, or willing to learn.

> If I understand correctly, liquidctl is now in semi-hiatus mode?

Not yet. For now I'm just admitting that I don't have the bandwidth anymore to promptly react to every issue/PR/discussion. A hiatus is indeed a possibility, but only if the bus factor remains one during the next few months.

---

@amezin ,

I disagree with a few things you wrote, but we don't need to discuss it now (see above my response to Aleksa regarding the mention of Rust).

> BTW, if you need any help with existing liquidctl - especially in areas of CI, test automation - let me know.

Our CI setup is nothing to be proud of, but I don't think it's the critical problem.

Besides bug fixes and new planned features/drivers, the codebase (including the tests) needs some cleanup and refactoring. Plus something I forgot on the original comment: type annotations (and type checking in CI).

jonasmalacofilho, 2023-10-25:
@Serphentas is joining the team. Thank you, and welcome!

---

Also, thank you too, @aleksamagicka! This year you've been much more active than me both here and on liquidtux.

aleksamagicka, 2023-10-25:
Hi @Serphentas! Glad to have you on board.

@jonasmalacofilho Thank you! I plan to be more active than now, there's lots of stuff to implement.

luksquaresma, 2023-10-31:
Hello all,

I got to know this repo searching how to control/optimize my `Corsair iCUE H115i RGB Pro XT 280mm`. Sadly I did not have much luck (also I'm not sure about this model support due to some naming overlaps on Corsair products).

I have some experience with python, rust and java and I would like to help in any way I can. Please contact me if you're interested @jonasmalacofilho , so we can arrange a online meeting.

Serphentas, 2023-11-04:
@aleksamagicka thanks for the welcome !

Ekmansoft, 2024-11-16:
I have a Asus Ryujin (1) that I'm trying to get to work. 

I have managed to get it to send requests to the device, so I can see firmware version, and some (probably faulty) fan speeds. 

I have not been able to initialize periodical reporting as seem to be active in Windows. 
There seem to be a "SET CONFIGURATION Request" sent in Windows that is missing in liquidctl. 
Could someone give a hint on how I can get that to be sent?

jonasmalacofilho, 2024-11-17:
@Ekmansoft take a look at our `asetek` driver, those devices require a similar request. By the way, in the future, please find or open a new _specific_ issue to discuss a device family, instead of posting on unrelated or general issues like this one. You can also reach out to me through email or Discord.