liquidctl/liquidctl #572: error in help page regarding fanspeeds

zeus86, 2023-02-22:
### Describe the bug

The help-page lists the options for fan-speed control as pairs of temperature and (pwm) percentage. This looks right, because setting a static speed will also work like that. however, the readme of this repo lists in it's examples, that it's actually pairs of temperature and rpm (which is the way, it actually had to be done). 

...this confused me quite a while. 
installed via pip on mint 22

### Commands executed

`liquidctl --help | grep \<temperature`

### Output of all relevant commands with `--debug` flag

` liquidctl [options] set <channel> speed (<temperature> <percentage>) ...`

### Affected device

Corsair commander Pro

### Operating system and version

Linux Mint 22, x86_64

### Installation method

pip

### Version of liquidctl

liquidctl v1.12.1 (Linux-5.15.0-60-generic-x86_64-with-glibc2.35)

zeus86, 2023-02-22:
edit: maybe the behaviour has changed here, because e.G. CoolerControl falls also short here, and tries to apply pwm-percentages instead of rpm on custom-curves

jonasmalacofilho, 2023-02-23:
> however, the readme of this repo lists in it's examples, that it's actually pairs of temperature and rpm (which is the way, it actually had to be done).

Can you point to where in `./README.md` that's said?

<details>
<summary>grep -n 'set .\+ speed' README.md</summary>

```
54:# liquidctl --match kraken set fan speed  20 30  30 50  34 80  40 90  50 100
55:# liquidctl --match kraken set pump speed 70
512:# liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
513:# liquidctl [options] set <channel> speed <percentage>
683:ExecStart=liquidctl --match kraken set pump speed 90
684:ExecStart=liquidctl --match kraken set fan speed  20 30  30 50  34 80  40 90  50 100
685:ExecStart=liquidctl --match "smart device" set sync speed 55
711:liquidctl --match kraken set pump speed 90
712:liquidctl --match kraken set fan speed  20 30  30 50  34 80  40 90  50 100
713:liquidctl --match "smart device" set sync speed 55
```

</details>

---

In any case, the `--help` output is correct in the general case, and the profile should generally be specified as temperature–percentage pairs.

However, due to a hack/bug, the Commander Pro works a bit differently, and we have yet to fix that (see: #480, #485, 9c3ca0be8e4c2df3ca4225a833f303c771c69be2).

zeus86, 2023-02-23:
uhm, i'm not able to find it in the readme as well, maybe i missread, and found it in the corsair commander pro subpage. 
However, i find this quite confusing anyways andthe help page should IMHO  list, that the behaviour there is sometimes device-dependent (especially because this a pretty wide-spread device).

jonasmalacofilho, 2023-02-23:
See if you can improve the documentation (`--help`, `man` page and guides) further and send me a PR.

Just note that while the Commander Pro is a fairly popular controller, it's not particularly more important to us than other devices. And given how fragmented the Corsair ecoysystem is, there might be other devices we support that are more popular than it, at least among those using liquidctl.

So don't make the documentation a mess just because of (our hack for) the Commander Pro. Also, the correct behavior takes precedence over the hack/exception.

jonasmalacofilho, 2023-02-23:
Alternatively, maybe you can work on #480 and on fixing the Commander Pro driver behavior?

closed 2023-05-11 by jonasmalacofilho