liquidctl/liquidctl #575: liquidctl fails with OverflowError on Corsair HX1500i

dsully, 2023-02-26:
### Describe the bug

I'm getting an error with a Corsair HX1500i on Ubuntu (22.10):

ERROR: Corsair HX1500i (experimental): unexpected error: OverflowError('Python int too large to convert to C int')

This looks similar to #463

### Commands executed

```
$ sudo liquidctl status
NZXT Kraken Z (Z53, Z63 or Z73) (experimental)
├── Liquid temperature    25.4  °C
├── Pump speed             986  rpm
├── Pump duty               25  %
├── Fan speed              553  rpm
└── Fan duty                20  %

ERROR: Corsair HX1500i (experimental): unexpected error: OverflowError('Python int too large to convert to C int')
```

### Output of all relevant commands with `--debug` flag

```
[DEBUG] (cli) (_log_env_infos): script: /home/dsully/dev/src/liquidctl/.direnv/python-3.10.7/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0.dev7+gdd2d1c1
[DEBUG] (cli) (_log_env_infos): platform: Linux-5.19.0-31-generic-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.10.7 (main, Nov 24 2022, 19:45:47) [GCC 12.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.13.1
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1e71:3008 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (open): selected interface: 0
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken Z (Z53, Z63 or Z73) (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1c1f (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): CorsairHidPsu identified: Corsair HX1500i (experimental)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1e71:201b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1e71:201b
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 174c:3074
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 0e8d:0608
[DEBUG] (usb) (find_devices): USB device: 05ac:024f
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 174c:2074
[DEBUG] (usb) (find_devices): USB device: 1b1c:1c1f
[DEBUG] (usb) (find_devices): USB device: 1e71:3008
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT Kraken Z (Z53, Z63 or Z73) (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writing report 0x74 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 75:01:3d:00:48:00:11:51:31:31:36:33:36:35:01:19:04:da:03:19:19:01:02:27:02:14:14:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (cli) (main): device: Corsair HX1500i (experimental)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writing report 0x00 with 64 bytes: 02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 02:00:00:c6:00:00:41:49:52:20:48:58:31:35:30:30:69:20:50:53:55:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writing report 0x00 with 64 bytes: 03:88:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 03:88:e6:f8:00:00:41:49:52:20:48:58:31:35:30:30:69:20:50:53:55:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (write): writing report 0x00 with 64 bytes: 03:d2:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 03:d2:13:c6:00:00:41:49:52:20:48:58:31:35:30:30:69:20:50:53:55:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (cli) (log): detailed error: Corsair HX1500i (experimental): unexpected error: OverflowError('Python int too large to convert to C int')
Traceback (most recent call last):
  File "/home/dsully/dev/src/liquidctl/.direnv/python-3.10.7/lib/python3.10/site-packages/liquidctl/cli.py", line 469, in main
    status = dev.get_status(**opts)
  File "/home/dsully/dev/src/liquidctl/.direnv/python-3.10.7/lib/python3.10/site-packages/liquidctl/driver/corsair_hid_psu.py", line 245, in get_status
    return self._get_status_directly()
  File "/home/dsully/dev/src/liquidctl/.direnv/python-3.10.7/lib/python3.10/site-packages/liquidctl/driver/corsair_hid_psu.py", line 170, in _get_status_directly
    ('Current uptime', self._get_timedelta(_CORSAIR_READ_UPTIME), ''),
  File "/home/dsully/dev/src/liquidctl/.direnv/python-3.10.7/lib/python3.10/site-packages/liquidctl/driver/corsair_hid_psu.py", line 310, in _get_timedelta
    return timedelta(seconds=secs)
OverflowError: Python int too large to convert to C int
[ERROR] (cli) (log): Corsair HX1500i (experimental): unexpected error: OverflowError('Python int too large to convert to C int')
NZXT Kraken Z (Z53, Z63 or Z73) (experimental)
├── Liquid temperature    25.4  °C
├── Pump speed             986  rpm
├── Pump duty               25  %
├── Fan speed              551  rpm
└── Fan duty                20  %
```

### Affected device

Corsair HX1500i

### Operating system and version

Ubuntu 22.10 / 5.19.0-31-generic

### Installation method

git clone + virtualenv

### Version of liquidctl

liquidctl v1.13.0.dev7+gdd2d1c1 (Linux-5.19.0-31-generic-x86_64-with-glibc2.36)

jonasmalacofilho, 2023-02-26:
That's actually a different issue, as the response is supposed to be valid in this case. More precisely, it matches the command that was sent, which is a first when we get these overflow errors.

But I think I fixed the issue in 2dd235e. Please test it and let me know.

closed 2023-02-26 by jonasmalacofilho

dsully, 2023-02-26:
That fixed it - thanks for the quick turn around:

```
$ sudo /home/dsully/dev/src/liquidctl/.direnv/python-3.10.7/bin/liquidctl status
NZXT Kraken Z (Z53, Z63 or Z73) (experimental)
├── Liquid temperature    25.5  °C
├── Pump speed             991  rpm
├── Pump duty               25  %
├── Fan speed              551  rpm
└── Fan duty                20  %

Corsair HX1500i (experimental)
├── Current uptime             15:52:25
├── Total uptime               17:42:25
├── VRM temperature                49.5  °C
├── Case temperature                0.0  °C
├── Fan control mode           Hardware
├── Fan speed                         0  rpm
├── Input voltage                115.00  V
├── +12V OCP mode            Multi rail
├── +12V output voltage           11.91  V
├── +12V output current            9.50  A
├── +12V output power            112.00  W
├── +5V output voltage             5.06  V
├── +5V output current             3.50  A
├── +5V output power              17.50  W
├── +3.3V output voltage           3.30  V
├── +3.3V output current           0.00  A
├── +3.3V output power             0.00  W
├── Total power output           128.00  W
├── Estimated input power       1753.00  W
└── Estimated efficiency              7  %
```

commit 2dd235e5 (2023-02-26): corsair_hid_psu: read exactly 4 bytes for timedeltas

The response from the device may contain non-nil bytes beyond the range
applicable to the command being replied to.

This is generally taken care of by LINEAR11, ULINEAR16, or when reading
a single byte. However, as we weren't sure how many bytes could actually
be returned by _get_timedelta, it was allowed to try and parse the
entire response as an integer.

That was fine since we were reading with little endianess, and no
devices were known that actually send extra non-nil bytes. That has
changed with the H1500i: only 4 bytes can be safely [... cut]