liquidctl/liquidctl #579: MSI MEG Coreliquid S280

hikkidev, 2023-03-12:
### Device type

AIO Coreliquid Cooler

### Product page

https://www.msi.com/Liquid-Cooling/MEG-CORELIQUID-S280

### First-party software

MSI Center

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles, voltages, currents or power measurements

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, voltages, current or power limits, lighting of embedded LEDs, lighting of accessories like RGB fans or LED strips

### Physical connection

USB + 3-pin CPU

### Additional information about the device

```
$ usb-devices
T:  Bus=04 Lev=01 Prnt=01 Port=10 Cnt=01 Dev#=  4 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0db0 ProdID=6a04 Rev=01.00
S:  Manufacturer=MSI
S:  Product=MEG Core Liquid S280
S:  SerialNumber=<HIDDEN_BY_ME>
C:  #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=00 Driver=(none)
E:  Ad=04(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=83(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:  If#= 1 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=02(O) Atr=03(Int.) MxPS=  64 Ivl=5ms
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=5ms
```
Details:
```
Bus 004 Device 004: ID 0db0:6a04 Micro Star International MEG Core Liquid S280
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0db0 Micro Star International
  idProduct          0x6a04 
  bcdDevice            1.00
  iManufacturer           1 MSI
  iProduct                2 MEG Core Liquid S280
  iSerial                 3 <HIDDEN_BY_ME>
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     154
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

nyok1912, 2024-10-11:
I have the same device and msi center is horrible! 
Hope it will be supported soon!

sarzeaud, 2024-11-15:
I have the S360 version, and I would also like some help to have this device work on linux.

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.