liquidctl/liquidctl #583: unspecified liquidctl error / response does not match command

p1r473, 2023-03-20:
### Describe the bug

Hello,
I am testing out ChimeraOS and am using liquidctl.
I am trying to get my Corsair Commander Core XT to match my GPU speeds, whereas my Corsair H170i Elite use my custom fan profile.

I have created a startup script:
```
#!/bin/bash
liquidctl initialize all
liquidctl --serial REDACTED set pump speed 100
/home/gamer/gpu.sh &
/home/gamer/yoda.py --serial REDACTED control fans with "(20,20),(30,30),(40,40),(50,50),(60,60),(70,70),(80,80),(100,90)" on coretemp.package_id_0 &
```

and the contents of gpu.sh:
```
!/bin/bash
while true; do
    temp=$(nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader)
    if [ "$temp" -ge 90 ]; then
        liquidctl --serial REDACTED set fans speed 0
    else
        liquidctl --serial REDACTED set fans speed "$temp"
    fi
    sleep 1
done
```

However, on the latter part, where my Core XT should match my GPU speeds, I get either one of the two following errors:
unspecified liquidctl error / response does not match command

Also please let me know if there is an easier way. There was no GPU sensor so I am just getting the output from `nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader)`

### Commands executed

`liquidctl initialize all`  : works fine

`liquidctl --serial REDACTED set fans speed 0 --verbose` :  unspecified liquidctl error
`liquidctl --serial REDACTED  set pump speed 100 --verbose` : no output. works fine

`/home/gamer/yoda.py --serial REDACTED control fans with "(20,20),(30,30),(40,40),(50,50),(60,60),(70,70),(80,80),(100,90)" on coretemp.package_id_0 --verbose` : works fine

### Output of all relevant commands with `--debug` flag

`liquidctl initialize all`
```
Corsair Commander Core XT (experimental)
├── Firmware version        1.3.54  
├── RGB port 1 LED count       N/A  
├── RGB port 2 LED count         4  
├── RGB port 3 LED count         4  
├── RGB port 4 LED count         4  
├── RGB port 5 LED count         4  
├── RGB port 6 LED count         4  
├── RGB port 7 LED count         4  
├── Fan port 1 connected       Yes  
├── Fan port 2 connected       Yes  
├── Fan port 3 connected       Yes  
├── Fan port 4 connected       Yes  
├── Fan port 5 connected       Yes  
├── Fan port 6 connected       Yes  
├── Temperature sensor 1       Yes  
└── Temperature sensor 2       Yes  

Corsair Commander Core (experimental)
├── Firmware version            2.11.221  
├── AIO LED count                    N/A  
├── RGB port 1 LED count               4  
├── RGB port 2 LED count               4  
├── RGB port 3 LED count               4  
├── RGB port 4 LED count               4  
├── RGB port 5 LED count               4  
├── RGB port 6 LED count               4  
├── AIO port connected               Yes  
├── Fan port 1 connected             Yes  
├── Fan port 2 connected             Yes  
├── Fan port 3 connected             Yes  
├── Fan port 4 connected             Yes  
├── Fan port 5 connected             Yes  
├── Fan port 6 connected             Yes  
├── Water temperature sensor         Yes  
└── Temperature sensor 1             Yes  

ASUS Aura LED Controller (experimental)
└── Firmware version    AULA3-AR32-0207  
```

`liquidctl --serial REDACTED set fans speed 0 --verbose`
```
INFO: detailed error: Corsair Commander Core XT (experimental): unspecified liquidctl error: ExpectationNotMet('device returned incorrect data type')
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 475, in main
    _device_set_speed(dev, args, **opts)
  File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 294, in _device_set_speed
    dev.set_fixed_speed(args['<channel>'].lower(), int(args['<percentage>'][0]), **opts)
  File "/usr/lib/python3.10/site-packages/liquidctl/driver/commander_core.py", line 147, in set_fixed_speed
    res = self._read_data(_MODE_HW_SPEED_MODE, _DATA_TYPE_HW_SPEED_MODE)
  File "/usr/lib/python3.10/site-packages/liquidctl/driver/commander_core.py", line 207, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: unspecified liquidctl error
ERROR: Corsair Commander Core XT (experimental): unspecified liquidctl error
```

`liquidctl --serial REDACTED  set pump speed 100 --verbose` : no output. works fine
`/home/gamer/yoda.py --serial REDACTED control fans with "(20,20),(30,30),(40,40),(50,50),(60,60),(70,70),(80,80),(100,90)" on coretemp.package_id_0 --verbose` : works fine
```
INFO: device: Corsair Commander Core (experimental) on bus hid and address /dev/hidraw27
INFO: channel: fans following profile [(20, 20), (30, 30), (40, 40), (50, 50), (60, 60), (70, 70), (80, 80), (100, 100)] on coretemp.package_id_0
INFO: update interval: 2 s; cutoff frequency (low-pass): 0.05 Hz; ema alpha: 0.27
INFO: starting...
INFO: fans control: lpf(coretemp.package_id_0) = lpf(29.0°C) = 29.0°C => duty := 29%
INFO: fans control: lpf(coretemp.package_id_0) = lpf(29.0°C) = 29.0°C => duty := 29%
INFO: fans control: lpf(coretemp.package_id_0) = lpf(29.0°C) = 29.0°C => duty := 29%
```

### Affected device

Corsair Commander Cote XT

### Operating system and version

ChimeraOS (ArchLinux)

### Installation method

came built with ChimeraOS

### Version of liquidctl

liquidctl v1.12.1 (Linux-6.1.11-arch1-1-x86_64-with-glibc2.37)

jonasmalacofilho, 2023-03-25:
The Commander Core XT issue seems to be related to #520. We think we fixed what was causing _liquidctl_ to put the device in that state, but that fix could be insufficient for the Core XT.

Another possibility is that something _else_ is putting your device in that "confused" state. We also have a patch in review for auto-correcting corrupted device state, regardless of what caused it (#522), but it hasn't been tested yet, and I authored it without having access to one of these devices (so it could be trivially broken). It would be great if you could review and possibly test it. (The discussions in the issues linked from that PR describe what we suspect is going on).

---

Regarding yoda, like you said:

> Its (temperature,duty)

If you "want 100% at 90 degrees and above", you should end your profile with `(90, 100)` (temperature, 90°C, duty, 100%). The same goes for the rest of your profile.

Additionally, you can omit points that don't change the derivative; another way to think about it is that if a point already lies in the line defined by its immediate neighbors, it's redundant.

```
$ /home/gamer/yoda.py --serial REDACTED control fans with "(20,0),(30,30),(80,80),(90,100)" on coretemp.package_id_0 --verbose
```

p1r473, 2023-03-25:
> Regarding yoda, like you said:
>
>
> If you "want 100% at 90 degrees and above", you should end your profile with `(90, 100)` (temperature, 90°C, duty, 100%). The same goes for the rest of your [...]

D'oh!
Thanks- good find!

I'll test the PR of liquidctl with my CoCo XT

jonasmalacofilho, 2024-12-16:
Merged with other similar issues into #753.

closed 2024-12-16 by jonasmalacofilho