liquidctl/liquidctl #587: NZXT Kraken X63 not listing on PopOS!

crazyrobinhood, 2023-04-10:
Hello there,
 here is the output on my Pop!_OS 22.04 LTS x86_64 with 6.2.6-76060206-generic kernel.
Any idea why isn't listed? The list is totally empty.
Pump is plugged and lighted up.
(Bios recognize it).

Thank you in advance for your work and your time.

```
liquidctl list --debug --verbose
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 12cf:0182
[DEBUG] liquidctl.driver.usb: found HID device 12cf:0182
[DEBUG] liquidctl.driver.usb: found HID device 12cf:0182
[DEBUG] liquidctl.driver.usb: found HID device 12cf:0182
[DEBUG] liquidctl.driver.usb: found HID device 046d:c299
[DEBUG] liquidctl.driver.usb: found HID device 0b05:1939
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: found HID device 258a:0190
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 0781:5591
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 0806:0806
[DEBUG] liquidctl.driver.usb: found USB device 258a:0190
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1235:8205
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 0b05:1939
[DEBUG] liquidctl.driver.usb: found USB device 046d:c299
[DEBUG] liquidctl.driver.usb: found USB device 12cf:0182
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
```

aleksamagicka, 2023-04-11:
`[DEBUG] liquidctl.cli: running liquidctl v1.5.1`

You're using liquidctl v1.5.1. That's over two years old and the minimum recommended version for the Kraken X63, which you can look up in [the table](https://github.com/liquidctl/liquidctl#supported-devices) in the README, is 1.11.1.

crazyrobinhood, 2023-04-13:
Thank you for your time.

closed 2023-04-13 by crazyrobinhood