liquidctl/liquidctl #592: weird issue with rgb

Trae132, 2023-04-20:
Im using liquidctl with fan control and love it however ever since i started using the plugin my rgb lights go out on motherboard as soon as it launches?  Any way to disable the lighting effects of the program?  Im assuming its the liquidctl because i made no other changes of any kind and it goes out right as it launches.

Trae132, 2023-04-21:
Yup thats the plugin and Asus motherboard indeed, thanks for the reply I thought I was losing my mind for a minute but I guess it was in fact linked to the liquidctl plugin.  Ive reached out to the plugin author hopefully they will help.  This is not something I could fix im guessing, of course id have no clue how but just a sort of is it possible thing?

aleksamagicka, 2023-04-21:
Of course it's possible, and seems rather easy - it parses all available devices [here](https://github.com/jmarucha/FanControl.Liquidctl/blob/master/LiquidctlPlugin.cs#L30), and for each it gets its `description`, which can be used to selectively initialize it: `liquidctl initialize -m <description>`. As there aren't that many drivers, the easiest way would be to just hardcode it. Check if the current device is the Aura LED controller and if it is, don't initialize it. Otherwise, initialize and carry on as usual.

Disclaimer, I don't know what the main program does with the plugin methods, but something like this should work.

Trae132, 2023-04-21:
Big thank you for taking the time to help me with this.  So I open cmd from the plugin folder and type this stuff there?  What exactly do I put in cmd?  Aura LED is absolutely what it is because in bios thats what I selected off at shutdown and it turned the same light off.

aleksamagicka, 2023-04-21:
No, I described a strategy for the code. I'll try to implement that in the
plugin.

closed 2023-06-16 by jonasmalacofilho

Trae132, 2023-08-06:
> No, I described a strategy for the code. I'll try to implement that in the plugin.
> […](#)

Heya just checking back in with you, its been a few months and wondering if you might of made some changes or remind you of this convo we had.  Thanks aleks!!

aleksamagicka, 2023-08-07:
I couldn't get liquidctl to build as an exe with the script from that repo and I wasn't in a mood to fight PyInstaller. The one exe in releases (from some time ago) does not seem to support the ASUS LEDs. If you have an idea on how to build it from git master, I can try and see what I can do regarding the RGB init. (But at that repo.)